sony/flutter-elinux #17: Build error

suyulin, 2021-07-22:
hi,
[image]
[image]

HidenoriMatsubayashi, 2021-07-26:
Could you inform me of your environment? It seems the version of clang is a little bit old.

Nagendra-Bankupalli, 2021-07-28:
@HidenoriMatsubayashi 
got similar error while cross building flutter test application 
Downloading elinux-x64-release tools...                          1,608ms

💪 Building with sound null safety 💪

Building an eLinux application with wayland backend in release mode for arm64 target...        17.4s
Failed to cmake:
-- Toolchain file defaulted to '/opt/tdx-wayland/5.1.0/sysroots/x86_64-abcsdk-linux/usr/share/cmake/OEToolchainConfig.cmake'
-- The CXX compiler identification is Clang 10.0.1
-- Check for working CXX compiler: /opt/ABC-wayland/5.1.0/sysroots/x86_64-abcsdk-linux/usr/bin/clang++
-- Check for working CXX compiler: /opt/ABC-wayland/5.1.0/sysroots/x86_64-abcsdk-linux/usr/bin/clang++ -- broken
-- Configuring incomplete, errors occurred!
See also "~/test_app/build/elinux/arm64/release/CMakeFiles/CMakeOutput.log".
See also "~/test_app/build/elinux/arm64/release/CMakeFiles/CMakeError.log".

CMake Error at /opt/tdx-wayland/5.1.0/sysroots/x86_64-abcsdk-linux/usr/share/cmake-3.16/Modules/CMakeTestCXXCompiler.cmake:53 (message):
  
could you help me in fixing it?

my flutter-elinux doctor says 
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel unknown, 2.4.0-4.2.pre, on Ubuntu 20.04.1 LTS 5.8.0-63-generic, locale en_IN)
[✓] eLinux toolchain - develop for embedded Linux devices
[!] Android toolchain - develop for Android devices (Android SDK version 30.0.3)
    ✗ cmdline-tools component is missing
      Run `path/to/sdkmanager --install "cmdline-tools;latest"`
      See https://developer.android.com/studio/command-line for more details.
    ✗ Android license status unknown.
      Run `flutter doctor --android-licenses` to accept the SDK licenses.
      See https://flutter.dev/docs/get-started/install/linux#android-setup for more details.
[✓] Chrome - develop for the web
[✓] Linux toolchain - develop for Linux desktop
[✓] Android Studio (version 4.1)
[✓] Android Studio
[✓] Connected device (4 available)

regsrds
Nagendra

HidenoriMatsubayashi, 2021-07-29:
@Nagendra-Bankupalli You try to cross-compile, right? If so, please see https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps#2-cross-building-from-x64-to-arm64

Nagendra-Bankupalli, 2021-07-29:
@HidenoriMatsubayashi ,
Yes , the  Link suggests to use tool-chain (sdk) specific to the target . I am using sdk built using yocto for my target , which includes flutter engine as well. the same tool chain worked flawlessly to build flutter-embedder "flutter-client" .

HidenoriMatsubayashi, 2021-07-29:
I understood. Is it possible to try the build by specifying only your sysroot **without using the Yocto SDK's toolchain**? It means 
using the host toolchain with the target sysroot.

e.g:
```Shell
$ flutter-elinux build elinux --target-arch=arm64 --target-sysroot=<absolute_path_to_yocto_sdk_sysroot>
```

Nagendra-Bankupalli, 2021-07-29:
@HidenoriMatsubayashi , below is the error

Building an eLinux application with wayland backend in release mode for arm64 target...        17.5s
Failed to cmake:
-- The CXX compiler identification is Clang 10.0.0
-- Check for working CXX compiler: /usr/bin/clang++
-- Check for working CXX compiler: /usr/bin/clang++ -- broken
-- Configuring incomplete, errors occurred!
See also "/home/<path>/test_app/build/elinux/arm64/release/CMakeFiles/CMakeOutput.log".
See also "/home/<path>/test_app/build/elinux/arm64/release/CMakeFiles/CMakeError.log".

CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCXXCompiler.cmake:53 (message):
  The C++ compiler

    "/usr/bin/clang++"

  is not able to compile a simple test program.

HidenoriMatsubayashi, 2021-07-29:
Thanks. Can you pass self-build?

```Shell
$ flutter-elinux build elinux
```

Nagendra-Bankupalli, 2021-07-29:
It works for x64 target (weston) and I can even run profile (dev tool) as well.

HidenoriMatsubayashi, 2021-07-29:
Did you install required tools to cross-build?

```Shell
$ sudo apt install gcc-aarch64-linux-gnu g++-aarch64-linux-gnu
$ sudo apt install libstdc++-8-dev-arm64-cross
```

Nagendra-Bankupalli, 2021-07-29:
No, I thought target tool chain would be sufficient , will try and let you know

HidenoriMatsubayashi, 2021-07-29:
Yes, we need these tools to cross-build when we use host's toolchain (clang/llvm).

Nagendra-Bankupalli, 2021-07-29:
@HidenoriMatsubayashi , thank you for the inputs,
now the build is success, and able to run the app on target using (from host)
flutter-elinux  run -d  <My-device> 

on profiling, I have updated .flutter_custom_devices.json with all required info about my target
but flutter-elinux attach says  (running on another host window)
"Waiting for a connection from Flutter on eLinux..."

HidenoriMatsubayashi, 2021-07-29:
Sounds good.

`$ flutter-elinux run -d My-device` is debug-mode (not profile mode), and currently, custom-devices support only debug mode. If you want to use the observatory, write `forwardPort` and `forwardPortSuccessRegex` in `.flutter_custom_devices.json`. When we attach debugger on remote devices, we need to port-forwarding.

```JSON
      "runDebug": [
        "ssh", "user@192.168.0.100", "/tmp/${appName}/${appName} -b ."
      ],
      "forwardPort": [
        "ssh", "-o", "ExitOnForwardFailure=yes",
        "-L", "127.0.0.1:${hostPort}:127.0.0.1:${devicePort}", "user@192.168.0.100"
      ],
      "forwardPortSuccessRegex": "Linux"
```

Nagendra-Bankupalli, 2021-07-29:
Yes, forwardPort and forwardPortSuccessRegex have been added to my flutter_custom_devices.json
According to 
https://github.com/sony/flutter-elinux/wiki/Remote-target-devices

below commands can be used to install and uninstall app from the target 
flutter-elinux install -d  my-device ,flutter-elinux uninstall -d my-device
in the above commands uninstall and forwardPort does not work for me throwing unknown commands , 

Tried running app on target and shows below line, does this means we can observe it on host, if portForward is success, if yes please let me know how to do it? am I missing anything ?
flutter: Observatory listening on http://127.0.0.1:41285/Q3dpAt3Ta3g=/

HidenoriMatsubayashi, 2021-07-29:
Could you inform me of your trying command and error messages?

Nagendra-Bankupalli, 2021-07-29:
with below commands I could get observatory data on host chrome from target (my-device)

target :  
`root@my-device:/tmp/test_app# ./test_app  -b .
flutter: Observatory listening on http://127.0.0.1:34549/CA1Zk5rr-LQ=/
`
host 
ssh -o ExitOnForwardFailure=yes -L 127.0.0.1:34549:127.0.0.1:34549 root@ip-my-device

open chrome on host  "http://127.0.0.1:34549/CA1Zk5rr-LQ=/"

[image]

HidenoriMatsubayashi, 2021-07-29:
Yes, `$ flutter-elinux run -d My-device` does the same thing automatically as above.

Internal steps:
1. [uninstall comamnd] uninstall your old flutter app from `My-device`
2. [install command] install your flutter app to `My-device`
3. [runDebug command] run your flutter app on `My-device`
4. [forwardPort command] portforwarding
   You will get log message like "http://127.0.0.1:34549/CA1Zk5rr-LQ=/" and you can open it on host web-browser

Nagendra-Bankupalli, 2021-07-29:
Yes,I expected the same, but the command did not provide any log , so I tried it manually , it worked
between , this is not actual profile data right ? like DevTool ?

``` 
host $  flutter-elinux -v run -d my-device
Launching lib/main.dart on eLinux in debug mode...
Uninstall test_app from my-device.
Uninstallation Success
Install test_app (build/elinux/arm64/debug/bundle) to my-device
Installation Success
Launch test_app.name on my-device

```
`

HidenoriMatsubayashi, 2021-07-30:
In my environment, it works fine.
```Shell
$ flutter-elinux run -d raspberry-pi4
Launching lib/main.dart on eLinux in debug mode...
Uninstall sample from raspberry-pi4.
hidenori@192.168.0.10's password: 
Uninstallation Success
Install sample (build/elinux/arm64/debug/bundle) to raspberry-pi4
hidenori@192.168.0.10's password: 
Installation Success
Launch sample.name on raspberry-pi4
hidenori@192.168.0.10's password: 
hidenori@192.168.0.10's password: 
Syncing files to device eLinux...                                  202ms

Flutter run key commands.
r Hot reload. 🔥🔥🔥
R Hot restart.
h List all available interactive commands.
d Detach (terminate "flutter run" but leave application running).
c Clear the screen
q Quit (terminate the application on the device).

💪 Running with sound null safety 💪

An Observatory debugger and profiler on eLinux is available at: http://127.0.0.1:43237/xd6sW5hdEoE=/
Activating Dart DevTools...                                      1,830ms
The Flutter DevTools debugger and profiler on eLinux is available at:
http://127.0.0.1:9100?uri=http://127.0.0.1:43237/xd6sW5hdEoE=/
```

I couldn't reproduce this issue. So, could you do debugging by adding log messages to around https://github.com/sony/flutter-elinux/blob/main/lib/elinux_device.dart#L141?

> between , this is not actual profile data right ? is profile mode supported in future ?

Why do you want to use profile mode? I think debug mode is well enough. The difference between debug and profile mode is that debug mode is debugging + profiling, whereas profile mode is profiling only.

HidenoriMatsubayashi, 2021-07-30:
> is profile mode supported in future ?

I added it just now. #34

Nagendra-Bankupalli, 2021-07-30:
@HidenoriMatsubayashi , 
1. Thank you for updating profile mode
2. Yes, I could get timeline info in debug mode, I observed that the profile mode is more sophisticated than debug mode, let me check again.
3. Regarding "flutter-elinux run" command, Could be my host settings? I will get back with more details later as it works with manual commands  for me now

HidenoriMatsubayashi, 2021-08-19:
@Nagendra-Bankupalli 

> Yes,I expected the same, but the command did not provide any log (app runs on target successfully), so I tried it manually , it worked

I can reproduce the same issue using i.mx8m device, but if I use raspberry-pi4, it's no problem. So, it looks depending on the target device or environment.

Nagendra-Bankupalli, 2021-08-19:
Thanks for updating, Yes mine is similar platform, I did not look into it as it is not blocking.

Nagendra-Bankupalli, 2021-08-22:
@HidenoriMatsubayashi , Have you tried flutter-gallery app on embedded target ?

HidenoriMatsubayashi, 2021-08-22:
Yes, I have. It worked fine.

Nagendra-Bankupalli, 2021-08-25:
@HidenoriMatsubayashi , 
its not working at my end not sure what is the issue, the application loads but no widgets visible , its white , except the flutter animation. any issues with GPU support ?, or how do we know that app is GPU driven? any inputs on this would be helpful.

HidenoriMatsubayashi, 2021-08-25:
Proverbially, I guess you haven't set `LANG` environment var. Could you try the following on your target device?

```
$ export LANG=en_US.UTF-8
```

Rustam-Askerov, 2021-11-16:
Hi,I created a flutter custom device, but when I entered the command "flutter-elinux devices" I found that instead of one device, two sailfish-devices are displayed. Because of this, when I enter the command "flutter-elinux run -d sailfish-device", debugging does not work correctly for me. What could this be related to?
![Снимок экрана от 2021-11-16 15-40-22](https://user-images.githubuser
[image]
content.com/71459422/141989426-ff9d24fb-e65a-48d4-a029-80dff15b9219.png)
[image]

Rustam-Askerov, 2021-11-16:
Hi,I created a flutter custom device, but when I entered the command "flutter-elinux devices" I found that instead of one device, two sailfish-devices are displayed. Because of this, when I enter the command "flutter-elinux run -d sailfish-device", debugging does not work correctly for me. What could this 
[image]
[image]
[image]

HidenoriMatsubayashi, 2021-11-17:
@Rustam-Askerov
Your target device (`sailfish-device`) looks localhost, it means you want to install your host machine instead of remote custom devieces, right?

Rustam-Askerov, 2021-11-17:
I run virtualbox on my pc and as a custom device I use a virtual machine on which sailfish os is installed. In the instructions for the Sailfish SDK, it says ssh
the connection to the virual box is done with the command "ssh -p 2222 -i ~ / SailfishOS / vmshare / ssh / private_keys / sdk root @ localhost". So I use localhost.
[image]

Rustam-Askerov, 2021-11-23:
Also I tried to build the app from x64 host for arm64 using cross compilation but when compilation reaches 100% the error appears:
"/usr/bin/aarch64-linux-gnu-ld: warning: libxkbcommon.so.0, needed for 
/home/raskerov/test_app/elinux/flutter/ephemeral/libflutter_elinux_wayland.so, not found
(try using -rpath or -rpath-link)"

> flutter-elinux build elinux --debug --target-arch=arm64 --target-sysroot=/opt/poky/2.1/sysroots/aarch64-poky-linux --system-include-directories=/usr/aarch64-linux-gnu/include/c++/10/aarch64-linux-gnu

>💪 Building with sound null safety 💪

>Building an eLinux application with wayland backend in debug mode for arm64 target...             12,1s
Failed to cmake build:
Scanning dependencies of target flutter_assemble
[  0%] Built target flutter_assemble
Scanning dependencies of target flutter_wrapper_app
[  7%] Building CXX object
flutter/CMakeFiles/flutter_wrapper_app.dir/ephemeral/cpp_client_wrapper/core_implementations.cc.o
[ 15%] Building CXX object
flutter/CMakeFiles/flutter_wrapper_app.dir/ephemeral/cpp_client_wrapper/standard_codec.cc.o
[ 23%] Building CXX object
flutter/CMakeFiles/flutter_wrapper_app.dir/ephemeral/cpp_client_wrapper/flutter_engine.cc.o
[ 30%] Building CXX object
flutter/CMakeFiles/flutter_wrapper_app.dir/ephemeral/cpp_client_wrapper/flutter_view_controller.cc.o
[ 38%] Linking CXX static library libflutter_wrapper_app.a
[ 38%] Built target flutter_wrapper_app
Scanning dependencies of target flutter_wrapper_plugin
[ 46%] Building CXX object
flutter/CMakeFiles/flutter_wrapper_plugin.dir/ephemeral/cpp_client_wrapper/core_implementations.cc.o
[ 53%] Building CXX object
flutter/CMakeFiles/flutter_wrapper_plugin.dir/ephemeral/cpp_client_wrapper/standard_codec.cc.o
[ 61%] Building CXX object
flutter/CMakeFiles/flutter_wrapper_plugin.dir/ephemeral/cpp_client_wrapper/plugin_registrar.cc.o
[ 69%] Linking CXX static library libflutter_wrapper_plugin.a
[ 69%] Built target flutter_wrapper_plugin
Scanning dependencies of target test_app
[ 76%] Building CXX object runner/CMakeFiles/test_app.dir/flutter_window.cc.o
[ 84%] Building CXX object runner/CMakeFiles/test_app.dir/main.cc.o
[ 92%] Building CXX object
runner/CMakeFiles/test_app.dir/__/flutter/generated_plugin_registrant.cc.o
[100%] Linking CXX executable test_app

>**/usr/bin/aarch64-linux-gnu-ld: warning: libxkbcommon.so.0, needed for 
/home/raskerov/test_app/elinux/flutter/ephemeral/libflutter_elinux_wayland.so, not found
(try using -rpath or -rpath-link)**
/usr/bin/aarch64-linux-gnu-ld: /opt/poky/2.1/sysroots/aarch64-poky-linux/usr/lib/libEGL.so.1:
.dynsym local symbol at index 3 (>= sh_info of 3)
/usr/bin/aarch64-linux-gnu-ld: /opt/poky/2.1/sysroots/aarch64-poky-linux/usr/lib/libEGL.so.1:
.dynsym local symbol at index 4 (>= sh_info of 3)
/usr/bin/aarch64-linux-gnu-ld: /opt/poky/2.1/sysroots/aarch64-poky-linux/usr/lib/libEGL.so.1:
.dynsym local symbol at index 5 (>= sh_info of 3)
/usr/bin/aarch64-linux-gnu-ld: /opt/poky/2.1/sysroots/aarch64-poky-linux/usr/lib/libEGL.so.1:
.dynsym local symbol at index 6 (>= sh_info of 3)
/usr/bin/aarch64-linux-gnu-ld: /opt/poky/2.1/sysroots/aarch64-poky-linux/usr/lib/libEGL.so.1:
.dynsym local symbol at index 7 (>= sh_info of 3)
/usr/bin/aarch64-linux-gnu-ld: /opt/poky/2.1/sysroots/aarch64-poky-linux/usr/lib/libEGL.so.1:
.dynsym local symbol at index 8 (>= sh_info of 3)
/usr/bin/aarch64-linux-gnu-ld: /opt/poky/2.1/sysroots/aarch64-poky-linux/usr/lib/libEGL.so.1:
.dynsym local symbol at index 9 (>= sh_info of 3)
/usr/bin/aarch64-linux-gnu-ld: /opt/poky/2.1/sysroots/aarch64-poky-linux/usr/lib/libgbm.so.1:
.dynsym local symbol at index 3 (>= sh_info of 3)
/usr/bin/aarch64-linux-gnu-ld: /opt/poky/2.1/sysroots/aarch64-poky-linux/usr/lib/libgbm.so.1:
.dynsym local symbol at index 4 (>= sh_info of 3)
/usr/bin/aarch64-linux-gnu-ld: /opt/poky/2.1/sysroots/aarch64-poky-linux/usr/lib/libgbm.so.1:
.dynsym local symbol at index 5 (>= sh_info of 3)
/usr/bin/aarch64-linux-gnu-ld: /opt/poky/2.1/sysroots/aarch64-poky-linux/usr/lib/libgbm.so.1:
.dynsym local symbol at index 6 (>= sh_info of 3)
/usr/bin/aarch64-linux-gnu-ld: /opt/poky/2.1/sysroots/aarch64-poky-linux/usr/lib/libgbm.so.1:
.dynsym local symbol at index 7 (>= sh_info of 3)
/usr/bin/aarch64-linux-gnu-ld: /opt/poky/2.1/sysroots/aarch64-poky-linux/usr/lib/libgbm.so.1:
.dynsym local symbol at index 8 (>= sh_info of 3)
/usr/bin/aarch64-linux-gnu-ld: /opt/poky/2.1/sysroots/aarch64-poky-linux/usr/lib/libgbm.so.1:
[... 26 lines cut ...]
/usr/bin/aarch64-linux-gnu-ld:
/home/raskerov/test_app/elinux/flutter/ephemeral/libflutter_elinux_wayland.so: undefined
link to «xkb_keysym_to_utf32@V_0.5.0»
/usr/bin/aarch64-linux-gnu-ld:
/home/raskerov/test_app/elinux/flutter/ephemeral/libflutter_elinux_wayland.so: undefined
link to «xkb_keymap_new_from_names@V_0.5.0»
/usr/bin/aarch64-linux-gnu-ld:
/home/raskerov/test_app/elinux/flutter/ephemeral/libflutter_elinux_wayland.so: undefined
link to «wl_proxy_marshal_constructor_versioned»
/usr/bin/aarch64-linux-gnu-ld:
/home/raskerov/test_app/elinux/flutter/ephemeral/libflutter_elinux_wayland.so: undefined
link to «xkb_keymap_mod_get_index@V_0.5.0»
/usr/bin/aarch64-linux-gnu-ld:
/home/raskerov/test_app/elinux/flutter/ephemeral/libflutter_elinux_wayland.so: undefined
link to «xkb_state_key_get_one_sym@V_0.5.0»
/usr/bin/aarch64-linux-gnu-ld:
/home/raskerov/test_app/elinux/flutter/ephemeral/libflutter_elinux_wayland.so: undefined
link to «xkb_context_new@V_0.5.0»
/usr/bin/aarch64-linux-gnu-ld:
/home/raskerov/test_app/elinux/flutter/ephemeral/libflutter_elinux_wayland.so: undefined
link to «xkb_state_update_key@V_0.5.0»
clang: error: linker command failed with exit code 1 (use -v to see invocation)
gmake[2]: *** [runner/CMakeFiles/test_app.dir/build.make:136: runner/test_app] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:219: runner/CMakeFiles/test_app.dir/all] Error 2
gmake: *** [Makefile:149: all] Errors 2

Rustam-Askerov, 2021-11-23:
But libxkbcommon0 library is installed:
[image]

suyulin, 2021-12-04:
hi,
on host device: 
```
flutter-elinux run -d mixpad-x
Launching lib/main.dart on eLinux in debug mode...
Uninstall hello_remote_device from mixpad-x.
Uninstallation Success
Install hello_remote_device (build/elinux/arm64/debug/bundle) to mixpad-x
Installation Success
Launch hello_remote_device.name on mixpad-x
```
no flutter hot load 
on target device:
```
./video_player_example -b ./
arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '3'.embedder.cc (1933): 'FlutterEnginePostDartObject' returned 'kInvalidArguments'. Engine not running.
[ERROR:engine/src/flutter/runtime/dart_vm_data.cc(18)] VM snapshot invalid and could not be inferred from settings.
[ERROR:engine/src/flutter/runtime/dart_vm.cc(256)] Could not setup VM data to bootstrap the VM from.
[ERROR:engine/src/flutter/runtime/dart_vm_lifecycle.cc(84)] Could not create Dart VM instance.
[FATAL:engine/src/flutter/shell/common/shell.cc(297)] Check failed: vm. Must be able to initialize the VM.
```

endless7, 2022-02-18:
Hi, @HidenoriMatsubayashi, i got this when i try to cross-compiling...
[image]

it was ok when i run `flutter-elinux run -d elinux-x11`, thx.

nguyenlkdn, 2024-04-09:
> ```shell
> Activating Dart DevTools...                                      1,830ms
> The Flutter DevTools debugger and profiler on eLinux is available at:
> http://127.0.0.1:9100?uri=http://127.0.0.1:43237/xd6sW5hdEoE=/
> ```

How can we setup for activating Dart DevTools on your RPi board?