sony/flutter-elinux #48: Add toolchain option to cross-build

HidenoriMatsubayashi, 2021-08-20:
Current cross-build uses the toolchain installed in the host so, we need to add user custom toolchain option to cross-build.

HidenoriMatsubayashi, 2022-04-07:
Now, I added `target-compiler-triple` option for cross-building.

seanchann, 2022-08-11:
hi, if  target-compiler-triple=aarch64-none-linux-gnu will broken compiler that  output error
```
    /usr/bin/ld: unrecognised emulation mode: aarch64linux
    Supported emulations: elf_x86_64 elf32_x86_64 elf_i386 elf_iamcu elf_l1om elf_k1om i386pep i386pe
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    gmake[1]: *** [CMakeFiles/cmTC_0ca6b.dir/build.make:100: cmTC_0ca6b] Error 1
    gmake[1]: Leaving directory '/home/seanchann/source/opensrc/flutter-extension/decodetest/build/elinux/arm64/release/CMakeFiles/CMakeTmp'
    gmake: *** [Makefile:127: cmTC_0ca6b/fast] Error 2
```

PR #110 (2022-08-12): Add custom toolchain option
Added `--target-toolchain` option.

Fixed #48
commit ed16e0bd (2022-08-12): Add custom toolchain option (#110)

Fixed #48
closed 2022-08-12 by HidenoriMatsubayashi