sony/flutter-elinux #56: GBM backend fails to start.

Charlie9830, 2021-09-15:
### Problem
Running the GBM backend build on the target device (rpi4) most commonly results in the
```"Failed to create view controller"``` error. If I reboot and try again, I can usually get it to work maybe 1 out of every 10 reboot cycles.

### Background
- I am using flutter-elinux via Yocto to build for a Raspberry Pi 4. Using bitbake directly (Not via a Cross-Compile toolchain). 

- I have a recipe for building the flutter-elinux tools, as opposed to using the existing flutter-sdk recipes. This saves me having to build the engine locally and the extra complications associated with that.

- I don't experience this issue when building as the default ```wayland-client``` and running from a compositor (in this case Cage).

- I did not have this issue when I was building using the current implementation of the ```meta-flutter``` repo, ie: Building the engine, building the drm-gbm embedder, compiling the flutter code and providing it to the embedder at runtime. _I have moved away from this implementation as the build system was very brittle, engine falling out of sync with sdk etc_

I understand that I am am off the mainline path with this particular build and the issue is likely in the particular configuration of my image, but any leads would be greatly appreciated. the ```Failed to create view controller``` error is quite generic and I'm not quite experienced enough with C++ to be able to dig into the error via the source code.

### Recipes

## flutter-elinux-sdk-native
_Builds the flutter-elinux tools_
```
SUMMARY = "Flutter for Embedded Linux"
DESCRIPTION = "Sony flavour of the Flutter tools geared for Embedded Linux development"
AUTHOR = "Sony"
BUGTRACKER = "https://github.com/sony/flutter-elinux/issues"
SECTION = "graphics"
CVE_PRODUCT = ""

LICENSE = "BSD-3-Clause"
LIC_FILES_CHKSUM = "file://LICENSE;md5=647d87ee0850d0ee715550d324dd447c"

DEPENDS += "curl-native unzip-native"

SRC_URI = "git://github.com/sony/flutter-elinux.git;branch=main \ 
	   file://ca-certificates.crt"

SRCREV = "${AUTOREV}"
#SRCREV = "073e2c265b437c049c2f7f38a3b1d70cb7071008"

S = "${WORKDIR}/git"

inherit native

do_compile() {
    export CURL_CA_BUNDLE=${WORKDIR}/ca-certificates.crt

    export PATH=${S}/bin:$PATH
    export PUB_CACHE=${S}/.pub-cache

    flutter-elinux doctor
}

do_install() {
    install -d ${D}${datadir}/flutter-elinux/sdk
    cp -rTv ${S}/. ${D}${datadir}/flutter-elinux/sdk
}

FILES_${PN}-dev = "${datadir}/flutter-elinux/sdk/*"

INSANE_SKIP_${PN}-dev = "already-stripped"

BBCLASSEXTEND = "native nativesdk"

```

## Castboard Player
_This is the Flutter App that fails to start when built with the GBM backend_
_Note: This recipe is setup to build for both the GBM and wayland client backends_
_References to castboard_core and castboard_remote can be ignored, they are just app level dependencies_
```
SUMMARY = "Castboard Player"
DESCRIPTION = "Player for the Castboard software suite built on flutter-elinux"
AUTHOR = "Charlie Hall"
HOMEPAGE = "https://github.com/charlie9830"
BUGTRACKER = "https://github.com/charlie9830"
SECTION = "graphics"
CVE_PRODUCT = ""

LICENSE = "CLOSED"

DEPENDS += "\
	flutter-elinux-sdk-native \
	unzip-native \
	castboard-core \
	castboard-remote\
	glib-2.0 \
	libinput \
	libxkbcommon \
	libdrm \
	virtual/libgles2 \
	virtual/egl \
	virtual/mesa \
	udev \
	systemd \
	"
	
DEPENDS += "\
	wayland-protocols \
	wayland-native \
"

RDEPENDS_${PN} += "\
    xkeyboard-config \
    fontconfig \
"

SRC_URI = "git://github.com/Charlie9830/castboard_player.git;protocol=https;rev=${CASTBOARD_PLAYER_REV};branch=master;destsuffix=git"

S = "${WORKDIR}/git"

TOOLCHAIN = "clang"

inherit cmake

do_patch() {
    export PATH=${STAGING_DIR_NATIVE}/usr/share/flutter-elinux/sdk/bin:$PATH
    export PUB_CACHE=${STAGING_DIR_NATIVE}/usr/share/flutter-elinux/sdk/.pub-cache
}

do_patch[depends] += "flutter-elinux-sdk-native:do_populate_sysroot"

do_configure() {
    #
    # Castboard Core
    #
    install -d ${WORKDIR}/castboard_core/
    cp -r ${STAGING_DATADIR}/castboard-core/* ${WORKDIR}/castboard_core/

    #
    # Castboard Remote
    #
    # We only create the web_app folder now. We don't populate it yet. This is because the castboard-player pubspec.yaml only includes a listing for the web_app folder. Not all of the specific items inside it.
    # If we include it now. When the app is compiled, it will ignore the directories inside the web_app folder because it doesn't know about them. Therefore we create a blank web_app directory now, so that 
    # flutter won't complain that it can't find the asset. Then we will populate it in the install step.
    install -d ${S}/assets/web_app/
}

do_compile() {
    export PATH=${STAGING_DIR_NATIVE}/usr/share/flutter-elinux/sdk/bin:$PATH

    cd ${S}
    
    flutter-elinux create .
    flutter-elinux pub get
    
    # DRM Build
    flutter-elinux build elinux --target-arch=arm64 --target-sysroot=${STAGING_DIR_TARGET} --target-backend-type=gbm
    
    # Wayland Client Build
    #flutter-elinux build elinux --target-arch=arm64 --target-sysroot=${STAGING_DIR_TARGET}
}

do_install() {
    #
    # Flutter-elinux Layout
    #
    install -d ${D}${datadir}/${PN}/
    cp -rTv ${S}/build/elinux/arm64/release/bundle/. ${D}${datadir}/${PN}/

    # Rename exectuable to something sane because flutter-elinux uses the parent dir name, which in the case of this
    # recipe is "git"
    mv ${D}${datadir}/${PN}/git ${D}${datadir}/${PN}/player

    # Install the web_app Assets.
    cp -r ${STAGING_DATADIR}/castboard-remote/web/* ${D}${datadir}/${PN}/data/flutter_assets/assets/web_app/
}

FILES_${PN} = "${datadir}/${PN}/*"

do_package_qa[noexec] = "1"
```

HidenoriMatsubayashi, 2021-09-15:
@Charlie9830 

Thank you for you comment. It's just a guess, but the command you are running may be wrong. For the DRM (GBM) backend, you need to run the command by **root** user. Proberbully, [2. Run with DRM backend](https://github.com/sony/flutter-embedded-linux/wiki/How-to-run-Flutter-apps#2-run-with-drm-backend) helps you. In addition, the drm device of RPI4 is `/dev/dri/card1`.

```
$ sudo FLUTTER_DRM_DEVICE="/dev/dri/card1" <path_to_binary>
```

Charlie9830, 2021-09-15:
@HidenoriMatsubayashi 

Thank you very much for your quick reply. You were correct. Whilst migrating over to flutter-elinux I had completely forgotten about setting ```FLUTTER_DRM_DEVICE``` and having to run with input privileges .

Closing this now.

closed 2021-09-15 by Charlie9830