sony/flutter-elinux #65: Raspberry Pi 4 8GB - wayland - Xdg-shell is invalid

piotrwest, 2021-11-02:
Hi, I'm trying to run flutter-elinux on RPi 4 8GB. I've managed to successfully use gbm backend. Now I'm trying out wayland, as gbm backend has quite long (around 4 seconds - in release mode) startup time for a simple hello-world application.

Here is what I'm trying and the error:

1. Install and launch Weston with *weston-launch* command
2. In the terminal in launched Weston desktop:
3. Build flutter app: `flutter-elinux build elinux --debug`:
```
💪 Building with sound null safety 💪

Building an eLinux application with wayland backend in debug mode for arm64 target...       103.5s
```
4. Run the app:
```
cd build/elinux/arm64/debug/bundle
./testapp

[ERROR][CreateRenderSurface(963)] Xdg-shell is invalid
Failed to create view controller.
```

Some debugging info:
```
ldd testapp
        linux-vdso.so.1 (0x0000007f9e56e000)
        libflutter_engine.so => /home/pi/testapp/build/elinux/arm64/debug/bundle/./lib/libflutter_engine.so (0x0000007f9a580000)
        libflutter_elinux_wayland.so => /home/pi/testapp/build/elinux/arm64/debug/bundle/./lib/libflutter_elinux_wayland.so (0x0000007f9a4ca000)
        libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000007f9a33f000)
        libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007f9a282000)
        libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007f9a25e000)
        libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f9a0ec000)
        libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000007f9a0d8000)
        libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000007f9a0a9000)
        /lib/ld-linux-aarch64.so.1 (0x0000007f9e540000)
        libxkbcommon.so.0 => /lib/aarch64-linux-gnu/libxkbcommon.so.0 (0x0000007f9a05a000)
        libwayland-client.so.0 => /lib/aarch64-linux-gnu/libwayland-client.so.0 (0x0000007f9a03b000)
        libwayland-cursor.so.0 => /lib/aarch64-linux-gnu/libwayland-cursor.so.0 (0x0000007f9a023000)
        libwayland-egl.so.1 => /lib/aarch64-linux-gnu/libwayland-egl.so.1 (0x0000007f9a011000)
        libEGL.so.1 => /lib/aarch64-linux-gnu/libEGL.so.1 (0x0000007f99feb000)
        libffi.so.6 => /lib/aarch64-linux-gnu/libffi.so.6 (0x0000007f99fd3000)
        librt.so.1 => /lib/aarch64-linux-gnu/librt.so.1 (0x0000007f99fbb000)
        libGLdispatch.so.0 => /lib/aarch64-linux-gnu/libGLdispatch.so.0 (0x0000007f99e30000)
```

This is happening on 2021-05-07-raspios-buster-arm64-lite.img.

Could you help me please debug this?

HidenoriMatsubayashi, 2021-11-02:
What version of weston are you using? Older weston doesn't support XDG-protocol.

```
$ weston --version
```

piotrwest, 2021-11-02:
It turns out I had weston version 5.0.0. After upgrading to 9.0.0 this works without issues. Thank you again for your help.

closed 2021-11-02 by piotrwest