sony/flutter-elinux #66: Flutter default sample application hangs after a few seconds when running on i.MX8X ARM64 target.

mortela4, 2021-11-16:
The attached Dockerfile is used to native-build (via 'docker buildx') the 'default' Flutter sample application for test on a ARM64 target (based on i.MX8QXP SoC). The target runs XWayland using Weston graphical shell. "build"-folder is copied to target as "/usr/local/flutter_demo", and the executable is started as shown below.

```
root@ccimx8x-sbc-pro:/usr/local/flutter_demo/elinux/arm64/debug/bundle# **./flutter_elinux_testapp &**
[1] 969
root@ccimx8x-sbc-pro:/usr/local/flutter_demo/elinux/arm64/debug/bundle# flutter: **Observatory listening on http://127.0.0.1:39523/JChjgsj2Mz0=/**

```

However, the app turns unresponsive after a few seconds - typically I only get to increment the counter in the UI a few times before the app hangs. The line "Observatory listening ..." is all the application emits when built with "--debug" flag. If built with "--release" it emits **nothing** on the console, and hangs almost immediately after UI is shown on attached display. 

Launching the app with "-d" flag shows the UI with windowmanager decorations/controls, but none of those function once the application hangs. In short; the application is **functioning correctly for a few seconds**, then hang.

Note that building and running on x86-host (Ubuntu20.4) works fine; both with 'elinux-wayland' and 'elinux-x11' backends. See attached screendump. 

Target root filesystem is built using Yocto 3.2 (Gatesgarth) with minor patches for CCiMX8X evaluation kit from DIGI Inc (https://github.com/digi-embedded/meta-digi), so basically NXP's official Yocto distribution (on top of 'fslc-multimedia-image' recipe). 
Launching the Flutter app from serial console or from GUI (e.g. weston-terminal) makes no difference.

mortela4, 2021-11-16:
Attachments did not get attached, so here is Dockerfile (built using 'docker buildx --load'):

```
FROM debian:bullseye-slim

WORKDIR /root

# Create a minimal runtime environment for executing 'flutter-elinux' build
# with the smallest possible image size.

RUN set -eux ; \
     apt-get update ; \
     apt-get upgrade ; \
     apt-get install -y --no-install-recommends \
        ca-certificates \
        curl \
        openssh-client \
        unzip \
        dnsutils \
        git \
        nano \
        clang \
        cmake \
        ninja-build \
        pkg-config \
        libgtk-3-dev \
        build-essential \
        pkg-config \
        libegl1-mesa-dev \
        libxkbcommon-dev \
        libgles2-mesa-dev \
        libwayland-dev \
        wayland-protocols \
        weston \
        libdrm-dev \
        libgbm-dev \
        libinput-dev \
        libudev-dev \
        libsystemd-dev \
        libx11-dev \
    ; \
    rm -rf /var/lib/apt/lists/* ; \
    git clone https://github.com/sony/flutter-elinux.git ; \
    mv flutter-elinux /opt/ ; \
    /opt/flutter-elinux/bin/flutter-elinux doctor ; \
    /opt/flutter-elinux/bin/flutter-elinux devices ; \
    mkdir flutter_elinux_test ; \
    cd flutter_elinux_test ; \
    /opt/flutter-elinux/bin/flutter-elinux create sample ; \
    cd sample ; \
    /opt/flutter-elinux/bin/flutter-elinux build elinux --debug ;

```

And screendump when running on host:
[image]

HidenoriMatsubayashi, 2021-11-17:
This issue happens on Weston (version 9 and above). See also: https://github.com/sony/flutter-embedded-linux/issues/234

Could you use [version 2.5.0](https://github.com/sony/flutter-elinux/releases/tag/2.5.0) instead of the latest version? Or, you can replace `libflutter_elinux_wayland.so` to a new one that you build (See: https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps#how-to-build-libflutter_engineso-and-libflutter_elinux_so)

mortela4, 2021-11-23:
Confirmed to work with 2.5.0: I did a 'git checkout tags/2.5.0' in working copy of 'flutter-elinux', then rebuilt sample application (using Dockerfile) - and it worked fine on my i.MX8X target platform! 
Thanks for helping out :-)
PS: we would of course like to use a most current Flutter-version as possible. Hope this issue will be fixed in near future!

HidenoriMatsubayashi, 2021-11-23:
> we would of course like to use a most current Flutter-version as possible. Hope this issue will be fixed in near future!

Yes, this issue will be fixed in the next stable version (2.6.0, 2.7.0, 2.8.0, and so on).

HidenoriMatsubayashi, 2021-12-10:
Fixed https://github.com/sony/flutter-elinux/pull/68

closed 2021-12-10 by HidenoriMatsubayashi