sony/flutter-elinux #67: Ubuntu Core Missing Fonts

johncharris, 2021-11-18:
Environment
Raspberry Pi 4
Ubuntu or Ubuntu Core

I'm trying to deploy the example application at https://github.com/johncharris/libplctag_dart on Ubuntu Core with Ubuntu-Frame.

I have the snap being built and the app deployed, but there is no text. The titlebar and buttons are empty. I tried adding the google_fonts package and embedding a font but that doesn't work either. The fonts work fine when I run it on a Ubuntu desktop. I'm guessing this means there is a missing dependency, but I have no idea what that might be.

HidenoriMatsubayashi, 2021-11-18:
There are two ways, which are installing fonts to your target host and [using a custom font](https://docs.flutter.dev/cookbook/design/fonts).

johncharris, 2021-11-18:
Thank you, that did indeed fix it. I'm guessing google fonts didn't work from me packaging it wrong.

closed 2021-11-18 by HidenoriMatsubayashi