sony/flutter-elinux #70: Failure at runtime from version 2.8.0 onwards.

Charlie9830, 2021-12-28:
**Specs**
Device: Rpi4
Build System: Yocto Dunfell
_Note: I don't directly build the flutter-engine with Yocto myself, currently I build the flutter-elinux tools as native and use them to build my application within Yocto._
Runtime Mode: Wayland client with Cage as the compositor. 

**Problem**
Since upgrading to version 2.8.0 and later. My application fails to start on device. Cage boots up successfully but then the flutter application itself seems to be immediately exiting. As far as I can tell it does not reach the ```main()``` function of the dart code. Reverting to 2.5.3 solves the issue.

Unfortunately, as of yet I haven't been able to extract an error message, it may be that cage is eating the error messages emitted from the app.

I am going to rebuild with Weston as the compositor and see if the same issue occurs.

Charlie9830, 2021-12-28:
So rebuilding and using Weston as the compositor thankfully gives me an error message emitted from the embedder.

> [FATAL:flutter/runtime/dart_vm_initializer.cc(22)] Error while initializing the Dart VM: Snapshot not compatiable with the current VM configuration: the snapshot requires 'produce no-code_comments no-dwarf_stack_traces_mode lazy_async_stacks no-lazy_dispatchers use_bare_instructions dedup_instructions no-asserts arm64-sysv compressed-pointers null-safety' but the VM has 'produce no-code_comments no-dwarf_stack_traces_mode lazy_async_stacks no-lazy_dispatchers use_base_instructions dedup_instructions no-asserts arm64-sysv no-compressed-pointers null-safety'
Aborted

The key difference in the configurations listed above is the ```compressed-pointers``` property.

HidenoriMatsubayashi, 2022-01-05:
@Charlie9830 
(@kono0514)

Thank you for reporting the issue. I confirmed that it will occur in profile/release mode.

HidenoriMatsubayashi, 2022-01-05:
We've been using `gen_snapshot` of Android. However, some changes for Android only added in flutter 2.8.0 (https://github.com/flutter/engine/pull/27975). Now, we need to prepare `gen_snapshot` for elinux.

Taha-Firoz, 2022-01-05:
Just ran into the same issue when building for `gbm`, updated my flutter version to 2.8.1 and builds stopped running

Taha-Firoz, 2022-01-05:
The bug is present for both debug and release mode binaries

topolittle, 2022-01-10:
As a workaround, I tried to enable the compressed pointers feature in the flutter-engine but I get a segmentation fault error when I try to run it on the target.

HidenoriMatsubayashi, 2022-01-11:
>  tried to enable the compressed pointers feature in the flutter-engine

Currently, it seems to be supported in Android only.

Taha-Firoz, 2022-02-13:
@HidenoriMatsubayashi any update on this? I don't understand the snapshot workaround

HidenoriMatsubayashi, 2022-02-14:
I'm working on this. This issue will be fixed with next release version (2.10.1)

closed 2022-02-21 by HidenoriMatsubayashi