sony/flutter-elinux #71: Issue running sample app in release or profile mode

kono0514, 2021-12-28:
I'm getting this error when I run it in release or profile mode. Debug mode works fine.
Tried both elinux-x11 and elinux-wayland.
What am I missing here?

> [FATAL:flutter/runtime/dart_vm_initializer.cc(22)] Error while initializing the Dart VM: Snapshot not compatible with the current VM configuration: the snapshot requires 'product no-code_comments no-dwarf_stack_traces_mode lazy_async_stacks no-lazy_dispatchers use_bare_instructions dedup_instructions no-asserts x64-sysv compressed-pointers null-safety' but the VM has 'product no-code_comments no-dwarf_stack_traces_mode lazy_async_stacks no-lazy_dispatchers use_bare_instructions dedup_instructions no-asserts x64-sysv no-compressed-pointers null-safety'

Reproduce steps:
```bash
# Already installed dependencies
git clone https://github.com/sony/flutter-elinux.git
sudo mv flutter-elinux /opt/
export PATH=$PATH:/opt/flutter-elinux/bin

flutter-elinux doctor
flutter-elinux create sample
cd sample
flutter-elinux run -d elinux-wayland --release
```

Error:
```bash
$ flutter-elinux run -d elinux-x11 --release
Launching lib/main.dart on eLinux in release mode...
Building an eLinux application with x11 backend in release mode for x64 target...       134.1s

Flutter run key commands.
h List all available interactive commands.
c Clear the screen
q Quit (terminate the application on the device).
[FATAL:flutter/runtime/dart_vm_initializer.cc(22)] Error while initializing the Dart VM: Snapshot not compatible with the current VM configuration: the snapshot requires 'product no-code_comments no-dwarf_stack_traces_mode lazy_async_stacks no-lazy_dispatchers use_bare_instructions dedup_instructions no-asserts x64-sysv compressed-pointers null-safety' but the VM has 'product no-code_comments no-dwarf_stack_traces_mode lazy_async_stacks no-lazy_dispatchers use_bare_instructions dedup_instructions no-asserts x64-sysv no-compressed-pointers null-safety'
```

flutter-elinux doctor:
```bash
$ flutter-elinux doctor
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel unknown, 2.8.1, on Ubuntu 21.10 5.10.43-yocto-standard, locale en_US)
[✓] eLinux toolchain - develop for embedded Linux devices
[✗] Android toolchain - develop for Android devices
    ✗ Unable to locate Android SDK.
      Install Android Studio from: https://developer.android.com/studio/index.html
      On first launch it will assist you in installing the Android SDK components.
      (or visit https://flutter.dev/docs/get-started/install/linux#android-setup for detailed instructions).
      If the Android SDK has been installed to a custom location, please use
      `flutter config --android-sdk` to update to that location.

[✗] Chrome - develop for the web (Cannot find Chrome executable at google-chrome)
    ! Cannot find Chrome. Try setting CHROME_EXECUTABLE to a Chrome executable.
[!] Android Studio (not installed)
[✓] Connected device (2 available)

! Doctor found issues in 3 categories.
```

kono0514, 2021-12-28:
Realized there's already an issue about this: #70
But hopefully this adds a little more context

HidenoriMatsubayashi, 2022-01-05:
@kono0514 Thank you for reporting the issue. I confirmed this issue.

HidenoriMatsubayashi, 2022-01-05:
I'll close this issue because it's duplicated.

closed 2022-01-05 by HidenoriMatsubayashi