sony/flutter-elinux #74: [Question] Passing flutter-embedded-linux flags

psstoyanov, 2022-01-13:
### Overview

[flutter-embedded-linux](https://github.com/sony/flutter-embedded-linux/) has [several flags](https://github.com/sony/flutter-embedded-linux/wiki/How-to-run-Flutter-apps#command-options) to control certain aspects of the behaviour.

They can't be used with `flutter-elinux` and at least some are overlapping (example with `-d` used for targeting the device and enabling decorations in flutter-elinux and flutter-embedded-linux respectively).

### Question
Is it within the scope of this project to enable the use of the additional flags from `flutter-embedded-linux`?

psstoyanov, 2022-01-14:
Great!

I will try to familiarize myself with the code to see how this can be achieved in the next week or so.
Probably I will also check with the [Flutter Study Group](https://flutterstudygroup.com/) for ideas on how to do it?

psstoyanov, 2022-01-25:
A question for the generation of the tool:

- Which flutter tools are we looking for on this line? https://github.com/sony/flutter-elinux/blob/490317ad3eee0745027d1675986e1d5f6e727e22/pubspec.yaml#L12
  - Should those be the ones provided from `depot_tools` ? 
  - Could I use the installed flutter version in `/opt/flutter`

HidenoriMatsubayashi, 2022-01-27:
> Which flutter tools are we looking for on this line?

https://github.com/flutter/flutter/tree/master/packages/flutter_tools

> Should those be the ones provided from depot_tools ?

What does it mean?

> Could I use the installed flutter version in /opt/flutter

No. `flutter-elinux` automatically downloads the corresponding `flutter_tools` version. See: https://github.com/sony/flutter-elinux/blob/main/bin/internal/shared.sh

psstoyanov, 2022-01-27:
Thank you, @HidenoriMatsubayashi ! 
That was silly of me, I was trying to do `dart pub get` instead of letting the tool do it's job...

psstoyanov, 2023-05-28:
We can add new arguments for building:
- https://github.com/sony/flutter-elinux/blob/2b8096b0983c4312673b96aa921bb42d1badb9e1/lib/commands/build.dart#L38

<details>
<summary> Example build flags </summary>

```
  argParser.addFlag(
      'onscreen-keyboard',
      help: 'Enables onscreen-keyboard support.',
    );
    argParser.addFlag(
      'fullscreen',
      help: 'Sets the app to fullscreen.',
    );
```
</details>

and running:
- https://github.com/sony/flutter-elinux/blob/2b8096b0983c4312673b96aa921bb42d1badb9e1/lib/commands/run.dart#L14

<details>
<summary> Example run flags </summary>

```
  ELinuxRunCommand({super.verboseHelp}) {
    argParser.addFlag(
      'onscreen-keyboard',
      help: 'Enables onscreen-keyboard support.',
    );
    argParser.addFlag(
      'fullscreen',
      help: 'Sets the app to fullscreen.',
    );
  }
```
</details>

Now to find how to pass them further down the stack.

closed 2023-08-14 by HidenoriMatsubayashi