sony/flutter-elinux #87: Flutter embedder build with rotation option

Taha-Firoz, 2022-04-06:
I'm on the latest version of flutter-elinux 2.10.3 but it doesn't have the rotation option in the client yet. I'm using a display who's primary axis is portrait and flutters own orientation system doesn't work. Any idea when the rotation option on the client will be available, I see you've already added it to the [flutter-embedded-linux](https://github.com/sony/flutter-embedded-linux/commit/7e390a500dc4be92d52c7d13d5b061ecce0422ca) repo

HidenoriMatsubayashi, 2022-04-06:
This issue is related to https://github.com/sony/flutter-elinux/issues/74. We need to add the command option to handle it. Could you check it?

closed 2022-04-06 by HidenoriMatsubayashi

Taha-Firoz, 2022-04-07:
No no it's my bad I had to delete the previous elinux folder and run flutter-elinux create again