sony/flutter-elinux #89: UI integration tests not run on a remote device

GuillaumeMorinQc, 2022-04-07:
I'm trying to use flutter integration tests on a remote arm64 device. When I launched the following command the application is deployed and launched, but when it time to run tests nothing happens for 12 minutes. 
The application is the default app with counter and button to increment it. 

```
flutter-arm test integration_test/app_test.dart  -vv --disable-service-auth-codes -d imx8mm
[  +48 ms] executing: uname -m
[  +21 ms] Exit code 0 from: uname -m
[        ] x86_64
[   +9 ms] executing: [/home/dev/sdk/flutter-elinux/flutter/] git -c log.showSignature=false log -n 1 --pretty=format:%H
[   +5 ms] Exit code 0 from: git -c log.showSignature=false log -n 1 --pretty=format:%H
[        ] 097d3313d8e2c7f901932d63e537c1acefb87800
[        ] executing: [/home/dev/sdk/flutter-elinux/flutter/] git tag --points-at 097d3313d8e2c7f901932d63e537c1acefb87800
[  +12 ms] Exit code 0 from: git tag --points-at 097d3313d8e2c7f901932d63e537c1acefb87800
[        ] 2.10.2
[  +35 ms] executing: [/home/dev/sdk/flutter-elinux/flutter/] git rev-parse --abbrev-ref HEAD
[   +2 ms] Exit code 0 from: git rev-parse --abbrev-ref HEAD
[        ] HEAD
[        ] executing: [/home/dev/sdk/flutter-elinux/flutter/] git rev-parse --abbrev-ref --symbolic @{u}
[   +3 ms] Exit code 128 from: git rev-parse --abbrev-ref --symbolic @{u}
[        ] fatal: HEAD does not point to a branch
[  +20 ms] Unable to locate an Android SDK.
[ +129 ms] Found plugin integration_test at /home/dev/sdk/flutter-elinux/flutter/packages/integration_test/
[  +44 ms] Found plugin integration_test at /home/dev/sdk/flutter-elinux/flutter/packages/integration_test/
[  +28 ms] Found plugin integration_test at /home/dev/sdk/flutter-elinux/flutter/packages/integration_test/
[  +57 ms] Found plugin integration_test at /home/dev/sdk/flutter-elinux/flutter/packages/integration_test/
[   +6 ms] Found 1 files which will be executed as Integration Tests.
[   +4 ms] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
[   +2 ms] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'WindowsUwpEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
[  +14 ms] Artifact Instance of 'ELinuxEngineArtifacts' is not required, skipping update.
[  +19 ms] executing: ping -w 500 -c 1 192.168.2.227
[  +28 ms] PING 192.168.2.227 (192.168.2.227) 56(84) bytes of data.
           64 bytes from 192.168.2.227: icmp_seq=1 ttl=62 time=0.997 ms

           --- 192.168.2.227 ping statistics ---
           1 packets transmitted, 1 received, 0% packet loss, time 0ms
           rtt min/avg/max/mdev = 0.997/0.997/0.997/0.000 ms
[   +4 ms] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
[        ] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'WindowsUwpEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
[   +9 ms] Artifact Instance of 'ELinuxEngineArtifacts' is not required, skipping update.
[  +73 ms] Skipping pub get: version match.
[  +30 ms] Found plugin integration_test at /home/dev/sdk/flutter-elinux/flutter/packages/integration_test/
[  +74 ms] Found plugin integration_test at /home/dev/sdk/flutter-elinux/flutter/packages/integration_test/
[  +21 ms] Generating /workspaces/flutter-elinux-imx8mm-devcontainer/hello_world/android/app/src/main/java/io/flutter/plugins/GeneratedPluginRegistrant.java
[ +147 ms] running test package with arguments: [-r, compact, --timeout, 30s, --concurrency=1, --chain-stack-traces, --, /workspaces/flutter-elinux-imx8mm-devcontainer/hello_world/integration_test/app_test.dart]
00:00 +0: loading /workspaces/flutter-elinux-imx8mm-devcontainer/hello_world/integration_test/app_test.dart                                                                                                                [ +146 ms] test 0: starting test /workspaces/flutter-elinux-imx8mm-devcontainer/hello_world/integration_test/app_test.dart
[   +8 ms] Stopping scan for flutter_test_config.dart; found project root at /workspaces/flutter-elinux-imx8mm-devcontainer/hello_world
[        ] test 0: starting test device
[   +2 ms] test 0: awaiting connection to test device
[  +17 ms] Uninstall hello_world from imx8mm.
[        ] executing: true
[... 73 lines cut ...]
[✗] Android toolchain - develop for Android devices
    ✗ Unable to locate Android SDK.
      Install Android Studio from: https://developer.android.com/studio/index.html
      On first launch it will assist you in installing the Android SDK components.
      (or visit https://flutter.dev/docs/get-started/install/linux#android-setup for detailed instructions).
      If the Android SDK has been installed to a custom location, please use
      `flutter config --android-sdk` to update to that location.

[✗] Chrome - develop for the web (Cannot find Chrome executable at google-chrome)
    ! Cannot find Chrome. Try setting CHROME_EXECUTABLE to a Chrome executable.

[!] Android Studio (not installed)
    • Android Studio not found; download from https://developer.android.com/studio/index.html
      (or visit https://flutter.dev/docs/get-started/install/linux#android-setup for detailed instructions).

[✓] Connected device (3 available)
    • eLinux (desktop) • elinux-wayland • flutter-tester • Debian GNU/Linux 11 (bullseye) 5.10.102.1-microsoft-standard-WSL2
    • eLinux (desktop) • elinux-x11     • flutter-tester • Debian GNU/Linux 11 (bullseye) 5.10.102.1-microsoft-standard-WSL2
    • eLinux (mobile)  • imx8mm         • flutter-tester • i.MX8 (Hydrogen-Tritium) 2.10.2 - 0.0.0.22052.2

[✓] HTTP Host Availability
    • All required HTTP hosts are available

! Doctor found issues in 3 categories.
```

fredrmag, 2022-04-28:
First of all, thanks for the great work in making flutter work on embedded! 

I have a similar issue as above when running integration tests on target device. The tests timeouts after 12 minutes.

I am following the [flutter integration test example](https://docs.flutter.dev/cookbook/testing/integration/introduction) for testing out integration testing using `flutter-elinux` on a target device.

<details>
<summary><code>flutter-elinux test integration_test/counter_test.dart -vv -d imx8mm</code></summary>
<br>

```
flutter-elinux test integration_test/counter_test.dart -vv -d imx8mm
[  +59 ms] executing: uname -m
[  +34 ms] Exit code 0 from: uname -m
[        ] x86_64
[  +19 ms] executing: [/opt/flutter-elinux/flutter/] git -c log.showSignature=false log -n 1 --pretty=format:%H
[  +11 ms] Exit code 0 from: git -c log.showSignature=false log -n 1 --pretty=format:%H
[        ] 7e9793dee1b85a243edd0e06cb1658e98b077561
[   +1 ms] executing: [/opt/flutter-elinux/flutter/] git tag --points-at 7e9793dee1b85a243edd0e06cb1658e98b077561
[  +18 ms] Exit code 0 from: git tag --points-at 7e9793dee1b85a243edd0e06cb1658e98b077561
[        ] 2.10.3
[  +37 ms] executing: [/opt/flutter-elinux/flutter/] git rev-parse --abbrev-ref HEAD
[   +6 ms] Exit code 0 from: git rev-parse --abbrev-ref HEAD
[        ] HEAD
[        ] executing: [/opt/flutter-elinux/flutter/] git rev-parse --abbrev-ref --symbolic @{u}
[  +12 ms] Exit code 128 from: git rev-parse --abbrev-ref --symbolic @{u}
[        ] fatal: HEAD does not point to a branch
[ +168 ms] Found plugin integration_test at /opt/flutter-elinux/flutter/packages/integration_test/
[  +68 ms] Found plugin integration_test at /opt/flutter-elinux/flutter/packages/integration_test/
[  +40 ms] Found plugin integration_test at /opt/flutter-elinux/flutter/packages/integration_test/
[  +56 ms] Found plugin integration_test at /opt/flutter-elinux/flutter/packages/integration_test/
[   +3 ms] Found 1 files which will be executed as Integration Tests.
[   +4 ms] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
[   +1 ms] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'WindowsUwpEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
[  +36 ms] Artifact Instance of 'ELinuxEngineArtifacts' is not required, skipping update.
[  +23 ms] executing: /home/fm/Android/Sdk/platform-tools/adb devices -l
[  +38 ms] executing: ping -w 500 -c 1 192.168.12.154
[  +55 ms] List of devices attached
[   +2 ms] PING 192.168.12.154 (192.168.12.154) 56(84) bytes of data.
           64 bytes from 192.168.12.154: icmp_seq=1 ttl=63 time=2.35 ms

           --- 192.168.12.154 ping statistics ---
           1 packets transmitted, 1 received, 0% packet loss, time 0ms
           rtt min/avg/max/mdev = 2.345/2.345/2.345/0.000 ms
[   +4 ms] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
[        ] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'WindowsUwpEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
[  +13 ms] Artifact Instance of 'ELinuxEngineArtifacts' is not required, skipping update.
[  +57 ms] Skipping pub get: version match.
[  +27 ms] Found plugin integration_test at /opt/flutter-elinux/flutter/packages/integration_test/
[  +64 ms] Found plugin integration_test at /opt/flutter-elinux/flutter/packages/integration_test/
[... 364 lines cut ...]
        "enabled": true,
        "platform": "arm64",
        "backend": "wayland",
        "ping": [
          "ping", "-w", "500", "-c", "1", "192.168.12.154"
        ],
        "pingSuccessRegex": "ttl=",
        "install": [
          "scp", "-r", "${localPath}", "root@192.168.12.154:/tmp/${appName}"
        ],
        "uninstall": [
          "ssh", "root@192.168.12.154", "rm -rf \"/tmp/${appName}\""
        ],
        "runDebug": [
          "ssh", "root@192.168.12.154", "export XDG_RUNTIME_DIR=/run/user/0; export WAYLAND_DISPLAY=wayland-0;/tmp/${appName}/${appName}"
        ],
        "forwardPort": [
          "ssh", "-o", "ExitOnForwardFailure=yes",
          "-L", "127.0.0.1:${hostPort}:127.0.0.1:${devicePort}", "root@192.168.12.154"
        ],
        "forwardPortSuccessRegex": ""
      }
    ]
  }
```

HidenoriMatsubayashi, 2023-08-15:
Sorry for no updates for a long time. Can you please try it using the latest flutter-elinux?

I was able to run integration test ([flutter integration test example](https://docs.flutter.dev/cookbook/testing/integration/introduction)) on Raspberry Pi4 (Raspberry Pi OS).

```
$ flutter-elinux test integration_test -d raspberry-pi4 --verbose
[  +20 ms] executing: [/home/hidenori/work/flutter/flutter-elinux/flutter/] git -c log.showSignature=false log -n 1 --pretty=format:%H
[  +12 ms] Exit code 0 from: git -c log.showSignature=false log -n 1 --pretty=format:%H
[        ] f468f3366c26a5092eb964a230ce7892fda8f2f8
[        ] executing: [/home/hidenori/work/flutter/flutter-elinux/flutter/] git tag --points-at f468f3366c26a5092eb964a230ce7892fda8f2f8
[  +18 ms] Exit code 0 from: git tag --points-at f468f3366c26a5092eb964a230ce7892fda8f2f8
[        ] 3.10.6
[   +2 ms] executing: [/home/hidenori/work/flutter/flutter-elinux/flutter/] git rev-parse --abbrev-ref HEAD
[   +3 ms] Exit code 0 from: git rev-parse --abbrev-ref HEAD
[        ] HEAD
[        ] executing: [/home/hidenori/work/flutter/flutter-elinux/flutter/] git rev-parse --abbrev-ref --symbolic @{upstream}
[   +2 ms] Exit code 128 from: git rev-parse --abbrev-ref --symbolic @{upstream}
[        ] fatal: HEAD does not point to a branch
[  +12 ms] Unable to locate an Android SDK.
[   +5 ms] executing: uname -m
[   +1 ms] Exit code 0 from: uname -m
[        ] aarch64
[  +69 ms] Found plugin integration_test at /home/hidenori/work/flutter/flutter-elinux/flutter/packages/integration_test/
[  +15 ms] Found plugin integration_test at /home/hidenori/work/flutter/flutter-elinux/flutter/packages/integration_test/
[   +8 ms] Found plugin integration_test at /home/hidenori/work/flutter/flutter-elinux/flutter/packages/integration_test/
[  +15 ms] Found plugin integration_test at /home/hidenori/work/flutter/flutter-elinux/flutter/packages/integration_test/
[   +2 ms] Found 1 files which will be executed as Integration Tests.
[   +1 ms] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
[        ] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
[   +4 ms] Artifact Instance of 'ELinuxEngineArtifacts' is not required, skipping update.
[   +5 ms] executing: ping -w 500 -c 1 192.168.10.22
[ +107 ms] PING 192.168.10.22 (192.168.10.22) 56(84) bytes of data.
                    64 bytes from 192.168.10.22: icmp_seq=1 ttl=128 time=96.2 ms

                    --- 192.168.10.22 ping statistics ---
                    1 packets transmitted, 1 received, 0% packet loss, time 0ms
                    rtt min/avg/max/mdev = 96.241/96.241/96.241/0.000 ms
[   +3 ms] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
[        ] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
[   +5 ms] Artifact Instance of 'ELinuxEngineArtifacts' is not required, skipping update.
[   +5 ms] Skipping pub get: version match.
[   +9 ms] Found plugin integration_test at /home/hidenori/work/flutter/flutter-elinux/flutter/packages/integration_test/
[  +15 ms] Found plugin integration_test at /home/hidenori/work/flutter/flutter-elinux/flutter/packages/integration_test/
[   +5 ms] Found plugin integration_test at /home/hidenori/work/flutter/flutter-elinux/flutter/packages/integration_test/
[  +23 ms] Generating /home/hidenori/work/flutter/sample/android/app/src/main/java/io/flutter/plugins/GeneratedPluginRegistrant.java
[  +58 ms] running test package with arguments: [--concurrency=1, --chain-stack-traces, --,
file:///home/hidenori/work/flutter/sample/integration_test/app_test.dart]
[... cut ...]
[+32736 ms] Permission denied, please try again.
[        ] Uninstallation Success
[        ] test 0: deleting temporary directory
[        ] test 0: finished
02:49 +1: All tests passed!                                                                                                                     
[   +4 ms] test package returned with exit code 0
[   +4 ms] "flutter test" took 170,067ms.
[  +24 ms] ensureAnalyticsSent: 0ms
[        ] Running 1 shutdown hook
[        ] Shutdown hooks complete
[        ] exiting with code 0
```

HidenoriMatsubayashi, 2023-08-15:
Added [Integration test on target device](https://github.com/sony/flutter-elinux/wiki/Remote-target-devices#integration-test-on-target-device) to Wiki. Closing. Thank you.

closed 2023-08-15 by HidenoriMatsubayashi