sony/flutter-elinux #94: Cross building for arm64 on x64 host issues

toygg96, 2022-04-26:
Hi Matsubayashi,

I followed the guide in https://github.com/sony/meta-flutter to cross build using Yocto sdk (Skipped Cross-building using bitbake section). Everything succeeded. Then I tried to run "flutter-elinux build elinux --target-arch=arm64 \
     --target-compiler-triple=aarch64-poky-linux --system-include-directories=/usr/aarch64-linux-gnu/include/c++/9/aarch64-linux-gnu -v" in https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps#case-2-use-yocto-sdk. 

But it gave me the following errors:
Failed to cmake build:
const auto __intptr = reinterpret_cast<uintptr_t>(__ptr);
1 error genereated.

Additional information: 
OS: Ubuntu 20.04.4 LTS
Followed the instructions in https://github.com/sony/flutter-embedded-linux/issues/4#issuecomment-1090157925 to disable the compiler test for C and C++, set the target architecture and swap the x86_64 and the ARM64 linker as the Yocto SDK is using the wrong linker.

Regards,
Toy

HidenoriMatsubayashi, 2022-04-26:
How about the below?

```Shell
$ source /opt/poky/3.1.15/environment-setup-aarch64-poky-linux
$ export CC=${CLANGCC}
$ export CXX=${CLANGCXX}
$ flutter-elinux build elinux --target-arch=arm64 --target-compiler-triple=aarch64-poky-linux
```

HidenoriMatsubayashi, 2022-04-27:
You used `--system-include-directories=/usr/aarch64-linux-gnu/include/c++/9/aarch64-linux-gnu -v` option, right?

toygg96, 2022-04-27:
Hi Matsubayashi,

Commands Run:
$ source /opt/poky/3.1.15/environment-setup-aarch64-poky-linux
$ export CC=${CLANGCC}
$ export CXX=${CLANGCXX}
$ flutter-elinux build elinux --target-arch=arm64 --target-compiler-triple=aarch64-poky-linux

Failed to cmake build:
Scanning dependencies of target flutter_assemble
[  0%] Built target flutter_assemble
Scanning dependencies of target flutter_wrapper_plugin
[  7%] Building CXX object
flutter/CMakeFiles/flutter_wrapper_plugin.dir/ephemeral/cpp_client_wrapper/core_
implementations.cc.o

In file included from
/opt/flutter-elinux/sample/elinux/flutter/ephemeral/cpp_client_wrapper/core_impl
ementations.cc:16:
//lib/gcc-cross/aarch64-linux-gnu/9/../../../../include/c++/9/cassert:43:10:
fatal error: 'bits/c++config.h' file not found
#include <bits/c++config.h>
         ^~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [flutter/CMakeFiles/flutter_wrapper_plugin.dir/build.make:63:
flutter/CMakeFiles/flutter_wrapper_plugin.dir/ephemeral/cpp_client_wrapper/core_
implementations.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:118:
flutter/CMakeFiles/flutter_wrapper_plugin.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

If I use the "--system-include-directories" option (https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps#troubleshooting), it gives a different error.

Commands Run:
$ source /opt/poky/3.1.15/environment-setup-aarch64-poky-linux
$ export CC=${CLANGCC}
$ export CXX=${CLANGCXX}
$ flutter-elinux build elinux --target-arch=arm64 \
--target-compiler-triple=aarch64-poky-linux  --system-include-directories=/usr/aarch64-linux-gnu/include/c++/9/aarch64-linux-gnu -v

 Failed to cmake build:
           Scanning dependencies of target flutter_assemble
           [  0%] Built target flutter_assemble
           Scanning dependencies of target flutter_wrapper_plugin
           [  7%] Building CXX object

flutter/CMakeFiles/flutter_wrapper_plugin.dir/ephemeral/cpp_client_wrapper/core_
           implementations.cc.o
           
           In file included from

/opt/flutter-elinux/sample/elinux/flutter/ephemeral/cpp_client_wrapper/core_impl
           ementations.cc:21:
           In file included from

/opt/flutter-elinux/sample/elinux/flutter/ephemeral/cpp_client_wrapper/include/f
           lutter/engine_method_result.h:8:

//lib/gcc-cross/aarch64-linux-gnu/9/../../../../include/c++/9/memory:121:25:
           error: cast from pointer to smaller type 'uintptr_t' (aka 'unsigned
int') loses
           information
             const auto __intptr = reinterpret_cast<uintptr_t>(__ptr);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
           1 error generated.
           make[2]: ***
[flutter/CMakeFiles/flutter_wrapper_plugin.dir/build.make:63:

flutter/CMakeFiles/flutter_wrapper_plugin.dir/ephemeral/cpp_client_wrapper/core_
           implementations.cc.o] Error 1
           make[1]: *** [CMakeFiles/Makefile2:118:
           flutter/CMakeFiles/flutter_wrapper_plugin.dir/all] Error 2
           make: *** [Makefile:130: all] Error 2
[   +5 ms] 
           #0      throwToolExit
(package:flutter_tools/src/base/common.dart:10:3)
           #1      NativeBundle.build
           (package:flutter_elinux/elinux_build_target.dart:387:7)
           <asynchronous suspension>
           #2      ELinuxBuilder.buildBundle
           (package:flutter_elinux/elinux_builder.dart:135:7)
           <asynchronous suspension>
           #3      BuildPackageCommand.runCommand
           (package:flutter_elinux/commands/build.dart:121:5)
           <asynchronous suspension>
           #4      FlutterCommand.run.<anonymous closure>
           (package:flutter_tools/src/runner/flutter_command.dart:1161:27)
           <asynchronous suspension>
           #5      AppContext.run.<anonymous closure>
           (package:flutter_tools/src/base/context.dart:150:19)
           <asynchronous suspension>
           #6      CommandRunner.runCommand
(package:args/command_runner.dart:209:13)
           <asynchronous suspension>
           #7      FlutterCommandRunner.runCommand.<anonymous closure>
           (package:flutter_tools/src/runner/flutter_command_runner.dart:281:9)
           <asynchronous suspension>
           #8      AppContext.run.<anonymous closure>
           (package:flutter_tools/src/base/context.dart:150:19)
           <asynchronous suspension>
           #9      FlutterCommandRunner.runCommand
           (package:flutter_tools/src/runner/flutter_command_runner.dart:229:5)
           <asynchronous suspension>
           #10     run.<anonymous closure>.<anonymous closure>
           (package:flutter_tools/runner.dart:62:9)
           <asynchronous suspension>
           #11     AppContext.run.<anonymous closure>
           (package:flutter_tools/src/base/context.dart:150:19)
           <asynchronous suspension>
           #12     main (package:flutter_elinux/executable.dart:83:3)
           <asynchronous suspension>

HidenoriMatsubayashi, 2022-04-27:
Could you use `--target-sysroot` option? I successed the build using its option.

```shell
$ sudo mv /opt/poky/3.1.16/sysroots/x86_64-pokysdk-linux/usr/bin/ld /opt/poky/3.1.16/sysroots/x86_64-pokysdk-linux/usr/bin/ld.x86_64
$ sudo cp -a /opt/poky/3.1.16/sysroots/x86_64-pokysdk-linux/usr/bin/aarch64-poky-linux/aarch64-poky-linux-ld /opt/poky/3.1.16/sysroots/x86_64-pokysdk-linux/usr/bin/ld
```

```shell
$ source /opt/poky/3.1.15/environment-setup-aarch64-poky-linux
$ export CC=${CLANGCC}
$ export CXX=${CLANGCXX}
$ flutter-elinux build elinux --target-arch=arm64 --target-compiler-triple=aarch64-poky-linux --target-sysroot=/opt/poky/3.1.16/sysroots/aarch64-poky-linux
```

HidenoriMatsubayashi, 2022-04-27:
Also, I updated [Case 2: Use Yocto SDK](https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps#case-2-use-yocto-sdk).

HidenoriMatsubayashi, 2022-04-28:
I'll close this issue, but please let me know if you have still any problems.

closed 2022-04-28 by HidenoriMatsubayashi

toygg96, 2022-04-28:
Hi Matsubayashi,

This command "flutter-elinux build elinux --target-arch=arm64 --target-compiler-triple=aarch64-poky-linux --target-sysroot=/opt/poky/3.1.16/sysroots/aarch64-poky-linux" works.

It is strange that I tried that command before but it didn't work with the exact same steps. Probably now it works because my Poky version is now 3.1.16?

Anyways, Thanks for your help :)

HidenoriMatsubayashi, 2022-04-28:
It sounds nice. You're welcome.