sony/flutter-elinux #98: How to use it on Jetson Nano?

fayfive, 2022-06-19:
I have followed the steps
```
$ flutter-elinux create sample
$ cd sample
$ flutter-elinux run -d elinux-wayland
```
But it just run on ubuntu x86
I want to run it on DRM mode of Jetson Nano. How to use it?
Can I cross compile the flutter app?
Thanks!

HidenoriMatsubayashi, 2022-06-20:
Please check the following wikis:
- https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps
- https://github.com/sony/flutter-embedded-linux/tree/master/examples/flutter-drm-eglstream-backend

fayfive, 2022-06-20:
Hello.
Can you tell me what's different of [flutter-elinux](https://github.com/sony/flutter-elinux) and [flutter-embedded-linux](https://github.com/sony/flutter-embedded-linux) ?

HidenoriMatsubayashi, 2022-06-20:
[flutter-elinux](https://github.com/sony/flutter-elinux) is a build and debug tool for elinux (embedded linux). Whereas [flutter-embedded-linux](https://github.com/sony/flutter-embedded-linux) is a porting layer of Flutter engine for elinux. Please check READMEs.

HidenoriMatsubayashi, 2022-06-22:
> I see the jetson sample #define DRM_DEVICE_NAME "drm-nvdc"

I don't think so as "/dev/dri/card0" is common, but I don't remember it.

fayfive, 2022-06-22:
> > I see the jetson sample #define DRM_DEVICE_NAME "drm-nvdc"
> 
> I don't think so as "/dev/dri/card0" is common, but I don't remember it.

Maybe you use jetpack 4.3. But now I am using jetpack4.6.
```
~$ ls /dev/dri/card0
ls: cannot access '/dev/dri/card0': No such file or directory
```
How can I fix it? Maybe it can't use udev for drm.

fayfive, 2022-07-05:
@HidenoriMatsubayashi 
any suggestions?

fayfive, 2022-08-30:
@HidenoriMatsubayashi
I have a test on jetson nano jp4.3. But I can't find the `/dev/dri/card0` too.
How can you do it on jetson nano?

HidenoriMatsubayashi, 2022-08-31:
I'll close the issue. See https://github.com/sony/flutter-elinux/issues/121

closed 2022-08-31 by HidenoriMatsubayashi