sony/flutter-elinux #108: CMake error while building app

Tom3652, 2022-08-09:
I am running the following command to build my basic app (generated by `flutter-elinux create .`) : 

```
flutter-elinux build elinux --target-arch=arm64 --target-sysroot=/opt/arm64-sysroot \
    --system-include-directories=/usr/aarch64-linux-gnu/include/c++/9/aarch64-linux-gnu \
    --debug
```

And i have the following output : 

```
tom@tom-VirtualBox:~/StudioProjects/test$ flutter-elinux build elinux --target-arch=arm64 --target-sysroot=/opt/arm64-sysroot     --system-include-directories=/usr/aarch64-linux-gnu/include/c++/9/aarch64-linux-gnu     --debug
Flutter 3.0.5 • channel unknown • unknown source
Framework • revision f1875d570e (4 weeks ago) • 2022-07-13 11:24:16 -0700
Engine • revision e85ea0e79c
Tools • Dart 2.17.6 • DevTools 2.12.2

💪 Building with sound null safety 💪

Building an eLinux application with wayland backend in debug mode for arm64 target...      1,365ms
Failed to cmake:
-- The CXX compiler identification is Clang 14.0.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - failed
-- Check for working CXX compiler: /usr/bin/clang++
-- Check for working CXX compiler: /usr/bin/clang++ - broken
-- Configuring incomplete, errors occurred!
See also "/home/tom/StudioProjects/test/build/elinux/arm64/debug/CMakeFiles/CMakeOutput.log".
See also "/home/tom/StudioProjects/test/build/elinux/arm64/debug/CMakeFiles/CMakeError.log".

CMake Error at /usr/share/cmake-3.22/Modules/CMakeTestCXXCompiler.cmake:62 (message):
  The C++ compiler

    "/usr/bin/clang++"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/tom/StudioProjects/test/build/elinux/arm64/debug/CMakeFiles/CMakeTmp

    Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_84996/fast && /usr/bin/gmake  -f
    CMakeFiles/cmTC_84996.dir/build.make CMakeFiles/cmTC_84996.dir/build
    gmake[1]: Entering directory '/home/tom/StudioProjects/test/build/elinux/arm64/debug/CMakeFiles/CMakeTmp'
    Building CXX object CMakeFiles/cmTC_84996.dir/testCXXCompiler.cxx.o
    /usr/bin/clang++ --target=aarch64-linux-gnu --sysroot=/opt/arm64-sysroot    -MD -MT
    CMakeFiles/cmTC_84996.dir/testCXXCompiler.cxx.o -MF CMakeFiles/cmTC_84996.dir/testCXXCompiler.cxx.o.d -o
    CMakeFiles/cmTC_84996.dir/testCXXCompiler.cxx.o -c
    /home/tom/StudioProjects/test/build/elinux/arm64/debug/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
    Linking CXX executable cmTC_84996
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_84996.dir/link.txt --verbose=1
    /usr/bin/clang++ --target=aarch64-linux-gnu --sysroot=/opt/arm64-sysroot
    CMakeFiles/cmTC_84996.dir/testCXXCompiler.cxx.o -o cmTC_84996
    /usr/bin/ld: unrecognised emulation mode: aarch64linux
    Supported emulations: elf_x86_64 elf32_x86_64 elf_i386 elf_iamcu elf_l1om elf_k1om i386pep i386pe
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    gmake[1]: *** [CMakeFiles/cmTC_84996.dir/build.make:100: cmTC_84996] Error 1
    gmake[1]: Leaving directory '/home/tom/StudioProjects/test/build/elinux/arm64/debug/CMakeFiles/CMakeTmp'
    gmake: *** [Makefile:127: cmTC_84996/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)
```

And i have `cat` the file `/home/tom/StudioProjects/test/build/elinux/arm64/debug/CMakeFiles/CMakeError.log` : 

```
tom@tom-VirtualBox:~/StudioProjects/test$ cat /home/tom/StudioProjects/test/build/elinux/arm64/debug/CMakeFiles/CMakeError.log
Detecting CXX compiler ABI info failed to compile with the following output:
[... 37 lines cut ...]

Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_84996/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_84996.dir/build.make CMakeFiles/cmTC_84996.dir/build
gmake[1]: Entering directory '/home/tom/StudioProjects/test/build/elinux/arm64/debug/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_84996.dir/testCXXCompiler.cxx.o
/usr/bin/clang++ --target=aarch64-linux-gnu --sysroot=/opt/arm64-sysroot    -MD -MT CMakeFiles/cmTC_84996.dir/testCXXCompiler.cxx.o -MF CMakeFiles/cmTC_84996.dir/testCXXCompiler.cxx.o.d -o CMakeFiles/cmTC_84996.dir/testCXXCompiler.cxx.o -c /home/tom/StudioProjects/test/build/elinux/arm64/debug/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
Linking CXX executable cmTC_84996
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_84996.dir/link.txt --verbose=1
/usr/bin/clang++ --target=aarch64-linux-gnu --sysroot=/opt/arm64-sysroot CMakeFiles/cmTC_84996.dir/testCXXCompiler.cxx.o -o cmTC_84996 
/usr/bin/ld: unrecognised emulation mode: aarch64linux
Supported emulations: elf_x86_64 elf32_x86_64 elf_i386 elf_iamcu elf_l1om elf_k1om i386pep i386pe
clang: error: linker command failed with exit code 1 (use -v to see invocation)
gmake[1]: *** [CMakeFiles/cmTC_84996.dir/build.make:100: cmTC_84996] Error 1
gmake[1]: Leaving directory '/home/tom/StudioProjects/test/build/elinux/arm64/debug/CMakeFiles/CMakeTmp'
gmake: *** [Makefile:127: cmTC_84996/fast] Error 2

```

I have installed all the mandatory dependencies : 

```
$ sudo apt install curl unzip git clang cmake pkg-config
$ sudo apt install libegl1-mesa libgles2-mesa libxkbcommon-dev
```

Thanks in advance for your help

Tom3652, 2022-08-10:
I have tried running the `flutter-elinux clean` and the usual `flutter clean` as well, but i am so new to linux i have no real idea of what may cause this or how to fix it :(

Tom3652, 2022-08-10:
Clearly there are some missing packages to be installed here or something... I have tried to install : `sudo apt install binutils-aarch64-linux-gnu` as well but i am still getting the same kind of error : 

[image]

HidenoriMatsubayashi, 2022-08-11:
I don't know your environment, but please check https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps#use-cases.

Tom3652, 2022-08-11:
Well i have a simple Ubuntu 22.04 LTS. 
I have tried the buildroot toolchain that is not working, i will try with Yocto today and then finally docker.

Thanks

Tom3652, 2022-08-11:
I have got the following error using the docker solution : 

```
tom@tom-VirtualBox:~/test_app$ flutter-elinux build elinux --target-arch=arm64 --target-sysroot=/home/tom/ubuntu18-arm64-sysroot --system-include-directories=/usr/aarch64-linux-gnu/include/c++/9/aarch64-linux-gnu --debug
Flutter 3.0.5 • channel unknown • unknown source
Framework • revision f1875d570e (4 weeks ago) • 2022-07-13 11:24:16 -0700
Engine • revision e85ea0e79c
Tools • Dart 2.17.6 • DevTools 2.12.2

💪 Building with sound null safety 💪

Building an eLinux application with wayland backend in debug mode for arm64 target...      1,658ms
Failed to cmake:
-- The CXX compiler identification is unknown
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - failed
-- Check for working CXX compiler: /home/tom/ubuntu18-arm64-sysroot/usr/bin/clang++
-- Check for working CXX compiler: /home/tom/ubuntu18-arm64-sysroot/usr/bin/clang++ - broken
-- Configuring incomplete, errors occurred!
See also "/home/tom/test_app/build/elinux/arm64/debug/CMakeFiles/CMakeOutput.log".
See also "/home/tom/test_app/build/elinux/arm64/debug/CMakeFiles/CMakeError.log".

CMake Error at /usr/share/cmake-3.22/Modules/CMakeTestCXXCompiler.cmake:62 (message):
  The C++ compiler

    "/home/tom/ubuntu18-arm64-sysroot/usr/bin/clang++"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/tom/test_app/build/elinux/arm64/debug/CMakeFiles/CMakeTmp

    Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_848af/fast && /usr/bin/gmake  -f
    CMakeFiles/cmTC_848af.dir/build.make CMakeFiles/cmTC_848af.dir/build
    gmake[1]: Entering directory '/home/tom/test_app/build/elinux/arm64/debug/CMakeFiles/CMakeTmp'
    Building CXX object CMakeFiles/cmTC_848af.dir/testCXXCompiler.cxx.o
    /home/tom/ubuntu18-arm64-sysroot/usr/bin/clang++    -o CMakeFiles/cmTC_848af.dir/testCXXCompiler.cxx.o -c
    /home/tom/test_app/build/elinux/arm64/debug/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
    aarch64-binfmt-P: Could not open '/lib/ld-linux-aarch64.so.1': No such file or directory
    gmake[1]: *** [CMakeFiles/cmTC_848af.dir/build.make:78: CMakeFiles/cmTC_848af.dir/testCXXCompiler.cxx.o] Error 255
    gmake[1]: Leaving directory '/home/tom/test_app/build/elinux/arm64/debug/CMakeFiles/CMakeTmp'
    gmake: *** [Makefile:127: cmTC_848af/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)
```

HidenoriMatsubayashi, 2022-08-11:
I think this issue was fixed. Can you please try again?

closed 2022-08-11 by HidenoriMatsubayashi

Tom3652, 2022-08-11:
Thanks for working on this.

I have the same error for now unfortunately..
Do i need to upgrade `flutter-elinux` somehow ?
I have ran `flutter-elinux upgrade`  and i see as output :

```
tom@tom-VirtualBox:~/StudioProjects/test_app$ flutter-elinux upgrade
Flutter 3.0.5 • channel unknown • unknown source
Framework • revision f1875d570e (4 weeks ago) • 2022-07-13 11:24:16 -0700
Engine • revision e85ea0e79c
Tools • Dart 2.17.6 • DevTools 2.12.2
flutter-elinux is already up to date
```

HidenoriMatsubayashi, 2022-08-11:
Please try the following steps.

1. cd <path to flutter-elinux>
2. git pull
3. rm bin/cache/*

Tom3652, 2022-08-12:
Thanks for the update. 

I have the following error now : 

```

tom@tom-VirtualBox:~/StudioProjects/test$ flutter-elinux build elinux --target-arch=arm64 --target-sysroot=/home/tom/ubuntu18-arm64-sysroot
Flutter 3.0.5 • channel unknown • unknown source
Framework • revision f1875d570e (4 weeks ago) • 2022-07-13 11:24:16 -0700
Engine • revision e85ea0e79c
Tools • Dart 2.17.6 • DevTools 2.12.2
Running pub upgrade...
Resolving dependencies...
  _fe_analyzer_shared 34.0.0 (46.0.0 available)
  analyzer 3.2.0 (4.6.0 available)
  archive 3.1.11 (3.3.1 available)
  args 2.3.0 (2.3.1 available)
  async 2.8.2 (2.9.0 available)
  boolean_selector 2.1.0
  browser_launcher 1.1.0 (1.1.1 available)
  build 2.3.0
  built_collection 5.1.1
  built_value 8.1.4 (8.4.0 available)
  characters 1.2.0 (1.2.1 available)
  charcode 1.3.1
  cli_util 0.3.5
  clock 1.1.0 (1.1.1 available)
  code_builder 4.2.0
  collection 1.16.0
  completion 1.0.0
  convert 3.0.1 (3.0.2 available)
  coverage 1.2.0 (1.5.0 available)
  crypto 3.0.1 (3.0.2 available)
  csslib 0.17.1 (0.17.2 available)
  dart_style 2.2.1 (2.2.3 available)
  dds 2.2.0 (2.2.6 available)
  dds_service_extensions 1.3.0 (1.3.1 available)
  devtools_shared 2.11.4 (2.16.0 available)
  dwds 12.1.1 (15.0.0 available)
  fake_async 1.3.0 (1.3.1 available)
  file 6.1.2
  fixnum 1.0.0 (1.0.1 available)
  flutter 0.0.0 from sdk flutter
  flutter_template_images 4.0.0 (4.2.0 available)
  flutter_test 0.0.0 from sdk flutter
  flutter_tools 0.0.0 from path flutter/packages/flutter_tools
  frontend_server_client 2.1.2 (2.1.3 available)
  glob 2.0.2 (2.1.0 available)
  html 0.15.0
  http 0.13.4 (0.13.5 available)
  http_multi_server 3.2.0 (3.2.1 available)
  http_parser 4.0.0 (4.0.1 available)
  intl 0.17.0
  io 1.0.3
  js 0.6.4
  json_rpc_2 3.0.1 (3.0.2 available)
  logging 1.0.2
  matcher 0.12.11 (0.12.12 available)
  material_color_utilities 0.1.4 (0.2.0 available)
  meta 1.7.0 (1.8.0 available)
  mime 1.0.1 (1.0.2 available)
  mockito 5.3.0
  multicast_dns 0.3.2 (0.3.2+1 available)
  mustache_template 2.0.0
  native_stack_traces 0.4.5 (0.5.0 available)
  node_preamble 2.0.1
  package_config 2.0.2 (2.1.0 available)
  path 1.8.1 (1.8.2 available)
  pedantic 1.11.1 (discontinued replaced by lints)
  petitparser 5.0.0
[... 66 lines cut ...]
    Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_80374/fast && /usr/bin/gmake  -f
    CMakeFiles/cmTC_80374.dir/build.make CMakeFiles/cmTC_80374.dir/build
    gmake[1]: Entering directory '/home/tom/StudioProjects/test/build/elinux/arm64/release/CMakeFiles/CMakeTmp'
    Building CXX object CMakeFiles/cmTC_80374.dir/testCXXCompiler.cxx.o
    //bin/clang++ --sysroot=/home/tom/ubuntu18-arm64-sysroot    -MD -MT
    CMakeFiles/cmTC_80374.dir/testCXXCompiler.cxx.o -MF CMakeFiles/cmTC_80374.dir/testCXXCompiler.cxx.o.d -o
    CMakeFiles/cmTC_80374.dir/testCXXCompiler.cxx.o -c
    /home/tom/StudioProjects/test/build/elinux/arm64/release/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
    Linking CXX executable cmTC_80374
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_80374.dir/link.txt --verbose=1
    //bin/clang++ --sysroot=/home/tom/ubuntu18-arm64-sysroot CMakeFiles/cmTC_80374.dir/testCXXCompiler.cxx.o -o
    cmTC_80374
    //bin/ld: cannot find Scrt1.o: No such file or directory
    //bin/ld: cannot find crti.o: No such file or directory
    //bin/ld: cannot find -lm: No such file or directory
    //bin/ld: cannot find libgcc_s.so.1: No such file or directory
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    gmake[1]: *** [CMakeFiles/cmTC_80374.dir/build.make:100: cmTC_80374] Error 1
    gmake[1]: Leaving directory '/home/tom/StudioProjects/test/build/elinux/arm64/release/CMakeFiles/CMakeTmp'
    gmake: *** [Makefile:127: cmTC_80374/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)
```

PR #109 (2022-08-11): Fix wrong compiler path issue when using external sysroots
Fixed #108. This is the side effect of https://github.com/sony/flutter-elinux/pull/106.