sony/flutter-elinux #112: how to build flutter-elinux binary

seanchann, 2022-08-12:
we need a document for the developer.  the current project is not a  normal flutter project.  using `flutter build linux`  does not work for me

HidenoriMatsubayashi, 2022-08-12:
Please check https://github.com/sony/flutter-elinux/wiki

closed 2022-08-12 by HidenoriMatsubayashi

seanchann, 2022-08-12:
sorry, I may have confused you. I modify the file in the dir `flutter-elinux/lib`. so need to rebuild flutter-elinux binary.  the wiki is just given a cp command

HidenoriMatsubayashi, 2022-08-12:
Sorry, I'm not sure what your question is.

seanchann, 2022-08-12:
just see 
[image]
add a `--verbose`  with  cmake --build. so need manual build flutter-elinux

HidenoriMatsubayashi, 2022-08-12:
Please see https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps#3-cross-building-from-x64-to-arm64

HidenoriMatsubayashi, 2022-08-12:
Please use the latest version and try `flutter-elinux build elinux`

seanchann, 2022-08-12:
in `flutter-elinux` project. (recompile binary of `flutter-elinux`)

call  `flutter-elinux-bak build elinux bin/flutter_elinux.dart` not work 

```
Flutter 3.0.5 • channel unknown • unknown source
Framework • revision f1875d570e (4 weeks ago) • 2022-07-13 11:24:16 -0700
Engine • revision e85ea0e79c
Tools • Dart 2.17.6 • DevTools 2.12.2

Building without sound null safety
For more information see https://dart.dev/null-safety/unsound-null-safety

This project is not configured for eLinux.
To fix this problem, create a new project by running `flutter-elinux create <app-dir>`.
```

seanchann, 2022-08-12:
I made a mistake, directly modifying lib/*.dart in flutter-elinux will take effect directly. flutter-elinux is a shell script in bin dir, no need to generate or compiler and so on

hoangtoi25, 2024-11-28:
I have same issue. How to resolve this issue @seanchann

hoangtoi25, 2024-11-28:
> Please use the latest version and try `flutter-elinux build elinux`

[image]