sony/flutter-elinux #114: Latest release doesn't build for arm64 target

Taha-Firoz, 2022-08-12:
I've pulled the latest build for flutter-elinux and builds that work with the previous version are failing with this one. It's a very minimal test that compiled successfully with 0 changes on the old release

compile command
```
$ flutter-elinux build elinux -v --target-backend-type=gbm  --target-arch=arm64 --target-sysroot=/home/me/sony-flutter/ubuntu18-arm64-sysroot --system-include-directories=/usr/aarch64-linux-gnu/include/c++/9/aarch64-linux-gnu
```

```
 Failed to cmake:
           -- The CXX compiler identification is Clang 10.0.0
           -- Check for working CXX compiler: //bin/clang++
           -- Check for working CXX compiler: //bin/clang++ -- broken
           -- Configuring incomplete, errors occurred!
           See also "/home/me/sony-flutter/elinuxTestUI/build/elinux/arm64/release/CMakeFiles/CMakeOutput.log".
           See also "/home/me/sony-flutter/elinuxTestUI/build/elinux/arm64/release/CMakeFiles/CMakeError.log".

           CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCXXCompiler.cmake:53 (message):
             The C++ compiler

               "//bin/clang++"

             is not able to compile a simple test program.

             It fails with the following output:

               Change Dir: /home/me/sony-flutter/elinuxTestUI/build/elinux/arm64/release/CMakeFiles/CMakeTmp
               
 Change Dir: /home/me/sony-flutter/elinuxTestUI/build/elinux/arm64/release/CMakeFiles/CMakeTmp

               Run Build Command(s):/usr/bin/make cmTC_cca2a/fast && /usr/bin/make -f CMakeFiles/cmTC_cca2a.dir/build.make CMakeFiles/cmTC_cca2a.dir/build
               make[1]: Entering directory '/home/me/sony-flutter/elinuxTestUI/build/elinux/arm64/release/CMakeFiles/CMakeTmp'
               Building CXX object CMakeFiles/cmTC_cca2a.dir/testCXXCompiler.cxx.o
               //bin/clang++ --sysroot=/home/me/sony-flutter/ubuntu18-arm64-sysroot     -o CMakeFiles/cmTC_cca2a.dir/testCXXCompiler.cxx.o -c
               /home/me/sony-flutter/elinuxTestUI/build/elinux/arm64/release/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
               Linking CXX executable cmTC_cca2a
               /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_cca2a.dir/link.txt --verbose=1
               //bin/clang++ --sysroot=/home/me/sony-flutter/ubuntu18-arm64-sysroot       CMakeFiles/cmTC_cca2a.dir/testCXXCompiler.cxx.o  -o cmTC_cca2a
               //bin/ld: cannot find crt1.o: No such file or directory
               //bin/ld: cannot find crti.o: No such file or directory
               //bin/ld: cannot find -lm
               //bin/ld: cannot find libgcc_s.so.1
               clang: error: linker command failed with exit code 1 (use -v to see invocation)
               make[1]: *** [CMakeFiles/cmTC_cca2a.dir/build.make:87: cmTC_cca2a] Error 1
               make[1]: Leaving directory '/home/me/sony-flutter/elinuxTestUI/build/elinux/arm64/release/CMakeFiles/CMakeTmp'
               make: *** [Makefile:121: cmTC_cca2a/fast] Error 2
```

HidenoriMatsubayashi, 2022-08-12:
Fixed that. Thanks.

closed 2022-08-12 by HidenoriMatsubayashi

Taha-Firoz, 2022-08-13:
Is there a discord, zulip, element, mailing list or GitHub discussion for this project to discuss possible features. I want to work on some but there's no place to talk stuff over.

PR #115 (2022-08-12): Fix wrong default value of target-toolchain option
Fixed #114
commit 4c1998ac (2022-08-12): Fix wrong default value of target-toolchain option (#115)

Fixed #114