sony/flutter-elinux #121: Failed to create view controller

fayfive, 2022-08-21:
I follow your steps to cross compile with docker.
When I run it on the board, I find that `Failed to create view controller` `FLUTTER_LOG_LEVELS=TRACE FLUTTER_DRM_DEVICE="/dev/dri/card1" /root/bundle/elinux  --bundle=/root/bundle`
My CPU is rk3568 and my build command `flutter-elinux build elinux --target-arch=arm64 --target-sysroot=~/ubuntu18-arm64-sysroot/ --target-backend-type=gbm`

HidenoriMatsubayashi, 2022-08-21:
You should paste your log.

fayfive, 2022-08-21:
I have `FLUTTER_LOG_LEVELS=TRACE `. But it just print `Failed to create view controller`.
or where is the log?
```
[root@RK356X:~/bundle]# FLUTTER_LOG_LEVELS=TRACE FLUTTER_DRM_DEVICE="/dev/dri/ca
rd1" /root/bundle/elinux  --bundle=/root/bundle
Failed to create view controller.
```

HidenoriMatsubayashi, 2022-08-21:
Please use debug builds.

```
flutter-elinux build elinux --target-arch=arm64 --target-sysroot=~/ubuntu18-arm64-sysroot/ --target-backend-type=gbm --debug
```

fayfive, 2022-08-21:
HA! I see ^_^
```
[root@RK356X:~/bundle]# FLUTTER_LOG_LEVELS=TRACE FLUTTER_DRM_DEVICE="/dev/dri/ca
rd1" /root/bundle/elinux  --bundle=/root/bundle
[ERROR][ConfigureDisplay(53)] Couldn't get resources
[ERROR][CreateRenderSurface(118)] Failed to create the native window
Failed to create view controller.
```

HidenoriMatsubayashi, 2022-08-21:
According to the log message, `drmModeGetResources` failed. I don't know your device, "/dev/dri/card1" is correct?

fayfive, 2022-08-21:
I have tried 
```
[root@RK356X:~/bundle]# FLUTTER_LOG_LEVELS=TRACE FLUTTER_DRM_DEVICE="/dev/dri/ca
rd0" /root/bundle/elinux  --bundle=/root/bundle
[INFO][ConfigureDisplay(71)] resolution: 1280x800
/root/bundle/elinux: symbol lookup error: /root/bundle/lib/libflutter_elinux_gbm.so: undefined symbol: drmIsMaste
```
But there is error of `drmIsMaste`

HidenoriMatsubayashi, 2022-08-21:
Can you please comment-out `drmIsMaster` in https://github.com/sony/flutter-embedded-linux/blob/master/src/flutter/shell/platform/linux_embedded/window/native_window_drm_gbm.cc#L29 and try again?

HidenoriMatsubayashi, 2022-08-21:
Or can you update your kernel version to more newest version?

fayfive, 2022-08-21:
> Or can you update your kernel version to more newest version?

`Linux RK356X 4.19.206`
Maybe I can't update. Because I don't have all the drivers of this official kernel.

I just want to see the result of elinux flutter. So I don't compile it.

I see it has no the api. Can I update the libdrm to fix it?

HidenoriMatsubayashi, 2022-08-21:
I think the first one is easier than building of the libdrm. Can you try it? After modifying the source file, you can build `libflutter_elinux_gbm.so` by the following steps in **arm64** docker.

```shell
$ cmake -DBUILD_ELINUX_SO=ON -DBACKEND_TYPE=DRM-GBM -DCMAKE_BUILD_TYPE=Release -DENABLE_ELINUX_EMBEDDER_LOG=ON -DFLUTTER_RELEASE=OFF ..
```

See also: https://github.com/sony/flutter-embedded-linux/wiki/Building-Embedded-Linux-embedding-for-Flutter#build-shared-library-for-flutter-elinux

fayfive, 2022-08-21:
```
flutter-embedded-linux/build$ cmake -DBUILD_ELINUX_SO=ON -DBACKEND_TYPE=DRM-GBM -DCMAKE_BUILD_TYPE=Release -DENABLE_ELINUX_EMBEDDER_LOG=ON -DFLUTTER_RELEASE=OFF ..
-- The CXX compiler identification is Clang 14.0.0
-- The C compiler identification is Clang 14.0.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'egl'
--   No package 'egl' found
CMake Error at /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:603 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:825 (_pkg_check_modules_internal)
  cmake/package.cmake:6 (pkg_check_modules)
  CMakeLists.txt:41 (include)

-- Configuring incomplete, errors occurred!
See also "/home/alfred/working/da/flutter-embedded-linux/build/CMakeFiles/CMakeOutput.log".
```
Do I need to set the sysroot?

HidenoriMatsubayashi, 2022-08-21:
See https://github.com/sony/flutter-embedded-linux/wiki/Installing-dependent-libraries

fayfive, 2022-08-21:
I need to cross compile, it seem to have no `sysroot` option?

HidenoriMatsubayashi, 2022-08-21:
> After modifying the source file, you can build libflutter_elinux_gbm.so by the following steps in arm64 docker,

Simply, you can build .so file for arm64 on arm64 docker.

fayfive, 2022-08-21:
That means I must put the source to the docker, and compile it?

HidenoriMatsubayashi, 2022-08-21:
Yes, that's right. Because it's easy to build than cross-buildings.

fayfive, 2022-08-21:
HA! It's too slow.
```
[ 94%] Building CXX object CMakeFiles/flutter_elinux_gbm.dir/src/flutter/shell/platform/common/client_wrapper/standard_codec.cc.o
make[2]: *** No rule to make target 'libflutter_engine.so', needed by 'libflutter_elinux_gbm.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 97%] Building CXX object CMakeFiles/flutter_elinux_gbm.dir/src/flutter/shell/platform/common/client_wrapper/plugin_registrar.cc.o
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/flutter_elinux_gbm.dir/all' failed
make[1]: *** [CMakeFiles/flutter_elinux_gbm.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```
But it's failed

HidenoriMatsubayashi, 2022-08-21:
Please copy `libflutter_engine.so` (aarch64 version) to current directory in the docker.

fayfive, 2022-08-21:
```
[root@RK356X:~/bundle]# FLUTTER_LOG_LEVELS=TRACE FLUTTER_DRM_DEVICE="/dev/dri/ca
rd0" /root/bundle/elinux  --bundle=/root/bundle
[INFO][ConfigureDisplay(71)] resolution: 1280x800
[ERROR][EnvironmentEgl(22)] Failed to get the EGL display: eglGetError: EGL_SUCCESS
[ERROR][ContextEgl(33)] Failed to choose EGL surface config: eglGetError: EGL_BAD_DISPLAY
[ERROR][CreateOnscreenSurface(71)] Failed to create EGL window surface: eglGetError: EGL_BAD_DISPLAY
Failed to create view controller.
```

HidenoriMatsubayashi, 2022-08-21:
`eglGetDisplay` fails on your device, but I'm not sure why it failed.

fayfive, 2022-08-21:
`[ERROR][EnvironmentEgl(22)] Failed to get the EGL display: eglGetError: EGL_SUCCESS` this one is error?

HidenoriMatsubayashi, 2022-08-21:
Yes.

fayfive, 2022-08-21:
The device has tow cards `card0` and `card1`

HidenoriMatsubayashi, 2022-08-21:
Have you set `DISPLAY` env? Also x11 or wayland is running on your device?

fayfive, 2022-08-21:
```
[root@RK356X:~/bundle]# env
USER=root
GST_V4L2_PREFERRED_FOURCC=NV12:YU12:NV16:YUY2
WESTON_DISABLE_ATOMIC=1
WESTON_DRM_DISABLE_MODIFIER=1
SHLVL=3
OLDPWD=/
HOME=/
PIXMAN_USE_RGA=1
PAGER=/bin/more 
GST_VIDEO_CONVERT_USE_RGA=1
PS1=[\u@\h:\w]# 
QT_GSTREAMER_PLAYBIN=playbin3
QT_GSTREAMER_WIDGET_VIDEOSINK=waylandsink
TERM=vt102
WESTON_DRM_KEEP_RATIO=1
GST_VIDEO_CONVERT_PREFERRED_FORMAT=NV12:NV16:I420:YUY2
PATH=/bin:/sbin:/usr/bin:/usr/sbin
XDG_RUNTIME_DIR=/var/run
FLUTTER_DRM_DEVICE=/dev/dri/card1
GST_DEBUG_NO_COLOR=1
GST_V4L2_USE_LIBV4L2=1
SHELL=/bin/sh
storagemedia=emmc
PWD=/root/bundle
QT_QPA_PLATFORM=wayland
WESTON_DRM_MIRROR=1
PREFERED_VIDEOSINK=waylandsink
QT_GSTREAMER_WINDOW_VIDEOSINK=waylandsink
GST_VIDEO_FLIP_USE_RGA=1
EDITOR=/bin/vi
```

HidenoriMatsubayashi, 2022-08-21:
Did you stop Wayland & Weston before running flutter for DRM?

HidenoriMatsubayashi, 2022-08-21:
> And I find that cpu is high performance.

Because you are using debug mode.

fayfive, 2022-08-21:
if `flutter_wayland.so` I don't stop the weston.
if `flutter_gbm.so` I stop the weston, but I don't know how to stop wayland.

HidenoriMatsubayashi, 2022-08-21:
see https://github.com/sony/flutter-embedded-linux/wiki/How-to-run-Flutter-apps#2-run-with-drm-backend

fayfive, 2022-08-21:
> see https://github.com/sony/flutter-embedded-linux/wiki/How-to-run-Flutter-apps#2-run-with-drm-backend

I see that `$ Ctrl + Alt + F3 # Switching to CUI` but I am elinux not ubuntu. I stop weston is ok? `Ctrl + Alt + F3 ` doesn't work for the system.

HidenoriMatsubayashi, 2022-08-21:
Yes. Please don't launch any window manager using DRM.

fayfive, 2022-08-21:
> Yes. Please don't launch any window manager using DRM.

But it'still 
```
[root@RK356X:~/bundle]# FLUTTER_LOG_LEVELS=TRACE FLUTTER_DRM_DEVICE="/dev/dri/ca
rd0" /root/bundle/elinux  --bundle=/root/bundle
[INFO][ConfigureDisplay(71)] resolution: 1280x800
[ERROR][EnvironmentEgl(22)] Failed to get the EGL display: eglGetError: EGL_SUCCESS
[ERROR][ContextEgl(33)] Failed to choose EGL surface config: eglGetError: EGL_BAD_DISPLAY
[ERROR][CreateOnscreenSurface(71)] Failed to create EGL window surface: eglGetError: EGL_BAD_DISPLAY
Failed to create view controller.
```
~_~

HidenoriMatsubayashi, 2022-08-21:
Hmm, I'm afraid that I have no idea to resolve it. It would be great if you investigate/debug this issue.

fayfive, 2022-08-21:
It seem that `src/flutter/shell/platform/linux_embedded/surface/environment_egl_stream.cc` is not compiled, I can see `environment_egl_stream.cc.o`? 
Which option can compile it?

sullivanj91, 2022-08-31:
Hi I'm getting a very similar issue when trying to run inside a docker container on a Jetson Nano host. Here is my command/output:
```
ag@g010-0101-nano:~/mobile/dashboard$ FLUTTER_LOG_LEVELS=TRACE FLUTTER_DRM_DEVICE="/dev/dri/card0" flutter-elinux run -d elinux-x11
Flutter 3.0.5 • channel unknown • unknown source
Framework • revision f1875d570e (7 weeks ago) • 2022-07-13 11:24:16 -0700
Engine • revision e85ea0e79c
Tools • Dart 2.17.6 • DevTools 2.12.2
Launching lib/main.dart on eLinux in debug mode...
Building an eLinux application with x11 backend in debug mode for arm64 target...        68.3s
[ERROR][EnvironmentEgl(22)] Failed to get the EGL display: eglGetError: EGL_SUCCESS
[ERROR][ContextEgl(31)] Failed to choose EGL surface config: eglGetError: EGL_BAD_DISPLAY
[ERROR][NativeWindowX11(33)] Failed to get Visual info.
[ERROR][CreateRenderSurface(132)] Failed to create the native window
Failed to create view controller.
```

Is there anything else i can provide to help debug?

HidenoriMatsubayashi, 2022-08-31:
Can you try the same thing outside the docker container?

sullivanj91, 2022-08-31:
Looks like it was a permission issue, I added `--group-add=video` to my `docker run` and now it displays. However the container is not able to capture touchscreen input events even with `--device=/dev/input/ --group-add=input` mounted.

>Can you try the same thing outside the docker container?

Both display and input are working when running outside of the container.

HidenoriMatsubayashi, 2022-08-31:
Okay. BTW, which JetPack version are you using?

sullivanj91, 2022-08-31:
After some more debugging this is another permission issue it seems. Added `-v /run/udev/data:/run/udev/data` to give `libinput` ability to list devices and then the output is:
```
[ERROR][operator()(208)] Failed to open /dev/input/event0, error: Permission denied
[ERROR][operator()(208)] Failed to open /dev/input/event2, error: Permission denied
[ERROR][operator()(208)] Failed to open /dev/input/event1, error: Permission denied
```
I then added `--user root` and it worked. Trying to figure out now how to give other users permission. `--group-add=input` is insufficient.

>which JetPack version are you using?

Jetpack is `4.9.253-tegra`

HidenoriMatsubayashi, 2022-08-31:
Thanks for the information.

fayfive, 2022-09-03:
> Hi I'm getting a very similar issue when trying to run inside a docker container on a Jetson Nano host. Here is my command/output:
> 
> ```
> ag@g010-0101-nano:~/mobile/dashboard$ ./build/elinux/arm64/debug/bundle/dashboard 
> [WARNING][CreateRenderSurface(111)] FLUTTER_DRM_DEVICE is not set, use /dev/dri/card0
> [ERROR][NativeWindowDrm(22)] Couldn't open /dev/dri/card0
> [ERROR][CreateRenderSurface(118)] Failed to create the native window
> Failed to create view controller.
> ```
> 
> But the `card0` device is mounted to the container:
> 
> ```
> ag@g010-0101-nano:~/mobile/dashboard$ ls /dev/dri/
> card0  renderD128
> ```
> 
> Is there anything else i can provide to help debug?

Hello! How can you get the path `/dev/dri/card0` on jetson?  I can find the path.
Here is the link that jetson doesn't have the "/dev/dri/card0".
https://forums.developer.nvidia.com/t/why-doesnt-jetson-have-the-directory-of-dev-dri-drm/218578

Li-00, 2023-07-18:
After setting up the proxy, flutter-elinux build directly under demo, and run. / [app_name]-- bundle=~/build/elinux/arm64/release/bundle will prompt the same problem in the ~ / build/elinux/arm64/release/bundle path. Then use this article to change the Cmake.txt and place the so library and then re-execute build `flutter-elinux build elinux-- target-arch=arm64-- target-sysroot=/-- target-backend-type=gbm-- verbose`.
Stuck in [+ 14 ms] release_elinux_application: Complete.
[+ 916 ms] Persisting file store.
[+ 62 ms] Done persisting file store.
[+ 682ms] executing: [/ media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/demo/elinux_sample/build/elinux/arm64/release/] cmake-DCMAKE_BUILD_TYPE=Release-DFLUTTER_TARGET_BACKEND_TYPE=gbm.
-DFLUTTER_TARGET_PLATFORM=elinux-arm64 / media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/demo/elinux_sample/elinux

linaro@linaro-alip:/media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/demo/elinux_sample$ ls
README.md  analysis_options.yaml  android  build  elinux  elinux_sample.iml  ios  lib  linux  macos  pubspec.lock  pubspec.yaml  test  web  windows
linaro@linaro-alip:/media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/demo/elinux_sample$ flutter-elinux build elinux --target-arch=arm64 --target-sysroot=/ --target-backend-type=gbm --verbose
Flutter 3.10.5 • channel unknown • unknown source
Framework • revision 796c8ef792 (5 weeks ago) • 2023-06-13 15:51:02 -0700
Engine • revision 45f6e00911
Tools • Dart 3.0.5 • DevTools 2.23.1
[ +385 ms] executing: [/media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/env/flutter-elinux/flutter/] git -c log.showSignature=false log -n 1 --pretty=format:%H
[ +268 ms] Exit code 0 from: git -c log.showSignature=false log -n 1 --pretty=format:%H
[   +2 ms] 796c8ef79279f9c774545b3771238c3098dbefab
[   +5 ms] executing: [/media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/env/flutter-elinux/flutter/] git tag --points-at 796c8ef79279f9c774545b3771238c3098dbefab
[ +104 ms] Exit code 0 from: git tag --points-at 796c8ef79279f9c774545b3771238c3098dbefab
[  +10 ms] 3.10.5
[  +35 ms] executing: [/media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/env/flutter-elinux/flutter/] git rev-parse --abbrev-ref HEAD
[  +57 ms] Exit code 0 from: git rev-parse --abbrev-ref HEAD
[   +1 ms] HEAD
[   +4 ms] executing: [/media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/env/flutter-elinux/flutter/] git rev-parse --abbrev-ref --symbolic @{upstream}
[  +46 ms] Exit code 128 from: git rev-parse --abbrev-ref --symbolic @{upstream}
[   +1 ms] fatal: HEAD does not point to a branch
[ +203 ms] Unable to locate an Android SDK.
[  +90 ms] executing: uname -m
[  +41 ms] Exit code 0 from: uname -m
[   +1 ms] aarch64
[+2122 ms] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
[  +13 ms] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
[  +12 ms] Flutter assets will be downloaded from https://storage.flutter-io.cn. Make sure you trust this source!
[ +126 ms] Artifact Instance of 'ELinuxEngineArtifacts' is not required, skipping update.
[   +5 ms] Artifact Instance of 'MaterialFonts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'GradleWrapper' is not required, skipping update.
[   +1 ms] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
[        ] Artifact Instance of 'FlutterSdk' is not required, skipping update.
[        ] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
[  +13 ms] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[... cut ...]
[ +100 ms] Font asset "MaterialIcons-Regular.otf" was tree-shaken, reducing it from 1645184 to 1336 bytes (99.9% reduction). Tree-shaking can be disabled by providing the --no-tree-shake-icons flag when
building your
                    app.
[  +14 ms] release_elinux_application: Complete
[ +916 ms] Persisting file store
[  +62 ms] Done persisting file store
[ +682 ms] executing: [/media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/demo/elinux_sample/build/elinux/arm64/release/] cmake -DCMAKE_BUILD_TYPE=Release -DFLUTTER_TARGET_BACKEND_TYPE=gbm
-DFLUTTER_TARGET_PLATFORM=elinux-arm64 /media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/demo/elinux_sample/elinux

Li-00, 2023-07-19:
I also ran into this problem, which was eventually solved by finding the right graphical library, my device is 
rk3568
SO：Debain 10 aarch64 
GPU：libmali-bifrost-g52-g2p0-x11.so 
Belongs to the x11 graphical library
Projects generated by flutter-elinux build elinux --target-arch=arm64 --target-sysroot=/ --target-backend-type=x11 --verbose will work