sony/flutter-elinux #123: A question about drm

fayfive, 2022-08-21:
As I know, DRM divide into planes, but flutter elinux never shows which plane is using and qt can specify.
And my app want to using two planes, one is flutter, anther is video.
Can elinux flutter do that?

HidenoriMatsubayashi, 2022-08-21:
I'm not sure what you wan to do, but basically current embedder for DRM backend doesn't support overlay with multi plane.

> And my app want to using two planes, one is flutter, anther is video.

In Flutter, it's usually done with an embedder. No need to use post-DRM.

fayfive, 2022-08-21:
Similarly, QT can specify  which layer it runs.
For example, there are 2 planes. 
Flutter can run on number 1 or 2.
I want to do that flutter runs on the top layer. The second is to play a video. Because video to drm is faster. The flutter ui can be slower, and the  flutter ui  is about 50% transparent

HidenoriMatsubayashi, 2022-08-21:
I see. But it's out/against of flutter's design.

fayfive, 2022-08-21:
I don't think so. Because flutter just use one plane.
Flutter drm can not do it now?
And which plane does flutter drm use?

HidenoriMatsubayashi, 2022-08-21:
> Because flutter just use one plane.

Right, so flutter-elinux for drm doesn't support multi plane overlay.

HidenoriMatsubayashi, 2022-08-21:
I'll close the issue.

closed 2022-08-21 by HidenoriMatsubayashi