sony/flutter-elinux #125: Unable to build app in release mode while debug mode is working fine

Tom3652, 2022-08-23:
Hello ! 

I am trying to build my linux app with the following command for `release` : `flutter-elinux build elinux --target-backend-type=x11 -v` 
but it's not working and it's failing without telling what is the error. 

If i run the command `flutter-elinux build elinux --debug --target-backend-type=x11 -v` it's working fine i have my `bundle` directory created etc... 

I have no clue how to fix it since the error doesn't tell anything. 

I can't copy paste everything as my terminal is too small, but i show you the end : 

*RELEASE* : 

```
[  +20 ms] 💪 Building with sound null safety 💪
[   +2 ms] Building an eLinux application with x11 backend in release mode for arm64 target...
[        ] Initializing file store
[        ] Done initializing file store
[   +7 ms] Skipping target: gen_localizations
[   +5 ms] Skipping target: gen_dart_plugin_registrant
[        ] elinux_plugins: Starting due to {}
[        ] elinux_plugins: Complete
[        ] kernel_snapshot: Starting due to {}
[   +5 ms] /opt/flutter-elinux/flutter/bin/cache/dart-sdk/bin/dart --disable-dart-dev /opt/flutter-elinux/flutter/bin/cache/dart-sdk/bin/snapshots/frontend_server.dart.snapshot --sdk-root /opt/flutter-elinux/flutter/bin/cache/artifacts/engine/common/flutter_patched_sdk_product/ --target=flutter --no-print-incremental-dependencies -Ddart.vm.profile=false -Ddart.vm.product=true --aot --tfa --packages /home/ciele/test_app/.dart_tool/package_config.json --output-dill /home/ciele/test_app/.dart_tool/flutter_build/d2b0f2d38f1b85fb9fa059f0c286f4b6/app.dill --depfile /home/ciele/test_app/.dart_tool/flutter_build/d2b0f2d38f1b85fb9fa059f0c286f4b6/kernel_snapshot.d --source /home/ciele/test_app/.dart_tool/flutter_build/dart_plugin_registrant.dart --source package:flutter/src/dart_plugin_registrant.dart -Dflutter.dart_plugin_registrant=file:///home/ciele/test_app/.dart_tool/flutter_build/dart_plugin_registrant.dart test_app/main.dart
[+40900 ms] Persisting file store
[  +18 ms] Done persisting file store
[  +36 ms] Building an eLinux application with x11 backend in release mode for arm64 target... (completed in 55,4s)
[   +5 ms] "flutter elinux" took 55 892ms.
[  +18 ms] ensureAnalyticsSent: 7ms
[        ] Running shutdown hooks
[        ] Shutdown hooks complete
[   +1 ms] exiting with code 1
```

Along with this error but i think it's simply the command error : 
[image]

*DEBUG* : 

```
[ +105 ms] Building an eLinux application with x11 backend in debug mode for arm64 target... (completed in 6,3s)
[        ] "flutter elinux" took 6 690ms.
[   +2 ms] ensureAnalyticsSent: 0ms
[        ] Running shutdown hooks
[        ] Shutdown hooks complete
[        ] exiting with code 0
```

If you need the whole trace i can provide it but since it's working in debug mode, i don't think it's "package-related". I have already tried by removing all my non-linux packages but the release build still fails.

Tom3652, 2022-08-23:
I must specify things a bit for you to find faster where the error may come from.

If i create an empty project using `flutter-elinux create .` and build it in release mode, it's working fine.
Now in my existing project, when i ran the `flutter-elinux create .` command to add the support as mentioned [here](https://github.com/sony/flutter-elinux/wiki/Creating-flutter-projects#2-use-existing-official-flutter-projects) , it asks me what is my organization name because it's ambiguous (it found 2 names). 

So i did : `flutter-elinux create --org=com.myappname .` and it worked but i don't know the impact on my app. 

Then i run the command mentioned in the above comment and it's failing.

HidenoriMatsubayashi, 2022-08-23:
Are you using the latest version of flutter-elinux? If no, please update it.

Tom3652, 2022-08-24:
The issue is also reproductible with the flutter engine. It comes from one of my dependency that can't be built under `arm64`.

closed 2022-08-24 by Tom3652