sony/flutter-elinux #126: App crashes on any touch input

lionelmennig, 2022-08-25:
First of all, thank you for your amazing work!

### Steps to Reproduce

1. Build and launch any Flutter app with `flutter-elinux run` on Wayland (Ubuntu 20.04 LTS)
2. Touch the screen
3. App crashes

### Error message

`[FATAL:flutter/lib/ui/window/pointer_data_packet_converter.ccc(188)] Check failed: iter != states_.end().`

### Additional infos

This crash won't occur with classic Linux build of the same app, and I can reproduce it with the Flutter demo app (Counter app).

Thank you for your help!

HidenoriMatsubayashi, 2022-08-25:
Can you please put the whole log message?

> Build and launch any Flutter app with flutter-elinux run on Wayland (Ubuntu 20.04 LTS)

Did you use Weston?

HidenoriMatsubayashi, 2022-08-25:
X86_64 or arm64?

lionelmennig, 2022-08-25:
@HidenoriMatsubayashi
1. Weston was not installed
2. x64

Indeed, installing Weston and running the app in this context won't crash on touch, which basically is a fix for my issue. Thank you, I should've tried that before.

HidenoriMatsubayashi, 2022-08-25:
Still, I'm not sure.

> flutter-elinux run on Wayland (Ubuntu 20.04 LTS)

What kind of Wayland compositor do you use?

lionelmennig, 2022-08-26:
I simply used to log in Ubuntu with Wayland mode (from the cogs menu on launch screen), and since the title bar of my app was looking different when built with eLinux against default Linux, I thought it was sufficient to consider being in a Wayland context window.

HidenoriMatsubayashi, 2022-08-26:
Understood. Thanks, but I have never seen this error message/crash.

lionelmennig, 2022-08-27:
Can I provide you anything to help you out about this? The only error message I get in the console is the one I shared, is there any place I can find more details about the crash?

HidenoriMatsubayashi, 2022-08-27:
Can you please try the following commands and get the log messages?

```shell
$ flutter-elinux build elinux --debug
$ FLUTTER_LOG_LEVELS=TRACE ./build/elinux/<target_arch_name>/debug/bundle/<appname> -b .
```

lionelmennig, 2022-08-30:
I think you might be interested by this one, occuring with the `build` command:

> Determining if the CXX compiler works failed with the following output:
> Change Dir: /home/dev/Projects/hello_app/build/elinux/x64/debug/CMakeFiles/CMakeTmp
> 
> Run Build Command(s):/usr/bin/make cmTC_c8aad/fast && /usr/bin/make -f CMakeFiles/cmTC_c8aad.dir/build.make CMakeFiles/cmTC_c8aad.dir/build
> make[1]: Entering directory '/home/dev/Projects/hello_app/build/elinux/x64/debug/CMakeFiles/CMakeTmp'
> Building CXX object CMakeFiles/cmTC_c8aad.dir/testCXXCompiler.cxx.o
> /usr/bin/clang++ --target=aarch64-linux-gnu     -o CMakeFiles/cmTC_c8aad.dir/testCXXCompiler.cxx.o -c /home/dev/Projects/hello_app/build/elinux/x64/debug/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
> Linking CXX executable cmTC_c8aad
> /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_c8aad.dir/link.txt --verbose=1
> /usr/bin/clang++ --target=aarch64-linux-gnu       CMakeFiles/cmTC_c8aad.dir/testCXXCompiler.cxx.o  -o cmTC_c8aad 
> /usr/bin/ld: unrecognised emulation mode: aarch64linux
> Supported emulations: elf_x86_64 elf32_x86_64 elf_i386 elf_iamcu elf_l1om elf_k1om i386pep i386pe
> clang: error: linker command failed with exit code 1 (use -v to see invocation)
> make[1]: *** [CMakeFiles/cmTC_c8aad.dir/build.make:87: cmTC_c8aad] Error 1
> make[1]: Leaving directory '/home/dev/Projects/hello_app/build/elinux/x64/debug/CMakeFiles/CMakeTmp'
> make: *** [Makefile:121: cmTC_c8aad/fast] Error 2

closed 2023-08-16 by HidenoriMatsubayashi