sony/flutter-elinux #127: The getter 'port' was called on null.

Tom3652, 2022-09-01:
I am trying to run on a remote device but i am getting the following error : 

```
foxtom@debian11arm64v2:~/test_app$ flutter-elinux run -d regate-1
Flutter 3.0.5 • channel unknown • unknown source
Framework • revision f1875d570e (il y a 7 semaines) • 2022-07-13 11:24:16 -0700
Engine • revision e85ea0e79c
Tools • Dart 2.17.6 • DevTools 2.12.2
Launching lib/main.dart on eLinux in debug mode...
Uninstall test_app from regate-1.
foxtom@172.16.1.169's password: 
Uninstallation Success
Install test_app (build/elinux/arm64/debug/bundle) to regate-1
foxtom@172.16.1.169's password: 
Installation Success
Launch test_app.name on regate-1
foxtom@172.16.1.169's password: 
Usage: /tmp/test_app/test_app --bundle=<value> 
Global options:
-b, --bundle=<value>             Path to Flutter project bundle
-n, --no-cursor                  No mouse cursor/pointer
-r, --rotation=<value>           Window rotation(degree) [0(default)|90|180|270]
-k, --onscreen-keyboard          Enable on-screen keyboard
-d, --window-decoration          Enable window decorations
-f, --fullscreen                 Always full-screen display
-w, --width=<value>              Window width
-h, --height=<value>             Window height
bundle option is mandatory.

NoSuchMethodError: The getter 'port' was called on null.
Receiver: null
Tried calling: port
#0      Object.noSuchMethod (dart:core-patch/object_patch.dart:38:5)
#1      ELinuxDevice.startApp (package:flutter_elinux/elinux_device.dart:179:45)
<asynchronous suspension>
#2      FlutterDevice.runHot (package:flutter_tools/src/resident_runner.dart:441:33)
<asynchronous suspension>
#3      Future.wait.<anonymous closure> (dart:async/future.dart:521:21)
<asynchronous suspension>
#4      HotRunner.run (package:flutter_tools/src/run_hot.dart:424:34)
<asynchronous suspension>
#5      RunCommand.runCommand (package:flutter_tools/src/commands/run.dart:692:26)
<asynchronous suspension>
#6      FlutterCommand.run.<anonymous closure> (package:flutter_tools/src/runner/flutter_command.dart:1183:27)
<asynchronous suspension>
#7      AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
<asynchronous suspension>
#8      CommandRunner.runCommand (package:args/command_runner.dart:209:13)
<asynchronous suspension>
#9      FlutterCommandRunner.runCommand.<anonymous closure> (package:flutter_tools/src/runner/flutter_command_runner.dart:281:9)
<asynchronous suspension>
#10     AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
<asynchronous suspension>
#11     FlutterCommandRunner.runCommand (package:flutter_tools/src/runner/flutter_command_runner.dart:229:5)
<asynchronous suspension>
#12     run.<anonymous closure>.<anonymous closure> (package:flutter_tools/runner.dart:62:9)
<asynchronous suspension>
#13     AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
<asynchronous suspension>
#14     main (package:flutter_elinux/executable.dart:83:3)
<asynchronous suspension>
```

```
{
	"custom-devices": [
	  {
	    "id": "regate-1",
	    "label": "reGATE 1",
	    "sdkNameAndVersion": "reGATE 1",
	    "enabled": true,
	    "platform": "arm64",
	    "backend": "wayland",
	    "ping": [
	      "ping", "-w", "500", "-c", "1", "172.16.1.169"
	    ],
	    "pingSuccessRegex": "ttl=",
	    "install": [
	      "scp", "-r", "${localPath}", "foxtom@172.16.1.169:/tmp/${appName}"
	    ],
	    "uninstall": [
	      "ssh", "foxtom@172.16.1.169", "rm -rf \"/tmp/${appName}\""
	    ],
	    "runDebug": [
	      "ssh", "foxtom@172.16.1.169", "/tmp/${appName}/${appName}"
	    ],
	    "forwardPort": [
	      "ssh", "-o", "ExitOnForwardFailure=yes",
	      "-L", "127.0.0.1:${hostPort}:127.0.0.1:${devicePort}", "foxtom@172.16.1.169"
	    ],
	    "forwardPortSuccessRegex": "Linux"
	  }
	]
      }
```

I have tried the port forwarding command manually in the terminal and everything seemed to be working fine by using a port like `1295` randomly. 

What i am doing wrong please ?

HidenoriMatsubayashi, 2022-09-01:
Can you try the below? We needed to add "**-b .**". I'll update Wiki.

```
	    "runDebug": [
	      "ssh", "foxtom@172.16.1.169", "/tmp/${appName}/${appName} -b ."
	    ],
```

Tom3652, 2022-09-02:
Thanks that fixed the error and i understand why now, it should pass the `bundle` parameter as request when you run the app manually.

Tom3652, 2022-09-02:
Well actually i have another error now : 

```
[ERROR][ELinuxWindowX11(30)] Failed to open display.
[ERROR][InitializeEgl(52)] Failed to initialize the EGL display: eglGetError: EGL_NOT_INITIALIZED
[ERROR][ContextEgl(31)] Failed to choose EGL surface config: eglGetError: EGL_NOT_INITIALIZED

NoSuchMethodError: The getter 'port' was called on null.
```

I have done the `export DISPLAY=:0` on another SSH session on my target device. Isn't it enough ?

HidenoriMatsubayashi, 2022-09-02:
> I have done the export DISPLAY=:0 on another SSH session on my target device. Isn't it enough ?

I'm not sure, but I guess some settings for x11 are missing on your device.

Tom3652, 2022-09-02:
Well if i run the binary file on my device compiled with x11 backend it's working just fine :/

Tom3652, 2022-09-05:
Hey @HidenoriMatsubayashi, do you mean that it's actually something you may have to fix if i understand correctly ? 

Thanks !

HidenoriMatsubayashi, 2022-09-05:
Sorry for confusing you. Can you try launching any X11 apps on your device via SSH from your host PC?

Tom3652, 2022-09-05:
Oh yes sure, this is actually working fine and this is how i do right now : 
- Compile my app with x11 backend (in release mode) 
- `scp` the `bundle` to my device 
- Run the bundle locally on the device through SSH from my host PC
- My app is working fine and displayed

The problem is that the debug is really slow because all of these steps... I wish i could use the `flutter-elinux run` cmd to directly debug without having to do all these steps first :/

HidenoriMatsubayashi, 2022-09-05:
Okay, great. Then, can you try the following (set DISPLAY env before "/tmp/${appName}/${appName}")?

```
	    "runDebug": [
	      "ssh", "foxtom@172.16.1.169", "DISPLAY=:0.0 /tmp/${appName}/${appName} -b ."
	    ],
```

HidenoriMatsubayashi, 2022-09-05:
> i guess hot restart / hot reload are not supported since it's with SSH ?

They are supported.

Tom3652, 2022-09-05:
I feel stupid it's written. 
Thanks for your work on flutter-elinux it's a really amazing tool

closed 2022-09-05 by Tom3652