sony/flutter-elinux #128: Latest release still doesn't build arm64 targets

Taha-Firoz, 2022-09-01:
I've already reported on #114 the error has changed now, 
 I've tried both debug and release neither work, also all of my builds are automated so it's highly unlikely that I've changed anything in my setup. Any release prior to the flutter 3.0.0 release builds fine. Also I verified that the `sys_root` does contain the `ld-linux-aarch64.so.1` it maybe that it's not setting the sysroot correctly

The build command used
```
flutter-elinux build elinux --debug -v --target-backend-type=gbm  --target-arch=arm64 --target-sysroot=/home/me/rsc/arm64-sysroot --system-include-directories=/usr/aarch64-linux-gnu/include/c++/9/aarch64-linux-gnu
```

The error
```
[   +5 ms] Failed to cmake:
           -- The CXX compiler identification is Clang 10.0.0
           -- Check for working CXX compiler: /usr/bin/clang++
           -- Check for working CXX compiler: /usr/bin/clang++ -- broken
           -- Configuring incomplete, errors occurred!
           See also "/home/me/repos/prj/build/elinux/arm64/debug/CMakeFiles/CMakeOutput.log".
           See also "/home/me/repos/prj/build/elinux/arm64/debug/CMakeFiles/CMakeError.log".
           
           CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCXXCompiler.cmake:53 (message):
             The C++ compiler
           
               "/usr/bin/clang++"
           
             is not able to compile a simple test program.
           
             It fails with the following output:
           
               Change Dir: /home/me/repos/prj/build/elinux/arm64/debug/CMakeFiles/CMakeTmp
           
               Run Build Command(s):/home/me/rsc/arm64-sysroot/usr/bin/make cmTC_9cca1/fast && /lib/ld-linux-aarch64.so.1: No such file or directory
           
           
           
           
           
             CMake will not be able to correctly generate this project.
           Call Stack (most recent call first):
             CMakeLists.txt:2 (project)
[   +2 ms] 
           #0      throwToolExit (package:flutter_tools/src/base/common.dart:10:3)
           #1      NativeBundle.build (package:flutter_elinux/elinux_build_target.dart:390:7)
           <asynchronous suspension>
           #2      ELinuxBuilder.buildBundle (package:flutter_elinux/elinux_builder.dart:139:7)
           <asynchronous suspension>
           #3      BuildPackageCommand.runCommand (package:flutter_elinux/commands/build.dart:134:5)
           <asynchronous suspension>
           #4      FlutterCommand.run.<anonymous closure> (package:flutter_tools/src/runner/flutter_command.dart:1183:27)
           <asynchronous suspension>
           #5      AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
           <asynchronous suspension>
           #6      CommandRunner.runCommand (package:args/command_runner.dart:209:13)
           <asynchronous suspension>
           #7      FlutterCommandRunner.runCommand.<anonymous closure> (package:flutter_tools/src/runner/flutter_command_runner.dart:281:9)
           <asynchronous suspension>
           #8      AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
           <asynchronous suspension>
           #9      FlutterCommandRunner.runCommand (package:flutter_tools/src/runner/flutter_command_runner.dart:229:5)
           <asynchronous suspension>
           #10     run.<anonymous closure>.<anonymous closure> (package:flutter_tools/runner.dart:62:9)
           <asynchronous suspension>
           #11     AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
           <asynchronous suspension>
           #12     main (package:flutter_elinux/executable.dart:83:3)
           <asynchronous suspension>
```

Tom3652, 2022-09-02:
For me this kind of error only appeared when trying to do some cross-building compilation and using `sys_root` etc... 
If you are not cross-compiling, you can just do : 

```
flutter-elinux build elinux --debug -v --target-backend-type=gbm
```

It's working for me as long as i run this cmd on an `arm64` linux device.

Taha-Firoz, 2022-09-02:
Yup but I've been crosscompiling for months before this, these recent commits had a lot of changes and some of them changed how cross compilation was done

Tom3652, 2022-09-02:
Oh alright, well i won't be much of a help here then sorry

closed 2022-09-02 by HidenoriMatsubayashi