sony/flutter-elinux #136: Full screen flag doesn't work (X11)

Tom3652, 2022-09-22:
Hello, 

I am running my app under X11 and i have managed to make it run at a specific width / height using the `-w` and the `-h` flags.
However, my app has a title app with my flutter app name. 

I would like to remove this title bar, so i have checked and i saw that there is a full screen flag `-f` but it's not working i have a half sized window (because i don't use the  `-w` and `-h` flags along with the `-f`)

HidenoriMatsubayashi, 2022-09-22:
We don't basically support X11 backend. Please use Wayland backend, or send a patch to fix the issue if you want to use X11 backend.

See https://github.com/sony/flutter-embedded-linux/issues/37

Tom3652, 2022-09-23:
I can't find any way to run Wayland on my device i am not a linux expert... 

But i understand for this issue thanks

HidenoriMatsubayashi, 2023-09-29:
Closing as I added this option support (https://github.com/sony/flutter-embedded-linux/pull/389). The change will be included in the upcoming flutter-elinux release version.

closed 2023-09-29 by HidenoriMatsubayashi