sony/flutter-elinux #138: audioplayer  setSourceUrl() not found

jinbo1996, 2022-10-10:
log:
arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '7'.
flutter: The Dart VM service is listening on http://127.0.0.1:41859/8my5k4tgU9Q=/
flutter: ---------------------
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: MissingPluginException(No implementation found for method setSourceUrl on channel xyz.luan/audioplayers)
#0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:294:7)
<asynchronous suspension>

jinbo1996, 2022-10-10:
I add the dependencies audioplayers: ^1.1.1 in pubspec.yaml
it can play audio on the web,but can not found setSourceUrl on linux device

HidenoriMatsubayashi, 2022-10-10:
I don't know which plugin you are using, but it seems that there is a problem with the implementation of the plugin. Plugin implementation is not taken care of in this repository. Please report to the plugin you used.

closed 2022-10-10 by HidenoriMatsubayashi

jinbo1996, 2022-10-10:
ok