sony/flutter-elinux #140: Error Artefect download

thomtomdup, 2022-10-18:
Hello.
I tried to install the new release of e-linux.
I have create a new project with flutter-elinux to verify if it's my project but not.
I have this error when i run the build : 

Downloading an artifact that may not be reachable in some environments (e.g.
firewalled environments):
https://github.com/sony/flutter-embedded-linux/releases/download/c08d7d5efc/elin
ux-common.zip
This should not have happened. This is likely a Flutter SDK bug. Please file an
issue at https://github.com/flutter/flutter/issues/new?template=1_activation.md
Downloading elinux-common tools...                                     ⣯Downloading elinux-common tools...                                 6,0s
Downloading an artifact that may not be reachable in some environments (e.g.
firewalled environments):
https://github.com/sony/flutter-embedded-linux/releases/download/c08d7d5efc/elin
ux-arm64-debug.zip
This should not have happenedDownloading an artifact that may not be reachable in some environments (e.g.
firewalled environments):
https://github.com/sony/flutter-embedded-linux/releases/download/c08d7d5efc/elin
ux-common.zip
This should not have happened. This is likely a Flutter SDK bug. Please file an
issue at https://github.com/flutter/flutter/issues/new?template=1_activation.md
Downloading elinux-common tools...                                     ⣯Downloading elinux-common tools...                                 6,0s
Downloading an artifact that may not be reachable in some environments (e.g.
firewalled environments):
https://github.com/sony/flutter-embedded-linux/releases/download/c08d7d5efc/elin
ux-arm64-debug.zip
This should not have happened. This is likely a Flutter SDK bug. Please file an
issue at https://github.com/flutter/flutter/issues/new?template=1_activation.md
Downloading elinux-arm64-debug tools...                                ⣯Downloading elinux-arm64-debug tools...                             8,4s
Downloading an artifact that may not be reachable in some environments (e.g.
firewalled environments):
https://github.com/sony/flutter-embedded-linux/releases/download/c08d7d5efc/elin
ux-arm64-profile.zip
This should not have happened. This is likely a Flutter SDK bug. Please file an
issue at https://github.com/flutter/flutter/issues/new?template=1_activation.md
Downloading elinux-arm64-profile tools...                              ⣯Downloading elinux-arm64-profile tools...                           4,3s
Downloading an artifact that may not be reachable in some environments (e.g.
firewalled environments):
https://github.com/sony/flutter-embedded-linux/releases/download/c08d7d5efc/elin
ux-arm64-release.zip
This should not have happened. This is likely a Flutter SDK bug. Please file an
issue at https://github.com/flutter/flutter/issues/new?template=1_activation.md
Downloading elinux-arm64-release tools...                              ⣯Downloading elinux-arm64-release tools...                           3,4s
Downloading an artifact that may not be reachable in some environments (e.g.
firewalled environments):
https://github.com/sony/flutter-embedded-linux/releases/download/c08d7d5efc/elin
ux-x64-debug.zip
This should not have happened. This is likely a Flutter SDK bug. Please file an
issue at https://github.com/flutter/flutter/issues/new?template=1_activation.md
Downloading elinux-x64-debug tools...                                  ⣯Downloading elinux-x64-debug tools...                               7,6s
Downloading an artifact that may not be reachable in some environments (e.g.
firewalled environments):
https://github.com/sony/flutter-embedded-linux/releases/download/c08d7d5efc/elin
ux-x64-profile.zip
This should not have happened. This is likely a Flutter SDK bug. Please file an
issue at https://github.com/flutter/flutter/issues/new?template=1_activation.md
Downloading elinux-x64-profile tools...                                ⣯Downloading elinux-x64-profile tools...                             3,2s
Downloading an artifact that may not be reachable in some environments (e.g.
firewalled environments):
https://github.com/sony/flutter-embedded-linux/releases/download/c08d7d5efc/elin
ux-x64-release.zip
. This is likely a Flutter SDK bug. Please file an
issue at https://github.com/flutter/flutter/issues/new?template=1_activation.md
Downloading elinux-arm64-debug tools...                                ⣯Downloading elinux-arm64-debug tools...                             8,4s
Downloading an artifact that may not be reachable in some environments (e.g.
firewalled environments):
https://github.com/sony/flutter-embedded-linux/releases/download/c08d7d5efc/elin
ux-arm64-profile.zip
This should not have happened. This is likely a Flutter SDK bug. Please file an
issue at https://github.com/flutter/flutter/issues/new?template=1_activation.md
Downloading elinux-arm64-profile tools...                              ⣯Downloading elinux-arm64-profile tools...                           4,3s
Downloading an artifact that may not be reachable in some environments (e.g.
firewalled environments):
https://github.com/sony/flutter-embedded-linux/releases/download/c08d7d5efc/elin
ux-arm64-release.zip
This should not have happened. This is likely a Flutter SDK bug. Please file an
issue at https://github.com/flutter/flutter/issues/new?template=1_activation.md
Downloading elinux-arm64-release tools...                              ⣯Downloading elinux-arm64-release tools...                           3,4s
Downloading an artifact that may not be reachable in some environments (e.g.
firewalled environments):
https://github.com/sony/flutter-embedded-linux/releases/download/c08d7d5efc/elin
ux-x64-debug.zip
This should not have happened. This is likely a Flutter SDK bug. Please file an
issue at https://github.com/flutter/flutter/issues/new?template=1_activation.md
Downloading elinux-x64-debug tools...                                  ⣯Downloading elinux-x64-debug tools...                               7,6s
Downloading an artifact that may not be reachable in some environments (e.g.
firewalled environments):
https://github.com/sony/flutter-embedded-linux/releases/download/c08d7d5efc/elin
ux-x64-profile.zip
This should not have happened. This is likely a Flutter SDK bug. Please file an
issue at https://github.com/flutter/flutter/issues/new?template=1_activation.md
Downloading elinux-x64-profile tools...                                ⣯Downloading elinux-x64-profile tools...                             3,2s
Downloading an artifact that may not be reachable in some environments (e.g.
firewalled environments):
https://github.com/sony/flutter-embedded-linux/releases/download/c08d7d5efc/elin
ux-x64-release.zip

Can you help me about this issue ?

Thank you.
Regards

HidenoriMatsubayashi, 2022-10-19:
These message are not problem. Because the official flutter assumes the  URL which the artifact is downloaded. If there is a chance to improve it, I will improve it, so I will keep the issue open.

Thanks.

thomtomdup, 2022-10-19:
OK thank you but the problem is with that issue I can't build any project.
To be able to build, I had to used an old version of flutter e-linux that I had backup.

HidenoriMatsubayashi, 2022-10-19:
Are you using proxy? Can you delete current flutter-elinux directory once? Then, git clone and flutter-elinux run?

thomtomdup, 2022-10-19:
No proxy and the folder of flutter-elinux was renamed to flutter-elinux-old before git clone.

HidenoriMatsubayashi, 2022-10-19:
I guess your internet speed is a little bit slow. What is your internet speed?

Artifacts for elinux can be downloaded by using the following command.
```
$ flutter-elinux precache --elinux
```

thomtomdup, 2022-11-02:
Hello.

I tried your solution but the issue is not solved.

After this i tried to build my app like i did before but i have some errors of build.

My process is the same than the wiki document : https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps
( using docker to compile in arm64).

**Error details :**

`Failed to cmake:
-- The CXX compiler identification is Clang 11.0.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - failed
-- Check for working CXX compiler: /usr/bin/clang++
-- Check for working CXX compiler: /usr/bin/clang++ - broken
-- Configuring incomplete, errors occurred!
See also
"/home/debian/projectapp/build/elinux/arm64/release/CMakeFiles/CMakeOutput.log".
See also
"/home/debian/projectapp/build/elinux/arm64/release/CMakeFiles/CMakeError.log".

CMake Error at /usr/share/cmake-3.18/Modules/CMakeTestCXXCompiler.cmake:59
(message):
  The C++ compiler

    "/usr/bin/clang++"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir:
    /home/debian/projectapp/build/elinux/arm64/release/CMakeFiles/CMakeTmp

    Run Build Command(s):/usr/bin/gmake cmTC_b0a5a/fast && /usr/bin/gmake  -f
    CMakeFiles/cmTC_b0a5a.dir/build.make CMakeFiles/cmTC_b0a5a.dir/build
    gmake[1] : on entre dans le répertoire
    « /home/debian/projectapp/build/elinux/arm64/release/CMakeFiles/CMakeTmp »
    Building CXX object CMakeFiles/cmTC_b0a5a.dir/testCXXCompiler.cxx.o
    /usr/bin/clang++ --sysroot=/home/debian/ubuntu-arm64-sysroot    -o
    CMakeFiles/cmTC_b0a5a.dir/testCXXCompiler.cxx.o -c
    /home/debian/projectapp/build/elinux/arm64/release/CMakeFiles/CMakeTmp/testCX
    XCompiler.cxx
    Linking CXX executable cmTC_b0a5a
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_b0a5a.dir/link.txt
    --verbose=1
    /usr/bin/clang++ --sysroot=/home/debian/ubuntu-arm64-sysroot
    CMakeFiles/cmTC_b0a5a.dir/testCXXCompiler.cxx.o -o cmTC_b0a5a
    /usr/bin/ld : ne peut pas trouver crt1.o : Aucun fichier ou dossier de ce
    type
    /usr/bin/ld : ne peut pas trouver crti.o : Aucun fichier ou dossier de ce
    type
    /usr/bin/ld : ne peut trouver -lm
    /usr/bin/ld : ne peut trouver libgcc_s.so.1
    clang: error: linker command failed with exit code 1 (use -v to see
    invocation)
    gmake[1]: *** [CMakeFiles/cmTC_b0a5a.dir/build.make:106 : cmTC_b0a5a] Erreur
    1
    gmake[1] : on quitte le répertoire
    « /home/debian/projectapp/build/elinux/arm64/release/CMakeFiles/CMakeTmp »
    gmake: *** [Makefile:140 : cmTC_b0a5a/fast] Erreur 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:4 (project)
`

Could you please help me to solve this probleme ? :) 
Thank you very much.
Best regards.

thomtomdup, 2022-11-02:
Hello:
Just for information, this is since flutter 3.3.0 version. I did some tests with différents versions of flutter-elinux.

Did you change something between 3.0.5 and 3.3.0 ? 😊.

Thank you.

thomtomdup, 2022-12-22:
Hello.
I found the issue and solved it.
I needed to erase my elinux directory on my project and run flutter-elinux create to create the directory and all work good .
Also i saw that you forgot to add command parameter in the wiki .

[Troubleshooting Part](https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps#troubleshooting
)
```
flutter-elinux build elinux --target-arch=arm64 \
     --target-sysroot=<Absolute_path_to>/ubuntu18-arm64-sysroot \
     --system-include-directories=/usr/aarch64-linux-gnu/include/c++/${version}/aarch64-linux-gnu
```

You forgot to add this :
`     --target-compiler-triple=aarch64-linux-gnu \`

Thank you,
Regards

kero990, 2023-03-21:
> 我猜你的网速有点慢。你的网速好吗？
> 
> 可以使用以下命令下载 elinux 的工件。
> 
> ```
> $ flutter-elinux precache --elinux
> ```

Hello, it is a little difficult to access github where I am, so I can't use the download method of automatically connecting to github. I can download these packages manually, but I don't know which directory to put them in to skip the automatic download? I don't even know whether to download to the flutter directory, the flutter-elinux directory or the project's directory.
Can I download manually and skip the automatic download? What directory should it be placed in?
Or can I change a dart file to change the download url point?

HidenoriMatsubayashi, 2023-03-22:
> Can I download manually and skip the automatic download? What directory should it be placed in?

Unfortunately, no. But you might be able to use `ELINUX_ENGINE_BASE_URL`. Is it enough for you use cases?
See https://github.com/sony/flutter-elinux/blob/main/lib/elinux_cache.dart#L101

HidenoriMatsubayashi, 2023-07-16:
Created a new page:
https://github.com/sony/flutter-elinux/wiki/Use-local-elinux-artifacts-without-downloading-from-GitHub