sony/flutter-elinux #141: How to build the Flutter App for arm32 with flutter-elinux?

HerryGeng, 2022-10-21:
I have successfully built libflutter-engine.so and libflutter_elinux_wayland.so.
When I built arm32's flutter app with flutter-elinux, I failed.
What should I do to successfully build the flutter app for arm32?

Thanks very much！

HidenoriMatsubayashi, 2022-10-21:
Unfortunately, we are not going to support arm32 arch. Our target arch is only aarch64 (ARM64) in flutter-elinux.

HerryGeng, 2022-10-21:
I wanted to try it on my own. Can you offer any help or directions?
Thank you!

HidenoriMatsubayashi, 2022-10-21:
I think https://github.com/sony/flutter-embedded-linux/wiki/Building-Flutter-apps#2-building-flutter-apps will help your work for arm32.

HerryGeng, 2022-10-22:
Thank you very much!

> I think https://github.com/sony/flutter-embedded-linux/wiki/Building-Flutter-apps#2-building-flutter-apps will help your work for arm32.

LLLLL72, 2022-10-22:
@HerryGeng  I have the same need with you to run flutter on arm32 device. I have successfully built libflutter-engine.so for arm64. But faile when build flutter engine for arm32, Could you give me some instructions on how to build libflutter-engine.so for arm32? Thank you.

HidenoriMatsubayashi, 2023-01-03:
I'll close this issue.

closed 2023-01-03 by HidenoriMatsubayashi