sony/flutter-elinux #143: Add support for multiple dri cards.

callumhopkinsdev, 2022-10-23:
Currently using the DRM backend you can only select a single card via `FLUTTER_DRM_DEVICE`. However, on the Raspberry Pi 4 the vc4 driver can randomly select either `/dev/dri/card0` or `/dev/dri/card1` at boot time. 

Most other DRM based applications iterate through all of the devices in the `/dev/dri/` directory and attempt to open each one until they are successful. 

It would be great if either the `FLUTTER_DRM_DEVICE` variable allowed for a comma seperated list of devices to attempt to open, or if by default the DRM backend would attempt to open every device until one is successful.

HidenoriMatsubayashi, 2022-10-23:
Thanks for creating the issue.

> However, on the Raspberry Pi 4 the vc4 driver can randomly select either /dev/dri/card0 or /dev/dri/card1 at boot time.

Really? Where can I find that information?

callumhopkinsdev, 2022-10-23:
On this thread, a few posts down you can find a quote directly from one of the engineers. 

https://forums.raspberrypi.com/viewtopic.php?t=307392&sid=0519efbedbfc3f01ff2dbf41547ff41e

Moreover, I've experienced this behaviour personally when working on a C# GUI toolkit (Avalonia)

HidenoriMatsubayashi, 2022-10-24:
Thanks for the information. I'll implement this feature.

HidenoriMatsubayashi, 2022-10-24:
I created https://github.com/sony/flutter-embedded-linux/pull/295

HidenoriMatsubayashi, 2022-10-24:
@callumhopkinsdev The change will be included in a next release (flutter 3.3.6?). If you want to add more improvements, please create a PR. Thanks.

closed 2022-10-24 by HidenoriMatsubayashi

callumhopkinsdev, 2022-10-24:
Awesome, thanks @HidenoriMatsubayashi!