sony/flutter-elinux #145: [Question] icudtl.dat size

psstoyanov, 2022-11-03:
Hi, I'm looking at a few of the commands and how elinux is extending the default Flutter ones. `--analyze-size` caught my eye.

Understandably, it doesn't work with flutter-elinux however this is interesting:
icudtl.dat with flutter-elinux has a size of 10MB:
```
➜  data git:(main) ✗ ls -lh
total 10M
drwxr-xr-x 1 pak0stpc pak0stpc 140 Nov  3 12:36 flutter_assets
-rw-r--r-- 1 pak0stpc pak0stpc 10M Nov  3 12:36 icudtl.dat

```

Comparing this with the generic GTK wrapper:
```
➜  ecommerce_app git:(main) ✗ flutter build linux --analyze-size
Running "flutter pub get" in ecommerce_app...                    1,473ms

💪 Building with sound null safety 💪
Building Linux application...                                           
▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
  bundle                                                                   62 MB
  bundle/
    ecommerce_app                                                          28 KB
    data/
      icudtl.dat                                                          796 KB
```

Why would there be such a drastic difference in the size between the two files?

HidenoriMatsubayashi, 2022-11-04:
Thanks. Both issues need to be investigated.

HidenoriMatsubayashi, 2022-11-04:
> Understandably, it doesn't work with flutter-elinux however this is interesting:

Fixed.

HidenoriMatsubayashi, 2022-11-05:
icudtl.dat is one of the flutter engine build artifacts. The engine build options may have an effect. Current build options: https://github.com/sony/flutter-embedded-linux/wiki/Building-Flutter-Engine-from-source

psstoyanov, 2022-11-05:
I'm waiting for a build to complete with:
https://github.com/flutter/engine/blob/main/ci/builders/linux_host_engine.json#L184

Not sure if that's comparable but it seemed like a good idea to start with whatever the main CI builder uses and then drone to the next option.

psstoyanov, 2022-11-07:
Hmm.... I don't get it - I always get 10M for icudtl.dat in the resulting local build regardless of architecture or options for any Linux output.
 
How is it that the officially released GTK build has a footprint of just a few KB :confused:

psstoyanov, 2022-11-07:
**Update:**
I believe this is the source of information relevant to the above observations: https://flutter-review.googlesource.com/c/recipes/+/9740 

Ohhh... the GTK wrapper doesn't need to bundle `fontconfig` as it's already part of the GTK deps - https://github.com/flutter/engine/pull/23663#issuecomment-759848028 

Anecdotal comparison with `fontconfig` package on my x64 tower (Plasma) being roughly 1100 KiB in size, appears to be in the right ball-park in terms of size.

I'm not sure if the above would have anything to do with a potential Wayland session for `flutter-elinux` but it might have something to do with the fonts not being rendered in debug mode.

I don't believe there is something else that may be discovered by this ticket, @HidenoriMatsubayashi . 
Closing it down.

closed 2022-11-07 by psstoyanov

HidenoriMatsubayashi, 2022-11-07:
Thanks, @psstoyanov