sony/flutter-elinux #147: [Advantech x86_64 X11 2GB RAM] Deploying flutter app problem

alexmac98, 2022-11-09:
Hello. 

I have an Advantech (arch x86_64 | X11 enabled | OS Ubuntu 20.04 Desktop | 2GB RAM) and I'm wondering if flutter-elinux can be used to deploy some flutter application on the Advantech. 
I've already installed the flutter-elinux in it, and the command works. However, when I try to deploy/run the flutter application, it takes a while until it "freezes", and the app is not deployed. I'm supposing it has to do with extreme resource usage (2 CPU cores at 100%, lots of memory being used). The application is nothing special.

Am I misinterpreting something regarding the flutter-elinux purpose/capabilities?
Isn't it supposed to be a lightweight solution of Flutter that can be used to deploy Flutter apps on weaker linux systems, for example? 
Can you help me find a way to diminish resourse usage? 

Best regards,
Alex

HidenoriMatsubayashi, 2022-11-09:
I cannot give any solutions to help as of now, but flutter-elinux is almost same with the official flutter (flutter-elinux depends on the official flutter). Can you please try the official flutter on that machine? Is there any difference in terms of resource usage?

alexmac98, 2022-11-10:
Hello. Thank you for your response.
I've tested both flutter and flutter-elinux on my laptop and flutter-elinux is more lightweight and consumes less resources than flutter, hands down. The problem has to do with the building of the application. The 

### flutter(-elinux) run --no-build 

flag doesn't work properly, since it builds the application anyway. In other words, even though I can build the application on my laptop, and copy it to the deployment machine, the Advantech will still try to build the app when it is deploying the GUI and it will freeze since it doesn't have enough RAM to build it (it needs around 2.1 GB to build, tested in a DRPC machine with 4GB of RAM and it worked properly).

I know this isn't your fault, it's seems to be a Flutter issue.

Will flutter-elinux support -no--build? 

Commands used:

$ flutter-elinux run --no-build -d elinux-x11

This issue is from 2020 but I guess it isn't fixed.
https://github.com/flutter/flutter/issues/68529

HidenoriMatsubayashi, 2022-11-12:
> Will flutter-elinux support -no--build?

Yes, if it's possible.

HidenoriMatsubayashi, 2023-01-03:
I'll close this issue.