sony/flutter-elinux #151: Not running in AOT mode but could not resolve the kernel binary.

Lynchez, 2022-11-17:
run with x11 its work but when i try to get build
 `flutter-elinux build elinux --debug --target-backend-type=x11`
 it gives this error
`embedder.cc (1218): 'FlutterEngineInitialize' returned 'kInvalidArguments'. Not running in AOT mode but could not resolve the kernel binary.
[ERROR][RunWithEntrypoint(255)] Failed to start Flutter engine: error 2
Failed to create view controller.
`
flutter-elinux doctor seems fine. Device is Jetson nano jetpack 4.6

the other backend types gives error too /dev/dri/card0
Failed to connect to the Wayland display.

HidenoriMatsubayashi, 2022-11-17:
>run with x11 its work but when i try to get build
>flutter-elinux build elinux --debug --target-backend-type=x11
>it gives this error

I'm not sure, that is a command to build (not to run). Can you please share whole steps and logs?

Lynchez, 2022-11-17:
I can run only with `flutter-elinux run -d elinux-x11` the other running options gives error. But when i try to get build with this command `flutter-elinux build elinux --debug --target-backend-type=x11` it gives error too.

`
embedder.cc (1218): 'FlutterEngineInitialize' returned 'kInvalidArguments'. Not running in AOT mode but could not resolve the kernel binary. [ERROR][RunWithEntrypoint(255)] Failed to start Flutter engine: error 2 Failed to create view controller.
`

HidenoriMatsubayashi, 2023-06-16:
Sorry for the late response. I confirmed 'flutter-elinux build elinux --debug --target-backend-type=x11' worked fine on Raspberry Pi4. Please try the latest version.

closed 2023-06-16 by HidenoriMatsubayashi