sony/flutter-elinux #153: Run Flutter sample app in Weston Failed.

WolsYang, 2022-12-07:
I add  flutter-wayland-client to my yocto image recipe
`IMAGE_INSTALL:append = " flutter-wayland-client"`

And install flutter-elinux on my machine which is imx8qxp :
`$ git clone https://github.com/sony/flutter-elinux.git
$ sudo mv flutter-elinux /opt/
$ export PATH=$PATH:/opt/flutter-elinux/bin`

Then check device
`$ flutter-elinux devices                                                                                                                                                                      
   Woah! You appear to be trying to run flutter as root.                                                                                                                                                            
   We strongly recommend running the flutter tool without superuser privileges.                                                                                                                                     
  /                                                                                                                                                                                                                 
�📎                                                                                                                                                                                                                 
Flutter 3.3.9 �• channel unknown �• unknown source                                                                                                                                                                  
Framework �• revision b8f7f1f986 (2 weeks ago) �• 2022-11-23 06:43:51 +0900                                                                                                                                         
Engine �• revision 8f2221fbef                                                                                                                                                                                       
Tools �• Dart 2.18.5 �• DevTools 2.15.0                                                                                                                                                                             
3 connected devices:                                                                                                                                                                                                
                                                                                                                                                                                                                    
Linux (desktop)  �• linux          �• linux-arm64    �• NXP i.MX Release Distro 5.15-kirkstone (kirkstone) :) 5.15.52+g36363d8623ba                                                                                 
eLinux (desktop) �• elinux-wayland �• flutter-tester �• NXP i.MX Release Distro 5.15-kirkstone (kirkstone) :) 5.15.52+g36363d8623ba                                                                                 
eLinux (desktop) �• elinux-x11     �• flutter-tester �• NXP i.MX Release Distro 5.15-kirkstone (kirkstone) :) 5.15.52+g36363d8623ba`

Now run sample but failed.
`$ flutter-elinux run -d elinux-wayland                                                                                                                                                        
   Woah! You appear to be trying to run flutter as root.                                                                                                                                                            
   We strongly recommend running the flutter tool without superuser privileges.                                                                                                                                     
  /                                                                                                                                                                                                                 
�📎                                                                                                                                                                                                                 
Flutter 3.3.9 �• channel unknown �• unknown source                                                                                                                                                                  
Framework �• revision b8f7f1f986 (2 weeks ago) �• 2022-11-23 06:43:51 +0900                                                                                                                                         
Engine �• revision 8f2221fbef                                                                                                                                                                                       
Tools �• Dart 2.18.5 �• DevTools 2.15.0                                                                                                                                                                             
Launching lib/main.dart on eLinux in debug mode...                                                                                                                                                                  
Building an eLinux application with wayland backend in debug mode for arm64 target...       189.1s                                                                                                                  
Error waiting for a debug connection: The log reader stopped unexpectedly.                                                                                                                                          
Error launching application on eLinux.`

HidenoriMatsubayashi, 2022-12-07:
I guess your flutter_custom_devices.json is wrong or not enough.

WolsYang, 2022-12-08:
No, it doesn't work
```
$ flutter-client -b ./build/elinux/arm64/release/bundle/sample                                                                                                                                
embedder.cc (960): 'FlutterEngineCreateAOTData' returned 'kInvalidArguments'. Invalid ELF path specified.                                                                                                           
[ERROR][flutter_project_bundle.cc(82)] Failed to load AOT data from: /usr/bin/./build/elinux/arm64/release/bundle/sample/lib/libapp.so                                                                              
[ERROR][flutter_elinux_engine.cc(164)] Unable to start engine without AOT data.                                                                                                                                     
Failed to create view controller.  
```

HidenoriMatsubayashi, 2022-12-08:
You have to build with release mode...

```shell
flutter-elinux run -d elinux-wayland --release
```

WolsYang, 2022-12-09:
Is this correct?
But I didn't see any app come up on the screen.
Should I try to specify output screen?

```
~/sample# flutter-elinux run -d elinux-wayland --release
   Woah! You appear to be trying to run flutter as root.
   We strongly recommend running the flutter tool without superuser privileges.
  /
�📎
Flutter 3.3.9 �• channel unknown �• unknown source
Framework �• revision b8f7f1f986 (2 weeks ago) �• 2022-11-23 06:43:51 +0900
Engine �• revision 8f2221fbef
Tools �• Dart 2.18.5 �• DevTools 2.15.0
Launching lib/main.dart on eLinux in release mode...
Building an eLinux application with wayland backend in release mode for arm64 target...       152.7s

Flutter run key commands.
h List all available interactive commands.
c Clear the screen
q Quit (terminate the application on the device).
```

HidenoriMatsubayashi, 2022-12-09:
Can you check below?

```shell
$ ./build/elinux/arm64/release/bundle/sample -b <path_to_build>/build/elinux/arm64/release/bundle
```

WolsYang, 2022-12-12:
Here is the output
```
~/sample$ FLUTTER_LOG_LEVELS=TRACE ./build/elinux/arm64/release/bundle/sample -b .
Bus error
```

HidenoriMatsubayashi, 2022-12-12:
> Bus error

Please check your Wayland environment first...

HidenoriMatsubayashi, 2023-01-03:
I'll close this issue as no updates after my last comment.

closed 2023-01-03 by HidenoriMatsubayashi