sony/flutter-elinux #157: Pressing the 'q' key does not terminate the application on the remote device

makotosato-at, 2022-12-26:
Hello

I am trying the following commands on host pc(x86):
```
$ flutter-elinux run -d target_device_arm64
(snip)

Flutter run key commands.
r Hot reload. 🔥🔥🔥
R Hot restart.
h List all available interactive commands.
d Detach (terminate "flutter run" but leave application running).
c Clear the screen
q Quit (terminate the application on the device).

💪 Running with sound null safety 💪

An Observatory debugger and profiler on eLinux is available at:
http://127.0.0.1:37541/kkd7NI14PDo=/
The Flutter DevTools debugger and profiler on eLinux is available at:
http://127.0.0.1:9100?uri=http://127.0.0.1:37541/kkd7NI14PDo=/
```
The application starts correctly on the remote target device(arm64).
But, pressing the q key does not terminate the application.

How do I terminate an application on a remote target device?

HidenoriMatsubayashi, 2022-12-26:
Although this is the known issue, there is currently no resolution.

makotosato-at, 2022-12-28:
Thank you for your reply.
Well noted.

HidenoriMatsubayashi, 2023-08-16:
To address this issue, I made a change that adds `stopApp` command to custom_devices.json. Thanks.

HidenoriMatsubayashi, 2023-08-16:
Updated https://github.com/sony/flutter-elinux/wiki/Remote-target-devices#create-flutter_custom_devicesjson.

PR #202 (2023-08-16): custom-device: add customizable stopApp command
This change adds `stopApp` command to custom-devices json file to quit flutter app from host PC remotely.

~/.flutter_custom_devices.json example:
```
{
  "custom-devices": [
    {
      "id": "raspberry-pi4",
      "label": "Rasberry Pi 4",
      "sdkNameAndVersion": "Rasberry Pi 4",
      "enabled": true,
      "platform": "arm64",
      "backend": "wayland",
      "ping": [
        "ping", "-w", "500", "-c", "1", "192.168.10.22"
      ],
      "pingSuccessRegex": "ttl=",
      "install": [
        "scp", "-r", "${localPath}", "hidenori@192.168 [... cut]
commit a9573272 (2023-08-16): custom-device: add customizable stopApp command (#202)

This change adds `stopApp` command to custom-devices json file to quit flutter app from host PC remotely.

Fixed #157
closed 2023-08-16 by HidenoriMatsubayashi