sony/flutter-elinux #159: Unable to Cross Compile

johncharris, 2023-02-03:
I am unable to cross compile my application. I get an error saying `Could not open '/lib/ld-linux-aarch64.so.1': No such file or
    directory`

This file should exist:
```
ls -lh /home/john/ubuntu-sysroot/lib/ld-linux-aarch64.so.1 
lrwxrwxrwx 1 root root 39 Jul  6  2022 /home/john/ubuntu-sysroot/lib/ld-linux-aarch64.so.1 -> aarch64-linux-gnu/ld-linux-aarch64.so.1
```

I started the build with the command:
`flutter-elinux build elinux      --target-arch=arm64      --target-compiler-triple=aarch64-linux-gnu      --target-sysroot=/home/john/ubuntu-sysroot/`

Extended Output:
```
Failed to cmake:
-- The CXX compiler identification is Clang 14.0.6
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - failed
-- Check for working CXX compiler: /usr/bin/clang++
-- Check for working CXX compiler: /usr/bin/clang++ - broken
-- Configuring incomplete, errors occurred!
See also "/home/john/ioxstreams/iox_streams/build/elinux/arm64/release/CMakeFiles/CMakeOutput.log".
See also "/home/john/ioxstreams/iox_streams/build/elinux/arm64/release/CMakeFiles/CMakeError.log".

CMake Error at /snap/cmake/1216/share/cmake-3.25/Modules/CMakeTestCXXCompiler.cmake:63 (message):
  The C++ compiler

    "/usr/bin/clang++"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/john/ioxstreams/iox_streams/build/elinux/arm64/release/CMakeFiles/CMakeScratch/TryCompile-LsP2HP

    Run Build Command(s):/home/john/ubuntu-sysroot/usr/bin/gmake -f Makefile cmTC_453fa/fast && aarch64-binfmt-P: Could not open '/lib/ld-linux-aarch64.so.1': No such file or
    directory
```

HidenoriMatsubayashi, 2023-02-05:
Can you install binutils-aarch64-linu-gnu on Host?

```shell
sudo apt install binutils-aarch64-linux-gnu
```

johncharris, 2023-02-06:
I have that package. The system is looking for `/lib/ld-linux.so.1` while I have '/lib/ld-linux.so.2'