sony/flutter-elinux #161: can not run on rockchip device

herehere, 2023-02-07:
I got this error break while:
```
flutter-elinux run -d elinux-x11
```

and our device is rockchip & OS uname:
```
Linux firefly 4.19.232 #4 SMP Mon Sep 5 09:46:40 CST 2022 aarch64 aarch64 aarch64 GNU/Linux
```

error break:
```
Flutter 3.7.1 • channel unknown • unknown source
Framework • revision 7048ed95a5 (6 days ago) • 2023-02-01 09:07:31 -0800
Engine • revision 800594f1f4
Tools • Dart 2.19.1 • DevTools 2.20.1
Launching lib/main.dart on eLinux in debug mode...
Building an eLinux application with x11 backend in debug mode for arm64 target...         Compilation to SkSL failed.
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:
GLSL to SPIRV failed; Compilation error. 3 error(s) and 1 warning(s).
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:
warning: (version, profile) forced to be (460, core), while in source code it is (320, es)
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:9:
error: '#include' : Included file not found. for header name: flutter/runtime_effect.glsl
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:93:
error: 'FlutterFragCoord' : no matching overloaded function found
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:93:
error: '=' :  cannot convert from ' const float' to ' temp 2-component vector of float'

Building an eLinux application with x11 backend in debug mode for arm64 target...         ShaderCompilerException: Shader compilation of
"/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag"
to
"/home/firefly/Workspace/flutterELinuxSample/sample/build/elinux/flutter_assets/shaders/in
k_sparkle.frag" failed with exit code 1.
impellerc stdout:

impellerc stderr:
Compilation to SkSL failed.
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:
GLSL to SPIRV failed; Compilation error. 3 error(s) and 1 warning(s).
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:
warning: (version, profile) forced to be (460, core), while in source code it is (320, es)
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:9:
error: '#include' : Included file not found. for header name: flutter/runtime_effect.glsl
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:93:
error: 'FlutterFragCoord' : no matching overloaded function found
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:93:
error: '=' :  cannot convert from ' const float' to ' temp 2-component vector of float'

Building an eLinux application with x11 backend in debug mode for arm64 target...           193.0s
```

Could this project run on rockchip?

herehere, 2023-02-07:
with  ```flutter-elinux run -d elinux-wayland```
I got:
```
Flutter 3.7.1 • channel unknown • unknown source
Framework • revision 7048ed95a5 (6 days ago) • 2023-02-01 09:07:31 -0800
Engine • revision 800594f1f4
Tools • Dart 2.19.1 • DevTools 2.20.1
Launching lib/main.dart on eLinux in debug mode...
Building an eLinux application with wayland backend in debug mode for arm64 target...     Compilation to SkSL failed.
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:
GLSL to SPIRV failed; Compilation error. 3 error(s) and 1 warning(s).
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:
warning: (version, profile) forced to be (460, core), while in source code it is (320, es)
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:9:
error: '#include' : Included file not found. for header name: flutter/runtime_effect.glsl
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:93:
error: 'FlutterFragCoord' : no matching overloaded function found
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:93:
error: '=' :  cannot convert from ' const float' to ' temp 2-component vector of float'

Building an eLinux application with wayland backend in debug mode for arm64 target...     ShaderCompilerException: Shader compilation of
"/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag"
to
"/home/firefly/Workspace/flutterELinuxSample/sample/build/elinux/flutter_assets/shaders/in
k_sparkle.frag" failed with exit code 1.
impellerc stdout:

impellerc stderr:
Compilation to SkSL failed.
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:
GLSL to SPIRV failed; Compilation error. 3 error(s) and 1 warning(s).
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:
warning: (version, profile) forced to be (460, core), while in source code it is (320, es)
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:9:
error: '#include' : Included file not found. for header name: flutter/runtime_effect.glsl
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:93:
error: 'FlutterFragCoord' : no matching overloaded function found
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:93:
error: '=' :  cannot convert from ' const float' to ' temp 2-component vector of float'

Building an eLinux application with wayland backend in debug mode for arm64 target...       201.4s
```

HidenoriMatsubayashi, 2023-02-07:
Which glib version do you use on that device? Also, can you try the command below?

```
flutter-elinux run -d linux
```

HidenoriMatsubayashi, 2023-02-07:
> Linux firefly 4.19.232 #4 SMP Mon Sep 5 09:46:40 CST 2022 aarch64 aarch64 aarch64 GNU/Linux

It looks like your Linux platform is old.

herehere, 2023-02-07:
my glib version
```
ldd (Ubuntu GLIBC 2.31-0ubuntu9.9) 2.31
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```

and  here is the ```flutter-elinux run -d linux``` output:
```
Flutter 3.7.1 • channel unknown • unknown source
Framework • revision 7048ed95a5 (6 days ago) • 2023-02-01 09:07:31 -0800
Engine • revision 800594f1f4
Tools • Dart 2.19.1 • DevTools 2.20.1
Launching lib/main.dart on Linux in debug mode...
ERROR: Compilation to SkSL failed.
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag: warning: (version, profile) forced to be (460, core), while in source code it is (320, es)  
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:9: error: '#include' : Included file not found. for header name: flutter/runtime_effect.glsl 
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:93: error: 'FlutterFragCoord' : no matching overloaded function found                        
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:93: error: '=' :  cannot convert from ' const float' to ' temp 2-component vector of float'  
ERROR: Target debug_bundle_linux-arm64_assets failed: ShaderCompilerException: Shader compilation of "/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag" to "/home/firefly/Workspace/flutterELinuxSample/sample/build/flutter_assets/shaders/ink_sparkle.frag" failed with exit code 1.                                            
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag: warning: (version, profile) forced to be (460, core), while in source code it is (320, es)  
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:9: error: '#include' : Included file not found. for header name: flutter/runtime_effect.glsl 
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:93: error: 'FlutterFragCoord' : no matching overloaded function found                        
/opt/flutter-elinux/flutter/packages/flutter/lib/src/material/shaders/ink_sparkle.frag:93: error: '=' :  cannot convert from ' const float' to ' temp 2-component vector of float'  
Building Linux application...                                           
Exception: Build process failed
```

Which version of OS can work?
I'm using ubuntu 20.04.5 LTS

HidenoriMatsubayashi, 2023-02-07:
I was able to reproduce this issue. I need investigation.

HidenoriMatsubayashi, 2023-02-07:
This seems not to be flutter-elinux unique issue. I found  the same issue on official flutter project.
https://github.com/flutter/flutter/issues/116703

dawnhowe-sync, 2023-02-08:
I am having this issue trying to build in an docker container on a IMX8 target with  "TorizonCore with evaluation containers" (Toradex). This is my first attempt to try to build flutter for the IMX8 and am glad it's not just me.

herehere, 2023-02-08:
Is that mean this is the issue of flutter 3.7? Could I change flutter-elinux branch & version to test? with https://github.com/sony/flutter-embedded-linux

HidenoriMatsubayashi, 2023-02-08:
Yes. Can you try https://github.com/sony/flutter-elinux/releases/tag/3.3.10?

toplinuxsir, 2023-02-09:
I meet the same problem

herehere, 2023-02-10:
I got a new problem with testing 
```
https://github.com/sony/flutter-elinux/releases/tag/3.3.10
```

after run ```flutter-elinux --version``` the flutter version will set to ```0.0.0-unknown```

so I can't test flutter project, the error:
```
Flutter • channel unknown • unknown source
Framework • revision f3bc66195b (8 weeks ago) • 2022-12-14 16:33:18 -0800
Engine • revision 3316dd8728
Tools • Dart 2.18.6 • DevTools 2.15.0
fatal: not a git repository: '/opt/flutter-elinux/.git'
The current Flutter SDK version is 0.0.0-unknown.

Because provider_shopper depends on provider >=4.0.0-dev which requires Flutter SDK version >=1.12.1, version solving failed.                                                       
Running "flutter pub get" in provider_shopper...                        
pub get failed (1; Because provider_shopper depends on provider >=4.0.0-dev which requires
Flutter SDK version >=1.12.1, version solving failed.)
```

How could I change the version or I could checkout to a stable point?

HidenoriMatsubayashi, 2023-02-10:
Ah, please use https://github.com/sony/flutter-elinux/commit/19a1b05ea20675c36c705955c4148ad40b205161

sethmnielsen, 2023-03-30:
I was having this same issue on a Rock 5B board running Ubuntu Server 20.04.6. I tried the workaround mentioned in https://github.com/flutter/flutter/issues/116703 and I was able to get the sample demo to run successfully!

I downloaded the 3.7.0 flutter release, then copied the `shader_lib` directory to `flutter-elinux/flutter/bin/cache/artifacts/engine/linux-arm64`. Also, I kept `flutter-elinux` in my home directory so I could modify it (make sure `~/flutter-elinux/bin` is on your PATH):

```
curl -O https://storage.googleapis.com/flutter_infra_release/releases/stable/linux/flutter_linux_3.7.0-stable.tar.xz
tar -xvf flutter_linux_3.7.0-stable.tar.xz flutter/bin/cache/artifacts/engine/linux-x64/shader_lib
cp -r flutter/bin/cache/artifacts/engine/linux-x64/shader_lib flutter-elinux/flutter/bin/cache/artifacts/engine/linux-arm64
```
After that, the sample demo worked immediately. So the problem seems to be that the GLSL shader libary files are missing for `linux-arm64`.

CmST0us, 2023-04-11:
> Ah, please use [19a1b05](https://github.com/sony/flutter-elinux/commit/19a1b05ea20675c36c705955c4148ad40b205161)

I get my demo run on this commit.
could you make a tag on this commit or modify the current 3.3.10 tag?

HidenoriMatsubayashi, 2023-04-11:
Hi @CmST0us I did right now. Please use https://github.com/sony/flutter-elinux/releases/tag/3.3.10

HidenoriMatsubayashi, 2023-05-25:
Closing this issue as I think the issue has been fixed in flutter 3.10.x. I confirmed flutter-elinux worked fine on jetson orin nano developer kit.

closed 2023-05-25 by HidenoriMatsubayashi