sony/flutter-elinux #179: build error

toplinuxsir, 2023-04-26:
After create flutter elinux app
and run build
```
flutter-elinux build elinux --target-backend-type=x11
```
```
/usr/bin/clang++ CMakeFiles/cmTC_79067.dir/testCXXCompiler.cxx.o -o
    cmTC_79067
    /usr/bin/ld: can not find -lstdc++: no file or dir
    clang: error: linker command failed with exit code 1 (use -v to see
    invocation)
    gmake[1]: *** [CMakeFiles/cmTC_79067.dir/build.make:100：cmTC_79067] Error 1
    gmake[1]:
    leave directory “/home/orangepi/build/myapp/build/elinux/arm64/release/CMakeFiles/CM
akeT
    mp”
    gmake: *** [Makefile:127：cmTC_79067/fast] Error 2

```

fryette, 2023-06-05:
The same error occurred.
Any workarounds?

fryette, 2023-06-06:
The issue appears since in documentation needed to use Ubunty 18.04 and my machine was Ubunty 20.04
Changed docker to 20.04 and able to compile

HidenoriMatsubayashi, 2023-06-06:
Thanks. I see. I updated https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps#create-sysroot-for-arm64.

Closing this issue.

closed 2023-06-06 by HidenoriMatsubayashi