sony/flutter-elinux #181: Rotation issue

thomtomdup, 2023-05-03:
Hello.

Since 3.7 version i have a problem on Yocto projhect imx8 with wayland. When i want to rotate with parameter -r 90 half or all screen is freezed. It unfreeze when i touch many time the screen.

Have you seen this issue or not ?

Thank you,
Regards

HidenoriMatsubayashi, 2023-05-03:
Specifically, what version happens this issue? Also, have you tried the latest version?

thomtomdup, 2023-05-03:
The version is the last ( i have updated version today ) and did tests but same issue.

HidenoriMatsubayashi, 2023-05-05:
rotation option (`-r`) may not be working properly in the first place. Are you using 90 or 270 degree rotation?

thomtomdup, 2023-05-09:
Hello.
I tested it without -r and all work good.

HidenoriMatsubayashi, 2023-06-15:
Can you try to use Sway or other Wayland compositor instead of Weston?

HidenoriMatsubayashi, 2023-08-02:
I guess this issue is related to https://github.com/sony/flutter-embedded-linux/issues/349, which has already fixed.

closed 2023-08-02 by HidenoriMatsubayashi