sony/flutter-elinux #187: Bug on Scrolling

FizboIV, 2023-06-12:
Hi, 
I'm working on a simple machine interface using
- RaspberryPi 4b
- Raspberry 7" touch display 800x480
- OS: RaspberryPi OS Lite v 11 Bullseye
- Weston v.9.0.0

I implemented a scrollable table and every time I scroll everything except the table vanishes. 

How it looks like before scrolling:
[image]

How it looks like after scrolling:
[image]

I also build the flutter app for web. There's everything working fine and as expected.
Do you have any ideas how to fix it?

Thank you

HidenoriMatsubayashi, 2023-06-12:
Can you check if this issue happens flutter for Linux desktop or Android/iOS? I believe this doesn't depend on flutter-elinux. I think it depends on flutter-engine.

HidenoriMatsubayashi, 2023-06-12:
Just FYI.

> I also build the flutter app for web

flutter engine of the flutter for web is completely different from desktops and mobiles (iOS/Android). Thus, flutter for web is not helpful.

FizboIV, 2023-06-12:
I tested a bit today and it does work on iOS. I checked on iPhone and iPad. Unfortunately I can't test on other platforms. 

I also noticed that as long as I plug in a mouse and scroll with the mouse the bug doesn't not happen. Maybe it is a problem with the touch input?

HidenoriMatsubayashi, 2023-06-12:
Can you share the source code?

FizboIV, 2023-06-13:
code can be found here:
https://github.com/FizboIV/embedded_test

its the custom_table widget that is malfunctioning

HidenoriMatsubayashi, 2023-06-13:
Thanks. One more question. What display backend are you using on Raspberry Pi4?

HidenoriMatsubayashi, 2023-06-15:
> OS: RaspberryPi OS Lite v 11 Bullseye
> Weston v.9.0.0

How did you install weston?

HidenoriMatsubayashi, 2023-06-15:
Now, I could reproduce this issue.

HidenoriMatsubayashi, 2023-06-15:
@FizboIV 
This issue occurs on RPI4 + Weston v9, but it doesn't happen on RPI4 + Sway. Can you please try to run the app using Sway instead?

FizboIV, 2023-06-15:
That did the trick! Thank you very much! I really appreciate it. :)