sony/flutter-elinux #190: i.MX8 CPU to 100%

wimdewitte, 2023-06-23:
I'm evaluating elinux on an i.MX8 board, using wayland; with latest yocto of NXP.
App (= default test_app or from flutter/samples) is working, startup time is ok; if the app is running without being used, the CPU usage is negligible.
But from the moment a button is clicked; CPU usage for the flutter app increases to 100% for some seconds.
This make the app worthless and doesn't reflect the idea that flutter is lightweight.
I don't think the problem is related to graphical subsystem or whatever, but more like a process while loop running too fast.
Is anyone else experiencing the same problem on i.MX8?

HidenoriMatsubayashi, 2023-06-23:
Are you using release build? Does CPU usage 100% mean per All CPU cores? or Just 1 core? Also please ensure GPU H/W driver is enabled on your device.

wimdewitte, 2023-06-23:
yes, release build, gpu enabled, load is on all cores

HidenoriMatsubayashi, 2023-08-11:
Specifically, what i.mx8 series are you using? I haven't heard flutter utilization is so high. Can you share real logs like top command logs?

HidenoriMatsubayashi, 2023-08-11:
This is an example using flutter on i.MX8M board.
https://www.sequality.at/blog/2023/techtalk-flutter-on-embedded-linux/

[image]

HidenoriMatsubayashi, 2023-08-14:
Closing. Thank you.

closed 2023-08-14 by HidenoriMatsubayashi