sony/flutter-elinux #191: Debian 10 ARM64 run or build error

Li-00, 2023-07-14:
[image]

[image]

[image]
[image]

I can ping github.com but run or build demo Downloading elinux-common tools faild ， don's know how to fix

HidenoriMatsubayashi, 2023-07-14:
I guess this is your network environment issue. Are you accessing from China? Can you download https://github.com/sony/flutter-embedded-linux/releases/tag/cdbeda788a manually?

See also https://github.com/sony/flutter-elinux/issues/140#issuecomment-1477907892

Li-00, 2023-07-14:
> 

yes, I'm in china run this demo,and i download release so put in  cache path
[image]

Is there any other possibility? thank you for your answer.

Li-00, 2023-07-14:
My computer can access github through the proxy without hindrance, but the embedded device cannot use the proxy, and I tried the flutter-elinux precache --elinux command, and the problem was the same, stuck in Downloading elinux-common tools for a long time...

HidenoriMatsubayashi, 2023-07-15:
I created a new PR to add this use case.

1. Download the elinux artifacts from https://github.com/sony/flutter-embedded-linux/releases
2. export ELINUX_ENGINE_BASE_LOCAL_DIRECTORY=<path_to_directory_of_downloaded_artifacts>
3. run flutter commands

closed 2023-07-15 by HidenoriMatsubayashi

HidenoriMatsubayashi, 2023-07-15:
Created https://github.com/sony/flutter-elinux/wiki/Use-local-elinux-artifacts-without-downloading-from-GitHub.

Li-00, 2023-07-18:
With the latest update, you can set up the proxy by configuring /etc/profile

First of all
vim /etc/profile
Then, add it to the file 
http_proxy=http://ip:port
https_proxy=https://ip:port
export http_proxy
export https_proxy
Finally, refresh the file
source /etc/profile

But when I try to run the project after building the project, it prompts Failed to create view controller, my board is rk3568 Debain10 aarch64, and I'm following https://github.com/sony/flutter-elinux/issues/121 to try to solve it