sony/flutter-elinux #194: undefined reference to `drmIsMaster'

Li-00, 2023-07-19:
# My device detail: 
rk3568
SO：Debain 10   aarch64 
GPU so：libmali-bifrost-g52-g2p0-x11.so 

env version
cmake version 3.26.4
flutter 3.10.5   dart  3.0.5
flutter-elinux   3.10.5  3.0.5 

# My CMake.txt
elinux_sample/elinux/runner/CMakeLists.txt
`
cmake_minimum_required(VERSION 3.15)
project(runner LANGUAGES CXX)

if(FLUTTER_TARGET_BACKEND_TYPE MATCHES "gbm")
   add_definitions(-DFLUTTER_TARGET_BACKEND_GBM)
endif()

add_executable(${BINARY_NAME}
        "flutter_window.cc"
        "main.cc"
        "${FLUTTER_MANAGED_DIR}/generated_plugin_registrant.cc"
)
apply_standard_settings(${BINARY_NAME})
set(FLUTTER_LIB   ${CMAKE_CURRENT_SOURCE_DIR}/../flutter_lib/)
find_library(DRM_LIBRARY NAMES drm)
target_link_libraries(${BINARY_NAME} PRIVATE ${DRM_LIBRARY})
target_link_libraries(${BINARY_NAME} PRIVATE flutter)
target_link_libraries(${BINARY_NAME} PRIVATE flutter_wrapper_app)
target_link_libraries(${BINARY_NAME} PRIVATE ${FLUTTER_LIB}libffi.so )
target_link_libraries(${BINARY_NAME} PRIVATE ${FLUTTER_LIB}libwayland-cursor.so )
target_link_libraries(${BINARY_NAME} PRIVATE ${FLUTTER_LIB}libxkbcommon.so.0 )
target_link_libraries(${BINARY_NAME} PRIVATE ${FLUTTER_LIB}libmali-bifrost-g52-g2p0-x11.so)
target_link_libraries(${BINARY_NAME} PRIVATE ${FLUTTER_LIB}libdrm.so.2.4.0 )
target_link_libraries(${BINARY_NAME} PRIVATE ${FLUTTER_LIB}libwayland-server.so.0 )
target_link_libraries(${BINARY_NAME} PRIVATE ${FLUTTER_LIB}libc.so.6 )
target_link_libraries(${BINARY_NAME} PRIVATE ${FLUTTER_LIB}ld-linux-aarch64.so.1 )

target_include_directories(${BINARY_NAME} PRIVATE "${CMAKE_SOURCE_DIR}")
add_dependencies(${BINARY_NAME} flutter_assemble)
`
# The so I added
elinux_sample/elinux/flutter_lib$ ls
ld-linux-aarch64.so.1  libc.so.6  libdrm.so.2  libdrm.so.2.4.0  libffi.so  libmali-bifrost-g52-g2p0-x11.so  libwayland-cursor.so  libwayland-server.so.0  libxkbcommon.so.0

# Build error
flutter-elinux build elinux --target-arch=arm64 --target-sysroot=/ --target-backend-type=gbm --verbose

[  +21 ms] Failed to cmake build:
           [  0%] Built target flutter_assemble
           [  7%] Building CXX object flutter/CMakeFiles/flutter_wrapper_app.dir/ephemeral/cpp_client_wrapper/core_implementations.cc.o
           [ 30%] Building CXX object flutter/CMakeFiles/flutter_wrapper_plugin.dir/ephemeral/cpp_client_wrapper/core_implementations.cc.o
           [ 30%] Building CXX object flutter/CMakeFiles/flutter_wrapper_plugin.dir/ephemeral/cpp_client_wrapper/standard_codec.cc.o
           [ 30%] Building CXX object flutter/CMakeFiles/flutter_wrapper_app.dir/ephemeral/cpp_client_wrapper/standard_codec.cc.o
           [ 38%] Building CXX object flutter/CMakeFiles/flutter_wrapper_app.dir/ephemeral/cpp_client_wrapper/flutter_engine.cc.o
           [ 46%] Building CXX object flutter/CMakeFiles/flutter_wrapper_plugin.dir/ephemeral/cpp_client_wrapper/plugin_registrar.cc.o
           [ 53%] Building CXX object flutter/CMakeFiles/flutter_wrapper_app.dir/ephemeral/cpp_client_wrapper/flutter_view_controller.cc.o
           [ 61%] Linking CXX static library libflutter_wrapper_plugin.a
           [ 61%] Built target flutter_wrapper_plugin
           [ 69%] Linking CXX static library libflutter_wrapper_app.a
           [ 69%] Built target flutter_wrapper_app
           [ 76%] Building CXX object runner/CMakeFiles/elinux_sample.dir/flutter_window.cc.o
           [ 84%] Building CXX object runner/CMakeFiles/elinux_sample.dir/main.cc.o
           [ 92%] Building CXX object runner/CMakeFiles/elinux_sample.dir/__/flutter/generated_plugin_registrant.cc.o
           [100%] Linking CXX executable elinux_sample

           /usr/bin/ld: /media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/demo/elinux_sample/elinux/flutter/ephemeral/libflutter_elinux_gbm.so: undefined reference to `drmIsMaster'
           clang: error: linker command failed with exit code 1 (use -v to see invocation)
           make[2]: *** [runner/CMakeFiles/elinux_sample.dir/build.make:141: runner/elinux_sample] Error 1
           make[1]: *** [CMakeFiles/Makefile2:198: runner/CMakeFiles/elinux_sample.dir/all] Error 2
           make: *** [Makefile:136: all] Error 2
[   +7 ms]
           #0      throwToolExit (package:flutter_tools/src/base/common.dart:10:3)
           #1      NativeBundle.build (package:flutter_elinux/elinux_build_target.dart:412:7)
           <asynchronous suspension>
           #2      ELinuxBuilder.buildBundle (package:flutter_elinux/elinux_builder.dart:126:7)
           <asynchronous suspension>
           #3      BuildPackageCommand.runCommand (package:flutter_elinux/commands/build.dart:137:5)
           <asynchronous suspension>
           #4      FlutterCommand.run.<anonymous closure> (package:flutter_tools/src/runner/flutter_command.dart:1336:27)
           <asynchronous suspension>
           #5      AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
           <asynchronous suspension>
           #6      CommandRunner.runCommand (package:args/command_runner.dart:212:13)
           <asynchronous suspension>
           #7      FlutterCommandRunner.runCommand.<anonymous closure> (package:flutter_tools/src/runner/flutter_command_runner.dart:296:9)
           <asynchronous suspension>
           #8      AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
           <asynchronous suspension>
           #9      FlutterCommandRunner.runCommand (package:flutter_tools/src/runner/flutter_command_runner.dart:244:5)
           <asynchronous suspension>
           #10     run.<anonymous closure>.<anonymous closure> (package:flutter_tools/runner.dart:91:9)
           <asynchronous suspension>
           #11     AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
           <asynchronous suspension>
           #12     main (package:flutter_elinux/executable.dart:81:3)
           <asynchronous suspension>

[ +232 ms] ensureAnalyticsSent: 5ms
[   +9 ms] Running 0 shutdown hooks
[   +2 ms] Shutdown hooks complete
[   +5 ms] exiting with code 1

Li-00, 2023-07-19:
If I create a running project without adding a configuration so, what he prompts is https://github.com/sony/flutter-elinux/issues/121. And I'm not sure if this question is consistent with mine.

Li-00, 2023-07-19:
linaro@linaro-alip:/media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/demo/elinux_sample$ sudo find / -name libflutter_elinux_gbm.so
/media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/demo/elinux_sample/elinux/flutter/ephemeral/libflutter_elinux_gbm.so
/media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/demo/flutter_elinux2/elinux/flutter/ephemeral/libflutter_elinux_gbm.so
/media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/env/flutter-elinux/flutter/bin/cache/artifacts/engine/elinux-arm64-release/libflutter_elinux_gbm.so
/media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/env/flutter-elinux/flutter/bin/cache/artifacts/engine/elinux-arm64-debug/libflutter_elinux_gbm.so
/media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/env/flutter-elinux/flutter/bin/cache/artifacts/engine/elinux-x64-release/libflutter_elinux_gbm.so
/media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/env/flutter-elinux/flutter/bin/cache/artifacts/engine/elinux-x64-debug/libflutter_elinux_gbm.so
/media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/env/flutter-elinux/bin/cache/libflutter_elinux_gbm.so

I guess my libflutter_elinux_gbm.so version  can’t match，If so, which version should I specify

HidenoriMatsubayashi, 2023-07-19:
>  /usr/bin/ld: /media/linaro/ef0e9ea8-873f-4819-bc21-6790e2ddc7b7/demo/elinux_sample/elinux/flutter/ephemeral/libflutter_elinux_gbm.so: undefined reference to `drmIsMaster'
 
The cause is that `drmIsMaster' API doesn't seem to be supported in your environment for cross-building.

Li-00, 2023-07-19:
I found that my target-backend-type parameter was specified incorrectly, it should be x11 for my graph architecture, and it works fine after making the change

`flutter-elinux build elinux --target-arch=arm64 --target-sysroot=/ --target-backend-type=x11 --verbose`

Sorry, this is due to insufficient knowledge of the graphics backend library, thanks to HidenoriMatsubayashi for the follow-up reply

[image]