sony/flutter-elinux #197: Unification of Runtime Libraries

alberrizwan, 2023-07-28:
I need to create a runtime container for Flutter applications on the i.MX8 platform for Torizon platforms. My goal is to minimize the final image footprint as much as possible. Although I have already managed to reduce the size significantly, I still want to further optimize it. To achieve this, I plan to unify the following Flutter runtime libraries: elinux-arm64-debug, elinux-arm64-profile, and elinux-arm64-release. I would greatly appreciate any suggestions or input you may have on this matter.
[image]

HidenoriMatsubayashi, 2023-07-28:
> I plan to unify the following Flutter runtime libraries: elinux-arm64-debug, elinux-arm64-profile, and elinux-arm64-release.

Basically, there are all necessary. But if you don't need debug and profile flavors, elinux-arm64-debug and elinux-arm64-profile can be deleted. Naturally, in this case, flutter-elinux commands with debug/profile won't work.

HidenoriMatsubayashi, 2023-07-28:
Also, if there are unnecessary display backends that you don't support, their .so files can be deleted.

libflutter_elinux_eglstream.so for EGLStream
libflutter_elinux_gbm.so for GBM
libflutter_elinux_wayland.so for Wayland
libflutter_elinux_x11.so for X11

closed 2023-07-28 by HidenoriMatsubayashi