sony/flutter-elinux #200: Run command with screen size parameters

jvieira88, 2023-08-09:
Hi,

I'm running an application from WSL (using Ubuntu). I'm interested in running the application in portrait mode but I can't find a way besides running the compiled application in the bundle folder and specifying the `-w` and `-h` parameters. But if I do this, I can't attach and debug the application.
After configuring sway to use a different screen resolution (480 x 800) by setting the config file (under ~/.config/sway), and launching sway, I can see a window of the expected size. 
However, if I launch my application using `flutter-elinux run -d elinux-wayland`, the window defaults to 1280 x 720 I believe, and there's no way to resize it. I was expecting it to take the resolution from the configuration file.

Is there a way of specifying the screen resolution when using the `flutter-elinux run -d elinux-wayland` command or a way of taking the current sway configuration?

HidenoriMatsubayashi, 2023-08-09:
Hi, thank you for creating the issue. Yes, you are right. This is related to https://github.com/sony/flutter-elinux/issues/74. @psstoyanov any updates?

psstoyanov, 2023-08-10:
Not much - had been busy with a new role for awhile. 
Before that I got to the point of adding the new args in the elinux tool but couldn't figure out how to pass it down to the embedder.

Notes:
Possibly [flutter-elinux create](https://github.com/sony/flutter-elinux/blob/main/lib/commands/create.dart) could take the desired window size and set it as default for the project: https://github.com/sony/flutter-embedded-linux/blob/master/examples/flutter-wayland-client/flutter_embedder_options.h This looks to be more involved overall and likely requires adjustment in[ flutter-embedded-linux](https://github.com/sony/flutter-embedded-linux) to allow for such operation.

[flutter-elinux run](https://github.com/sony/flutter-elinux/blob/main/lib/commands/run.dart) should be less involved to implement as it would be limited to `flutter-elinux` itself. I got stuck trying to pass the argument down at the time.

HidenoriMatsubayashi, 2023-08-10:
Okay, then I'll address this issue.

HidenoriMatsubayashi, 2023-08-14:
I added `FLUTTER_ELINUX_CUSTOM_RUN_ARGS` env to pass elinux embedder options via flutter-elinux tool. It's not the best but there is no another way we can address as of now. Please use `FLUTTER_ELINUX_CUSTOM_RUN_ARGS` env to specify your favorite options of the embedder.

closed 2023-08-14 by HidenoriMatsubayashi