sony/flutter-elinux #205: Issue after update to flutter 3.13.0

jvieira88, 2023-08-17:
Hi,

There seems to be an issue after the flutter update to 3.13.0 (#204). Just had a quick look but it seems to affect some widgets such as the DropdownButton and other buttons. I reverted back to #201 and it seems fine.

Example with #204 
[image]

Example with #201:
[image]

Maybe it's not related with flutter-elinux but just thought to bring it up.

HidenoriMatsubayashi, 2023-08-17:
Sorry, what is your issue?

HidenoriMatsubayashi, 2023-08-17:
The following template source files were updated in flutter 3.13.0.

- [main.cc](https://github.com/sony/flutter-elinux/blob/main/templates/app/runner/main.cc)
- [flutter_embedder_options.h](https://github.com/sony/flutter-elinux/blob/main/templates/app/runner/flutter_embedder_options.h)

Can you check if the files you are using are up to date?

HidenoriMatsubayashi, 2023-08-18:
Can you please check if this issue happen on elinux-x11 backend too?

The root cause might be https://github.com/sony/flutter-embedded-linux/pull/376.

jvieira88, 2023-08-21:
Hi @HidenoriMatsubayashi,

Running with X11 results in the same issue. 
However, both `main.cc` and `flutter_embedder_options.h` local to the project (under `/elinux/runner/`) differ from the source files you mentioned. Is there a way of refreshing these local files without having to manually execute a run command every time  there is an update?

HidenoriMatsubayashi, 2023-08-21:
> However, both main.cc and flutter_embedder_options.h local to the project (under /elinux/runner/) differ from the source files you mentioned.

Okay. So please try https://github.com/sony/flutter-elinux/issues/205#issuecomment-1682880895.

> Is there a way of refreshing these local files without having to manually execute a run command every time there is an update?

Unfortunately, there is no way to upgrade local elinux c++ files in existing projects. Please replace your local files manually.

HidenoriMatsubayashi, 2023-08-21:
Great.

> Something to keep in mind the next time I need to update the repository :)

I'll consider automatic-migration way in flutter-elinux.

Closing. Thanks.

closed 2023-08-21 by HidenoriMatsubayashi