sony/flutter-elinux #211: Plugins working in debug mode but not in profile / release mode.

Arna-Maity, 2023-08-28:
I converted my existing flutter **(Flutter Version 3.10)** project into **flutter-elinux (Flutter Version 3.13)** based one as mentioned in the project wiki. Everything including plugins work fine when run in the debug mode, but the plugins don't working properly when run in profile / release mode. I am running **Ubuntu 20.04** and have followed the steps as mentioned in the wiki of this repo to setup flutter-elinux.
For example when running the camera plugin from the flutter-elinux-plugins repo in **debug mode** _( flutter-elinux run -d elinux-x11 )_:
( flutter-elinux run -d elinux-x11 )
```
══╡ EXCEPTION CAUGHT BY WIDGETS LIBRARY ╞═══════════════════════════════════════════════════════════
The following LateError was thrown building RtCamFeedView(dirty, state: _RtCamFeedViewState#d023d):
LateInitializationError: Field '_controller@277206264' has not been initialized.

The relevant error-causing widget was:
  RtCamFeedView
  RtCamFeedView:file:///home/arna/wasteboard_edge_application/wasteboard_ui_flutter/lib/ui/welcome/we
  lcome_view.dart:68:27

Pixel buffer size: width = 1280, height = 720
```

But when I run the same application in **release mode**  _( flutter-elinux run -d elinux-x11 --release )_ :

```
flutter: LateInitializationError: Field '_controller@552206264' has not been initialized.

flutter: #0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:308)
flutter: <asynchronous suspension>
flutter: #1      MethodChannel.invokeListMethod (package:flutter/src/services/platform_channel.dart:496)
flutter: <asynchronous suspension>
flutter: #2      AndroidCamera.availableCameras (package:camera_android/src/android_camera.dart:74)
flutter: <asynchronous suspension>
flutter: #3      _RtCamFeedViewState.startCamera (package:wasteboard_ui_flutter/ui/rtcamfeed/rtcamfeed_view.dart:30)
flutter: <asynchronous suspension>
```
Is there something I am missing that needs to be done extra to run the app in release mode?

HidenoriMatsubayashi, 2023-08-28:
I suppose this issue is the same as https://github.com/sony/flutter-elinux-plugins/issues/78. Are you using the latest video_player plugin (flutter-elinux-plugins)?

HidenoriMatsubayashi, 2023-08-28:
This is a flutter-elinux-plugin issue, but I might be wrong. camera plugin for elinux might need to be fixed to fix this issue.

Arna-Maity, 2023-08-28:
> I suppose this issue is the same as [sony/flutter-elinux-plugins#78](https://github.com/sony/flutter-elinux-plugins/issues/78). Are you using the latest video_player plugin (flutter-elinux-plugins)?

Didn't try it yet. Although, I tried some other plugins which are not specifically elinux plugins but plugins which support the Linux platform installed from pub.dev which worked fine in debug mode as well but not in profile/release. Will test and make some detailed observations and report back today. 

> This is a flutter-elinux-plugin issue, but I might be wrong. camera plugin for elinux might need to be fixed to fix this issue.

So this is specific to the flutter-elinux for flutter version 3.13 right and this does not exist for flutter 3.10 right?

HidenoriMatsubayashi, 2023-08-28:
> So this is specific to the flutter-elinux for flutter version 3.13 right and this does not exist for flutter 3.10 right?

Yes, correct.

HidenoriMatsubayashi, 2023-08-28:
This issue should be fixed in https://github.com/sony/flutter-elinux-plugins/pull/81. Thanks.

closed 2023-08-28 by HidenoriMatsubayashi

HidenoriMatsubayashi, 2023-08-28:
Okay, let me check.

HidenoriMatsubayashi, 2023-08-29:
@Arna-Maity Is it possible to share your test code (flutter project?) to reproduce a rotation issue?

Arna-Maity, 2023-08-29:
Have a look at this. The **StatefulWidget** managing the camera is the same as the one we are using in our project. 

Try running it in all the three debug, profile and release modes. In debug mode the orientation is correct but in profile and release modes it is rotated.

https://github.com/Arna-Maity/camera_test

Arna-Maity, 2023-08-29:
The camera rotation issue seems to be fixed after the **latest platform fix (x64)** (#213)   you merged into the **main** branch.  All the linux platform plugins which failed to run previously on 3.13.1 are now running fine. Thank you!

HidenoriMatsubayashi, 2023-08-29:
I see. Thanks.