sony/flutter-elinux #215: Error in running on wayland

baharanchi, 2023-08-31:
I have a [NanoPC-T4 ](https://www.friendlyelec.com/index.php?route=product/product&product_id=225) development board.
I tried to run the sample code on the board on Sway using the commands:
```
$ sway &
$ flutter-elinux run -d elinux-wayland
```

But I got the following error: 

```
Launching lib/main.dart on eLinux in debug mode...
Building an eLinux application with wayland backend in debug mode for arm64 target...        36.0s
[ERROR][EnvironmentEgl(22)] Failed to get the EGL display: eglGetError: EGL_SUCCESS
[ERROR][ContextEgl(79)] Failed to choose EGL surface config: eglGetError: EGL_BAD_DISPLAY
[ERROR][CreateOnscreenSurface(118)] Failed to create EGL window surface: eglGetError: EGL_BAD_DISPLAY
[ERROR:flutter/shell/platform/embedder/embedder_surface_gl.cc(121)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
[ERROR][MakeCurrent(72)] Failed to make the EGL context current: eglGetError: EGL_BAD_DISPLAY
[ERROR:flutter/shell/gpu/gpu_surface_gl_skia.cc(44)] Could not make the context current to set up the Gr context.
[ERROR][MakeCurrent(72)] Failed to make the EGL context current: eglGetError: EGL_BAD_DISPLAY
[ERROR:flutter/shell/gpu/gpu_surface_gl_skia.cc(82)] Could not make the context current to set up the Gr context.
[ERROR:flutter/shell/common/platform_view.cc(75)] Failed to create platform view rendering surface
[ERROR][DispatchEvent(1263)] Wayland display is invalid.
```

Here is the output of `flutter-elinux devices`:
```
3 connected devices:

Linux (desktop)  • linux          • linux-arm64    • Debian GNU/Linux 11 (bullseye) 4.19.193
eLinux (desktop) • elinux-wayland • flutter-tester • Debian GNU/Linux 11 (bullseye) 4.19.193
eLinux (desktop) • elinux-x11     • flutter-tester • Debian GNU/Linux 11 (bullseye) 4.19.193
```

I should note that I have no problem running the sample on X11 using the following command:
`$ flutter-elinux run -d elinux-x11`
The problem occurs only in Wayland mode.

HidenoriMatsubayashi, 2023-08-31:
Fist of all, Does any native Wayland client (not flutter) app work fine on your board (+sway)?

baharanchi, 2023-08-31:
> 
Sway itself works fine and I could run "LXTerminal" on it. But I don't know about any native Wayland client apps. Can you give me an example so I can test it on the board?

HidenoriMatsubayashi, 2023-08-31:
How about `weston-simple-egl`?

baharanchi, 2023-08-31:
`weston-simple-shm` works fine though.

HidenoriMatsubayashi, 2023-08-31:
Okay.

Can you please try the followings?

- Unset DISPLAY
- Set EGL_PLATFORM=wayland

baharanchi, 2023-08-31:
I tried but it made no difference. I got the exact same error.  And also `weston-simple-egl` still produces segmentation fault.

baharanchi, 2023-08-31:
Okay thanks. Is there any difference in terms of performance or features between X11 and Wayland?
I want to determine if I really need the Wayland backend.

HidenoriMatsubayashi, 2023-08-31:
It depends on system requirements and target board spec, but in general x11 system is heavier than wayland. And also x11 requires a number of dependent libraries.

HidenoriMatsubayashi, 2023-08-31:
> Do they behave the same in areas like graphics hardware acceleration?

Yes, I think so.

> Do you expect Canvas to be faster on Wayland than X11?

I'm not sure. I recommend you to evaluate it first on other arm64 devices such as Raspberry Pi 4 that I've confirmed flutter-elinux (both x11 and wayland) works fine.

HidenoriMatsubayashi, 2023-09-01:
@baharanchi Just FYI. If you run flutter with wayland backend, please try to use `--async-vblank` option to improve performance.
See the links below:
- https://github.com/sony/flutter-elinux/wiki/Running-flutter-apps#run-without-the-flutter-elinux-command
- https://github.com/sony/flutter-embedded-linux/wiki/How-to-run-Flutter-apps#command-options

baharanchi, 2023-09-18:
I solved the problem by changing the Linux image. At first I was using the official Debian image mentioned in the [NanoPC-T4 wiki](https://wiki.friendlyelec.com/wiki/index.php/NanoPC-T4) which had the problem. Then I tried the unofficial [Armbian image for the device](https://www.armbian.com/nanopc-t4/) and it resolved the issue. I think it has something to do with [this forum post](https://forum.armbian.com/topic/16516-rk3399-legacy-multimedia-framework/).

There was another problem though. I couldn't run Sway or the sample application with Wayland backend directly from the default desktop. So I decided to run Sway from a tty and then run the sample application from a terminal inside Sway. In order to start the terminal in Sway, you can use `Mod + Enter` key combinations (Mod is the Super key by default and can be changed in the Sway config file). Also the default terminal in Sway is `foot` which should be installed using `sudo apt install foot` (This can also be changed in the Sway config file). I hope this helps anyone who might be having the same problem.

I should also note that I feel Flutter applications have better performance on Wayland than X11 in my experience.

closed 2023-09-18 by baharanchi