sony/flutter-elinux #217: Mouse / Touch input support for DRM-GBM Backend

Arna-Maity, 2023-09-03:
Are mouse and touch inputs supported for DRM-GBM Backends?

HidenoriMatsubayashi, 2023-09-03:
Yes. Thanks.

closed 2023-09-03 by HidenoriMatsubayashi

Arna-Maity, 2023-09-05:
@HidenoriMatsubayashi Actually I was unable to use the mouse/trackpad using the GBM backend on an x64 PC.

Also, is there a way to terminate the app once it is already running?

HidenoriMatsubayashi, 2023-09-05:
Does an user who you are using for running your flutter app have a permission to access input devices(/dev/input/xxx)?

> Also, is there a way to terminate the app once it is already running?

This is a general question. Please google it.

Arna-Maity, 2023-09-06:
> Does an user who you are using for running your flutter app have a permission to access input devices(/dev/input/xxx)?

My user account does have **RW** access, does it also need the **execute (X)** bit set?

[image]

> This is a general question. Please google it.

Tried most of the general ways to close the application but the issue is once the application is opened (in **DRM/GBM** mode), it doesn't seem to respond to any kind of user input. It was also not possible to switch to a different **tty** session to kill the process. Maybe I suppose this is related to my inputs not working when the application is running **in DRM/GBM** mode.

talksik, 2023-09-09:
@Arna-Maity were you able to change the permissions and get this working?

@HidenoriMatsubayashi I believe a post like this should be re-opened if the issue is not resolved.

Arna-Maity, 2023-09-09:
> were you able to change the permissions and get this working?

Not yet. Still facing this issue.

talksik, 2023-09-09:
@Arna-Maity please let me know when you have changed the permissions with `chmod` and tested.

Arna-Maity, 2023-09-10:
So I tried giving **RW** permissions to the **Others** field of all the input devices `/dev/input/event*`, `/dev/input/mouse*` and `/dev/input/mice` using the following:

```
sudo chmod 666 /dev/input/event* /dev/input/mouse* /dev/input/mice
```

That seemed to do the job.

Thanks, @talksik  @HidenoriMatsubayashi .

talksik, 2023-09-10:
Perfect thank you for closing the loop!