sony/flutter-elinux #222: Flickering on Raspberry Pi 3 B+

baharanchi, 2023-09-19:
The sample application is not working properly on my Raspberry Pi 3 B+ on Sway. After hovering or clicking on the "+" button, the whole app flickers or sometimes goes black except for the button. 

I recorded a video of this behavior using my phone. you can watch it here:

https://github.com/sony/flutter-elinux/assets/25637303/bb3bf364-01ef-4ee1-9c2d-7396de92c58c

HidenoriMatsubayashi, 2023-09-19:
Please share Raspberry Pi OS and sway versions. First of all, this kind of H/W or SW specific issue will take a long time to be fixed or closed since it is not easy to reproduce the issue and it needs the target H/W.

baharanchi, 2023-09-20:
Sway version is v1.5.

I used the official Raspbian image "2023-05-03-raspios-bullseye-arm64.img".
Here are the contents of `/ets/os-release`:
```PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

And here is the result of `uname -a`:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

HidenoriMatsubayashi, 2023-09-20:
By the way, Raspberry Pi 3 has only 1GB memory, right? I guess it's not enough for flutter-elinux. How did you build the artifacts on Raspberry Pi 3?

baharanchi, 2023-09-20:
That's right. I increased the swap size to 1GB, stopped memory heavy
services like lightdm, ran the command from a tty and just waited for it to
finish :D

baharanchi, 2023-09-20:
Curently I don't have access to a Raspberry Pi 4. I need it to work on 3 B+
if possible.

HidenoriMatsubayashi, 2023-09-21:
For now, I've tested it on Raspberry Pi4 (completely same versions), but the issue didn't be reproduced.

HidenoriMatsubayashi, 2023-09-21:
I was also able to reproduce a similar rendering issue on Raspberry Pi 3 B+.

HidenoriMatsubayashi, 2023-09-21:
This issue is the same as https://github.com/sony/flutter-embedded-linux/issues/334. I'll disable dirty region management temporarily as a workaround. I'll close this issue.

Thanks.

HidenoriMatsubayashi, 2023-09-21:
Re-opened as this issue might not be related to https://github.com/sony/flutter-embedded-linux/issues/334. I can still rendering issue even using https://github.com/sony/flutter-embedded-linux/pull/385.

HidenoriMatsubayashi, 2023-09-22:
Waiting for flutter/engine's bug fix. See https://github.com/sony/flutter-embedded-linux/issues/334#issuecomment-1730838409

HidenoriMatsubayashi, 2023-11-10:
v3.13.9 should fix this issue as it includes https://github.com/sony/flutter-embedded-linux/pull/388 for now. Closing. Thanks.

closed 2023-11-10 by HidenoriMatsubayashi