sony/flutter-elinux #228: Error launching application on i.MX 8

hareendranmg, 2023-10-20:
Hi,

I'm running flutter-elinux on my Toradex Apalis i.MX 8 with Yocto build OS based on [tdx-reference-multimedia-image](https://developer.toradex.com/linux-bsp/os-development/bsp-layers-and-reference-images-for-yocto-project-software/)
[image]

But when i run the application, below error occurs.

```
[ +242 ms] Error waiting for a debug connection: The log reader stopped unexpectedly.
[  +17 ms] Error launching application on eLinux.
```

<details>
<summary>Full log
</summary>

```bash
root@apalis-imx8-07013820:~/sample# flutter-elinux run -v
[ +651 ms] Unable to locate an Android SDK.
[ +202 ms] executing: uname -m
[  +30 ms] Exit code 0 from: uname -m
[  +15 ms] aarch64
[+1935 ms] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
[        ] Artifact Instance of 'LegacyCanvasKitRemover' is not required, skipping update.
[  +10 ms] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
[ +107 ms] Artifact Instance of 'ELinuxEngineArtifacts' is not required, skipping update.
[   +4 ms] Artifact Instance of 'MaterialFonts' is not required, skipping update.
[        ] Artifact Instance of 'GradleWrapper' is not required, skipping update.
[        ] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
[        ] Artifact Instance of 'LegacyCanvasKitRemover' is not required, skipping update.
[        ] Artifact Instance of 'FlutterSdk' is not required, skipping update.
[        ] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
[   +7 ms] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FontSubsetArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'PubDependencies' is not required, skipping update.
[ +157 ms] Skipping pub get: version match.
[ +558 ms] Initializing file store
[  +50 ms] Skipping target: gen_localizations
[  +47 ms] gen_dart_plugin_registrant: Starting due to {InvalidatedReasonKind.inputChanged: The following inputs have updated contents: /home/root/sample/.dart_tool/package_config_subset}
[ +136 ms] gen_dart_plugin_registrant: Complete
[   +5 ms] Skipping target: _composite
[  +11 ms] complete
[  +30 ms] Launching lib/main.dart on eLinux in debug mode...
[  +20 ms] /opt/flutter/bin/cache/dart-sdk/bin/dart --disable-dart-dev /opt/flutter/bin/cache/dart-sdk/bin/snapshots/frontend_server.dart.snapshot --sdk-root
/opt/flutter/bin/cache/artifacts/engine/common/flutter_patched_sdk/ --incremental --target=flutter --experimental-emit-debug-metadata -DFLUTTER_WEB_AUTO_DETECT=true
-DFLUTTER_WEB_CANVASKIT_URL=https://www.gstatic.com/flutter-canvaskit/a794cf2681c6c9fe7b260e0e84de96298dc9c18b/ --output-dill /tmp/flutter_tools.AVUEYS/flutter_tool.BNTMXJ/app.dill --packages
/home/root/sample/.dart_tool/package_config.json -Ddart.vm.profile=false -Ddart.vm.product=false --enable-asserts --track-widget-creation --filesystem-scheme org-dartlang-root --initialize-from-dill
build/d1e25e033e83e0237fd5ee906d81ede1.cache.dill.track.dill --verbosity=error --flutter-widget-cache --enable-experiment=alternative-invalidation-strategy
[  +90 ms] Building app
[... 431 lines cut ...]
[   +1 ms] -- Set runtime path of "/home/root/sample/build/linux/arm64/debug/bundle/sample" to "$ORIGIN/lib"
[   +1 ms] -- Installing: /home/root/sample/build/linux/arm64/debug/bundle/data/icudtl.dat
[   +6 ms] -- Installing: /home/root/sample/build/linux/arm64/debug/bundle/lib/libflutter_linux_gtk.so
[ +258 ms] -- Installing: /home/root/sample/build/linux/arm64/debug/bundle/data/flutter_assets
[   +1 ms] -- Installing: /home/root/sample/build/linux/arm64/debug/bundle/data/flutter_assets/fonts
[        ] -- Installing: /home/root/sample/build/linux/arm64/debug/bundle/data/flutter_assets/fonts/MaterialIcons-Regular.otf
[  +11 ms] -- Installing: /home/root/sample/build/linux/arm64/debug/bundle/data/flutter_assets/AssetManifest.bin
[        ] -- Installing: /home/root/sample/build/linux/arm64/debug/bundle/data/flutter_assets/NOTICES.Z
[        ] -- Installing: /home/root/sample/build/linux/arm64/debug/bundle/data/flutter_assets/kernel_blob.bin
[ +253 ms] -- Installing: /home/root/sample/build/linux/arm64/debug/bundle/data/flutter_assets/FontManifest.json
[   +1 ms] -- Installing: /home/root/sample/build/linux/arm64/debug/bundle/data/flutter_assets/AssetManifest.json
[        ] -- Installing: /home/root/sample/build/linux/arm64/debug/bundle/data/flutter_assets/version.json
[        ] -- Installing: /home/root/sample/build/linux/arm64/debug/bundle/data/flutter_assets/packages
[        ] -- Installing: /home/root/sample/build/linux/arm64/debug/bundle/data/flutter_assets/packages/cupertino_icons
[   +1 ms] -- Installing: /home/root/sample/build/linux/arm64/debug/bundle/data/flutter_assets/packages/cupertino_icons/assets
[        ] -- Installing: /home/root/sample/build/linux/arm64/debug/bundle/data/flutter_assets/packages/cupertino_icons/assets/CupertinoIcons.ttf
[        ] -- Installing: /home/root/sample/build/linux/arm64/debug/bundle/data/flutter_assets/shaders
[   +1 ms] -- Installing: /home/root/sample/build/linux/arm64/debug/bundle/data/flutter_assets/shaders/ink_sparkle.frag
[  +41 ms] Building Linux application... (completed in 123.9s)
[ +525 ms] (sample:3844): Gdk-CRITICAL **: 07:33:03.713: gdk_window_get_state: assertion 'GDK_IS_WINDOW (window)' failed
[   +4 ms] ** (sample:3844): WARNING **: 07:33:03.726: Failed to start Flutter renderer: No GL implementation is available

```

</details>

HidenoriMatsubayashi, 2023-10-20:
> [ +242 ms] Error waiting for a debug connection: The log reader stopped unexpectedly.

This error may depend on your target board environment.

Can you try the following command and put its logs?

```
$ /home/root/sample/build/elinux/arm64/debug/bundle/sample -b .
```

If the command works fine, you will see the logs like below.

```
libEGL warning: pci id for fd 4: 1ab8:0010, driver (null)

pci id for fd 5: 1ab8:0010, driver (null)
flutter: The Dart VM service is listening on http://127.0.0.1:39287/hBEMannkxoA=/
```

HidenoriMatsubayashi, 2023-10-28:
I'll close this since there is no response. Please let me know if you have still this issue. I'll re-open it. Thanks.

closed 2023-10-28 by HidenoriMatsubayashi