sony/flutter-elinux #231: Segmentation fault

JGNS, 2023-11-21:
I have compiled the product and run it successfully with my friends, but it cannot be started with x11 on my linux development board. So after I get build/.../bundle/, what do I need to set up in the new linux? I used ./demo -b ./ -w 960 -h 640 -r 0 to start, and it prompted me that I could not find libflutter_engine.so, so I copied libflutter_engine.so in the bundle to /usr/lib, and now I was prompted with Segmentation fault , there is no other prompt information.

HidenoriMatsubayashi, 2023-11-21:
If you need a help, you should give more details such as how to build, logs and etc. No one can resolve your problems without the details. Thanks.

HidenoriMatsubayashi, 2023-11-21:
Closing. If you need help, please share more details. Thanks.

closed 2023-11-21 by HidenoriMatsubayashi