sony/flutter-elinux #245: Unable to cross compile with Yocto SDK

voloder, 2024-02-24:
Hello, I am trying to compile using [Variscite's Yocto toolchain](https://variwiki.com/index.php?title=DART-6UL_Yocto&release=mx6ul-yocto-kirkstone-5.15.71_2.2.0-v1.0), but I am getting the following errors:

```
.../elinux/flutter/ephemeral/cpp_client_wrapper/core_implementations.cc:16:10: fatal error:
'cassert' file not found
#include <cassert>
         ^~~~~~~~~

fatal error: 'gnu/stubs-soft.h' file not found
# include <gnu/stubs-soft.h>
          ^~~~~~~~~~~~~~~~~~
```
Things I've done:
- Added latest meta-flutter and meta-clang layers to the sdk
- Bitbaked with `-c populate_sdk`
- Installed SDK
- Symlinked ld to the arm one
`sudo cp -a arm-fslc-linux-gnueabi/arm-fslc-linux-gnueabi-ld ld` in /opt/fslc-framebuffer/4.0/sysroots/x86_64-fslcsdk-linux/usr/bin
- Added the following to OEToolchainConfig.cmake in the SDK sysroot
```
set ( CMAKE_C_COMPILER_WORKS 1 )
set ( CMAKE_CXX_COMPILER_WORKS 1 )
```
- Ran
```sh
flutter-elinux build elinux --target-arch=arm64 \
     --target-sysroot=/opt/fslc-framebuffer/4.0/sysroots/cortexa7t2hf-neon-fslc-linux-gnueabi \
     --target-backend-type=gbm
```

Any help would be appreciated, thank you!

HidenoriMatsubayashi, 2024-02-25:
Are you using PNC (persuade) build with Yocto? If yes, I don't think you need to use '--target-sysroot' and '--target-arch' option.

voloder, 2024-02-25:
> Are you using PNC (persuade) build with Yocto? If yes, I don't think you need to use `--target-sysroot` and `--target-arch` option.

Removed it, tried again, still happens. I am not sure what PNC is, could you elaborate?

andreascian, 2024-03-27:
@voloder you're using a 32 bit cross-toolchain (and this is correct because DART-6UL uses an iMX6UL SOC, which has a Cortex-A7 core) but using `--target-arch=arm64`

usually, when building with a Yocto toolchain, you just have to setup the environment and:

```
 . /opt/yocto/sdk/desk-mx8m-l-4.0.0/environment-setup-armv8a-poky-linux
flutter-elinux build elinux --target-arch=arm64 --target-compiler-triple=aarch64-poky-linux-
```

I'm pretty new to flutter-elinux but it seems that aarch32 is **NOT** supported

nguyenlkdn, 2024-04-03:
Refer https://bugs.llvm.org/show_bug.cgi?id=13796

I think it is clang with hardfp issue.